Koenkk/zigbee2mqtt #23453: zigbee2mqtt is not picking up new IP address for serial port in configuration.yaml

victorhooi, 2024-07-29:
### What happened?

I have zigbee2mqtt 1.39.0 running in a Docker container.

It is connected to a Ethernet-based Zigbee coordinator device ([UZG-01](https://uzg.zig-star.com/product/)).

I had to power-cycle the UZG-01 device, as I was trying to diagnose an issue with it. This caused the IP address to change, after a while - which then caused my zigbee2mqtt container to repeatedly crash and restart (assumedly because it couldn't open that serial port).

I edited the `/var/opt/zigbee2mqtt/data/configuration.yaml` to point to the new IP address for the UZG-01.

```
$ head configuration.yaml
homeassistant:
  legacy_triggers: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://10.5.1.36
serial:
  port: tcp://10.5.1.249:6638
  baudrate: 115200
  disable_led: false
```

However, ever after this, the zigbee2mqtt container is repeatedly crashing.

I checked the logs, and it seems that it's still attempting to find the UZG-01 at the old IP address (10.5.1.45) - even though the configuration.yaml file reflects the new IP address:
```
[2024-07-26 13:27:41] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-26 13:27:41] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-07-26 13:27:41] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-26 13:27:42] info: 	zh:zstack:znp: Opening TCP socket with 10.5.1.45:6638
[2024-07-26 13:27:45] info: 	zh:zstack:znp: Socket error
[2024-07-26 13:27:45] error: 	z2m: Error while starting zigbee-herdsman
[2024-07-26 13:27:45] error: 	z2m: Failed to start zigbee
[2024-07-26 13:27:45] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-26 13:27:45] error: 	z2m: Exiting...
[2024-07-26 13:27:45] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:191:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:188:25)
    at Logger.info (/app/lib/util/logger.ts:201:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:113:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:517:28)
    at TCP.<anonymous> (node:net:350:12)
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
```

It seems for some reason, zigbee2mqtt is not picking up the changed configuration.yaml - and is simply somehow remembering the old IP address somewhere?

### What did you expect to happen?

zigbee2mqtt should pickup the changed IP address, and start normally, without crashing.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20240710

### Adapter

UZG-01

### Setup

Docker container

maciekdnd, 2024-08-02:
Not sure if this is related, but to be sure check this:  #23355 
`rm opt/zigbee2mqtt/data/coordinator_backup.json
systemctl restart zigbee2mqtt.service`

nblount, 2024-09-25:
I have a similar problem running version 1.40.1, trying to change Zigbee adapter from Sonoff USB to SLZB-06 Ethernet and regardless of where I change the settings it always defaults back to using /dev/ttyUSB0. I have tried changing to tcp://slzb-06.local:6638/ in configuration.yaml and via the GUI but it just connects to /dev/ttyUSB0, the GUI switches back instantly but the configuration.yaml file has the correct settings.

configuration.yaml:
serial:
  port: tcp://slzb-06.local:6638
  baudrate: 115200
  adapter: zstack
  disable_led: true

GUI:
[image]

I have tried removing the coordinator_backup.json but that makes no difference.

closed 2024-12-28 by github-actions[bot]