Koenkk/zigbee2mqtt #23455: [New device support]: Soil sensor _TZE284_aao3yzhs

Barneybaer84, 2024-07-30:
### Link

https://a.aliexpress.com/_EvMR48F

### Database entry

{"id":94,"type":"EndDevice","ieeeAddr":"0xa4c13880ab293656","nwkAddr":19075,"manufId":4417,"manufName":"_TZE284_aao3yzhs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"'�;.\u00135�;.e6�;.\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":77,"manufacturerName":"_TZE284_aao3yzhs","zclVersion":3,"modelId":"TS0601","powerSource":3}},"genScenes":{"attributes":{"nameSupport":0}}},"binds":[{"cluster":25,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8469b","endpointID":1},{"cluster":61184,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8469b","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1722347988209}

### Comments

Here is my external converter for this soil sensor. Which I received today. The Tuya datapoints are the same as the TS0601_soil. The temperature must be divided by 10. 
[image]

Source Code:

`https://github.com/sprut/Hub/issues/3388#issue-2421419978`

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_aao3yzhs',
        },
    ],
    model: 'TS0601_soil_3',
    vendor: 'TuYa',
    description: 'Soil sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.soil_moisture(), tuya.exposes.temperatureUnit(), e.battery(), tuya.exposes.batteryState()],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                [3, 'soil_moisture', tuya.valueConverter.raw],
                [5, 'temperature', tuya.valueConverter.divideBy10],
                [9, 'temperature_unit', tuya.valueConverter.temperatureUnit],
                [14, 'battery_state', tuya.valueConverter.batteryState],
                [15, 'battery', tuya.valueConverter.raw],
            ],
        
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

Koenkk, 2024-08-01:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Barneybaer84, 2024-08-01:
PR Done :)

closed 2024-08-03 by Koenkk

andiukas, 2024-08-08:
Soil Moisture Sensor does not update

rrakso, 2024-08-29:
`M***K` thanks for your comment [under this product](https://pl.aliexpress.com/item/1005007443398328.html) with issue URL pointing here! 😀

Ramotny, 2024-08-29:
Hello.
Sorry to reopen the issue but I've entered all data described here above in configuration.yaml and definition file as "Support new device" of Zigbee2MQTT help said but still can NOT read this sensor. Can You help me?

jjvelar, 2024-09-01:
> Hello. Sorry to reopen the issue but I've entered all data described here above in configuration.yaml and definition file as "Support new device" of Zigbee2MQTT help said but still can NOT read this sensor. Can You help me?

hi!
check identation in configuration file.
That's my configuration. I paired and the device was identified.
Temperature sensor is working well and humidity is "null", but I guess it will take some time to show up.
I hope it helps.

[image]

[image]

Ramotny, 2024-09-02:
Hello. In the meantime case has been solved "by itself" because Z2M update 1,40 already includes this sensor and has been (finally) idnetified and works ok. Honestly I will not go backwards to test Your work - sorry for that. Surely comes another case.
I think my error was that I added another "advanced" section instead of adding "- external converter" to the already existing one.
I'm still dumb about Z2M and HA :(
PS: You can force sensor transmission (data refresh) by pressing shortly button under cover.

konehead01, 2024-10-13:
Hi, i have bought 3 of these sensors, but can't add them to Zigbee2mQTT ( 1.40.2-1). 

What i do:
- Set Zigbee2MQTT in pairing modus
- Add battery to plant monitor (green light is flashing) OR i hold the reset button for 4 seconds (green lights flashing as well)
- See nothing in the device list (bluethooth via app is working well. 

What do i wrong?

I bought this zigbee version: https://a.aliexpress.com/_EJOY9Mh

Ramotny, 2024-10-14:
Hi Konehead - I've ca 10 pcs of these sensors (from 3 sources) and there were no any problems at all neither with pairing not using. AFAIR only one needed the pairing 2-3 times untill finally paired.
Try adding near to the coordinator and when paired move to target location. Please be aware that the TX of these sensors is not very powerful.
Rgs Greg
PS: The only one bad thing I might say about these sensors it that they consume batteries quite quickly - I use NiMH 800mAh accu and they last ca 2-2,5 months. Also the manufacturer could make silicone caps green instead of blue. Last but not least there should be possibility to calibrate battery type or 0% voltage.
EDIT: with NiMH it lasted ca 10 days since 0% report until it died definitely (no data update).

konehead01, 2024-10-14:
Hi Remotny, 

Thanks for your reply, I appriciate. Will try to-night the follwoing:
1) Will sit next to the Zigbee main controler
2) Will remove the batteries, add, wait for 10 sec, add batterie en hit de reset button for 10 sec (untill green light is flashing again)
3) hit pairing button in Z2M

Is this the right process to follow?

Ramotny, 2024-10-14:
Start Z2M paring ("Permit
 all") before putting batteries back. It may take a while before pairing is done. These are mine (forget #3 - it's another type).
[image]

konehead01, 2024-10-15:
Really appriciate your input, but it doesn't work.  I have tried multiple pairings session, also to put the device 10 cm from the codinator... will contact the seller

MatthewRuddy, 2024-10-16:
> hi! check identation in configuration file. That's my configuration. I paired and the device was identified. Temperature sensor is working well and humidity is "null", but I guess it will take some [...]

Did you ever get yours to display humidity/moisture? I have gotten 2x of these today and both are still showing "NA" for humidity half a day later.

elzershark, 2024-10-22:
> Ich schätze Ihren Beitrag sehr, aber es funktioniert nicht. Ich habe mehrere Pairing-Sitzungen versucht und das Gerät auch 10 cm vom Kodiergerät entfernt platziert ... werde den Verkäufer kontaktieren

Du hast kein Zigbee, sondern Bluetooth gekauft.