Koenkk/zigbee2mqtt #23456: All the entities were duplicated after update of ha/core

Groeldor, 2024-07-30:
### What happened?

I've been updating HA/core to 2024.7.4, after service reboot, all my cards went gray, with the error: This entity is no longer being provided by the mqtt integration. If the entity is no longer in use, delete it in settings.

After having a look, it seems all my entities are duplicated, the old name keep the historical data, the new one old_name_2 are now the relevant one for new data.

### What did you expect to happen?

Nothing should have changed

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

v2.3.6

### Adapter

SLZB-06

### Setup

Add on on virtual machine on a QNAP nas.

CladStriffe, 2024-08-04:
Same issue here, but not related to the update of HA core. 
Instead on my side it is only due to update of Z2M to 1.39.0-1 (or 1.39.1-1 which I also tried before restoring to previous working version).
After the update of Z2M it looks OK, but when there is a reboot, all MQTT entities are duplicated with _2 at the end.

Koenkk, 2024-08-05:
Try removing the base topic from the z2m HA addon config page.

heisenberg2980, 2024-08-09:
Similar issue here. I have two z2m instances running in HA (each of them using a different base topic) and after a crash of HA all the devices linked to the second z2m instance appeared as unavailable and new entities with _2 were created.

@Koenkk removing the base topic in my second z2m instance is not an option in my case, as that would cause both z2m instances to write to the same topic, causing a lot of issues that I already experienced some months ago when I was setting up the second instance.

Koenkk, 2024-08-09:
@heisenberg2980 is the base topic on the addon config the same as the one in the z2m configuration.yaml?

heisenberg2980, 2024-08-09:
@Koenkk yes the base topic is zigbee2mqtt-garage in both places:

[image]

[image]

heisenberg2980, 2024-08-09:
@Koenkk I bit more info, let´s see if we find the reason (and a solution) for this. I have manually deleted the messages related to one of the devices from MQTT, and after deleting those messages the _2 entities disappeared from HA but the original entities (without _2) stayed. Opening one of these entities in HA, the message "This entity is no longer being provided by the mqtt integration. If the entity is no longer in use, delete it in settings." is displayed, so it seems somehow HA has unlinked those entities from MQTT, and when z2m try to register those entities again, HA generates de _2 entities to avoid duplicates.

The only solution seems to be to stop the z2m instance and manually delete each of the devices (which is a bit of a pain, specially if you have many devices in that z2m instance). After starting z2m the correct devices are created. Another side effect of this solution is that any info associated with the entity (like for instance the friendly name if it had been updated) is lost when the entity is deleted and restored.

Happy to hear any ideas of how to solve this issue with a more elegant solution, ideally not having to manually delete each entity and loosing some data in the process.

Koenkk, 2024-08-12:
@heisenberg2980 I'm not sure how the entities ended up being postfixed with `_2`, but since your base topic was the same on both the addon and configuration.yaml, I guess the HA crash caused this. Can't you easily bulk delete all the entities via HA?

[image]

Groeldor, 2024-08-12:
> @heisenberg2980 I'm not sure how the entities ended up being postfixed with `_2`, but since your base topic was the same on both the addon and configuration.yaml, I guess the HA crash caused this. Can't you easily bulk delete all the entities via HA?
> 
> <img alt="Screenshot 2024-08-12 at 13 11 37" width="1209" src="https://private-user-images.githubusercontent.com/2892853/357023538-a34863b0-bbe7-4fbf-8a73-4ceee1544f4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NjE2NzksIm5iZiI6MTcyMzQ2MTM3OSwicGF0aCI6Ii8yODkyODUzLzM1NzAyMzUzOC1hMzQ4NjNiMC1iYmU3LTRmYmYtOGE3My00Y2VlZTE1NDRmNGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTJUMTExNjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmNiZmFlYjNhZmY2YzE3NTQ3NDY0NTk0NmIwMTQ3MGExNmM2NjU1MDUxYTVmODJmZmIyNDE3Nzg3MjQ3NTlhMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.jZu6MGs8h0WhZmc6O-4Bi26UjNKfCBwV2d6OAyTM2_M">

I have some entities that I've not yet updated (some lights that I don't use as zigbee entites at the moment), so I'll give it a try today and let you know if this works.

CladStriffe, 2024-08-15:
> Try removing the base topic from the z2m HA addon config page.

Thank you for pointing me in the right direction. 

Before the update of Z2M, the base topic was indeed different between the addon configuration.yaml and the web front end settings. 
I don't know for how long it was the case. 
But after the update, the base topic in the web front end setting page was updated which caused the issue. 

So I restored my HA, updated the base topic in configuration.yaml to match the other one, and all was good even after updating Z2M. 
Thanks.

Groeldor, 2024-08-17:
I tried to remove some of the _2 entities, as a test, but got this message when I tried :
[image]

I stopped the z2m add on, but it didn't changed anything. Any clue on how to proceed ?

heisenberg2980, 2024-08-19:
@Groeldor after you stop the z2m add-on the _2 entities should disappear, and with the z2m stopped you can delete the entities without _2. After that you start z2m and it should create the entities without _2

Groeldor, 2024-08-20:
> @Groeldor after you stop the z2m add-on the _2 entities should disappear, and while z2m is stopped you can delete the entities without _2. After that, when you start z2m it should create the entities without _2

The _2 entites didn't disappeared after stoping z2m, neither did they "rename" automatically without _2.

I had to do it manually, like I did for the other one.

C47hub, 2024-09-25:
You can prevent this from happening ... I reverted to a previous snapshot and did it this way. If the _2 stuff already was created I couldn't get it fixed and didn't want to put in the work changing all this entities for 50+ devices.
So on a reverted snapshot I looked within zigbee2mqtt settings, not the addon, to the base topic. Then I stopped zigbee2mqtt, disabled autostart etc ... updated zigbee2mqtt. Restarted Home Assistant (do not skip this) and then on the ADDON settings, not within, I changed the base topic to what I had noted down before.  Starting zigbee2mqtt after that results in everything working correctly again. 
Needless to say, it took me a few attempts to figure this one out.

heisenberg2980, 2024-09-30:
@C47hub that is amazing, I will save your comment and test these steps if this happens again. @Groeldor you may want to close this issue as it seems the solution from @C47hub might solve the issue.

closed 2024-12-28 by github-actions[bot]