Koenkk/zigbee2mqtt #23457: Tuya TS011F_plug_1 display Null as sum of consumed energy

evilmumi, 2024-07-31:
### What happened?

The plug is not sending a value as energy:
[image]
[image]
i have tried to reset count via dev console - no change.
Also tried to play with calibration settings or something - no change.

Data:
`{"id":45,"type":"Router","ieeeAddr":"0xa4c1384b40863719","nwkAddr":54607,"manufId":4417,"manufName":"_TZ3000_cicwjqth","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"haElectricalMeasurement":{"attributes":{"activePower":0,"rmsVoltage":229,"rmsCurrent":0}},"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_cicwjqth","powerSource":1,"zclVersion":3,"appVersion":116,"stackVersion":0,"hwVersion":1,"swBuildId":""}},"genOnOff":{"attributes":{"32768":0,"moesStartUpOnOff":0,"tuyaBacklightMode":0,"onOff":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,423]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b4d0","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b4d0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b4d0","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":116,"stackVersion":0,"hwVersion":1,"swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722405323108}`

any more information / logs needed?

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

docker container

evilmumi, 2024-08-02:
still same in 1.39.1

Koenkk, 2024-09-12:
When pressing the configure (yellow refresh) button on the frontend, does this succeed?

evilmumi, 2024-09-13:
I get:
[image]

Koenkk, 2024-09-14:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/3eeb90dd20a3bf6f09e76880abfc9c74): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if configure succeeds now

evilmumi, 2024-09-19:
log:
`[2024-09-19 09:23:16] info: 	z2m: Loaded external converter 'ext_converter.js'
`
no custom in description:
[image]
`[2024-09-19 09:23:19] error: 	z2m: Failed to configure 'Tuya_Gartenpumpe', attempt 1 (Error: ZCL command 0xa4c1384b40863719/1 seMetering.configReport([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:648:9)
    at Object.currentSummDelivered (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:128:5)
    at configure (/app/data/extension/externally-loaded.js:61:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:193:21)
    at Configure.configure (/app/lib/extension/configure.ts:125:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
`

Koenkk, 2024-09-19:
`UNSUPPORTED_ATTRIBUTE` indicates it doesn't support reporting the `energy`, does it work when reading via the dev console? If yes, then we need to poll it

[image]

evilmumi, 2024-09-20:
if i do so i receive:
[image]

Koenkk, 2024-09-20:
`UNSUPPORTED_ATTRIBUTE` indicated this device doesn't support it. To confirm this you would have to check with the Tuya gateway. I think there is little we can do from the z2m side.

evilmumi, 2024-09-23:
hi , what excatly do you mean with Tuya gateway?

Koenkk, 2024-09-23:
I mean testing if it work with the original Tuya gateway

svmaris, 2024-09-23:
I have a couple of these power plugs (slightly different models, `_TZ3000_gjnozsaz` and `_TZ3000_w0qqde0g`) and noticed that they suddenly stopped reporting the total energy value a couple of days ago. Not sure if it's related, but I added a couple of these plugs and a couple of hours later they all stopped reporting the total energy.

Adding ext_converter.js didn't help. Only after downgrading the Docker image from `:dev-latest` to `:latest` (v1.40.1) they started reporting this value again and they've been working fine for the last couple of days.

evilmumi, 2024-09-24:
> I mean testing if it work with the original Tuya gateway

ah ok. I don't own that HW :/

Koenkk, 2024-09-24:
@svmaris thanks for reporting, latest-dev indeed had an issue, fix it now.

ganslmeier, 2024-09-28:
Same issue on my side with TS011F_plug_1. 
I figured out that it works well with the stable version. Dev shows NULL for energy.

ganslmeier, 2024-09-30:
Version: 1.40.1-dev not working. 
Version: 1.40.1 working. 

I have two lxc container running. If I switch to dev the energy is showing NULL. If I start the stable lxc the energy value is working.

briodan, 2024-10-02:
Seeing an issue with my [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3),
[image]
 seeing it via the HA addon on 1.40.0, 1.40.1 and 1.40.2

Koenkk, 2024-10-02:
@ganslmeier is 1.40.2 ok?

ganslmeier, 2024-10-02:
It is working. Thanks a lot for your great work. 
"version": "1.40.2-dev"

[image]

Koenkk, 2024-10-03:
@ others, can this be closed?

svmaris, 2024-10-03:
@Koenkk Just upgraded to 1.40.2-dev commit: [e5fac76](https://github.com/Koenkk/zigbee2mqtt/commit/e5fac76) and it's working fine. Thanks!