Koenkk/zigbee2mqtt #23458: Another tuya zigbee repeater _TZ3000_kxlmv9ag identified as TS0207_water_leak_detector

MartinNeurol, 2024-07-31:
### What happened?

New EMOS Smart zigbee repeater IP-2213Z - just a tuya repeater, _TZ3000_kxlmv9ag should be added to tuya.js among the other repeaters identified as water leak sensors:         

fingerprint: [{modelID: 'TS0207', manufacturerName: '_TZ3000_m0vaazab'},
            {modelID: 'TS0207', manufacturerName: '_TZ3000_ufttklsz'},
            {modelID: 'TS0207', manufacturerName: '_TZ3000_nkkl7uzv'},
            {modelID: 'TS0207', manufacturerName: '_TZ3000_misw04hq'},
            {modelID: 'TS0207', manufacturerName: '_TZ3000_gszjt2xx'},
            {modelID: 'TS0207', manufacturerName: '_TZ3000_5k5vh43t'}],
        model: 'TS0207_repeater',
        vendor: 'TuYa',
        description: 'Repeater',
        fromZigbee: [fz.linkquality_from_basic],
        toZigbee: [],
        exposes: [],

Many thanks!

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

 7.1.1.0 build 273

### Adapter

Skyconnect

### Setup

HomeAssistant OS on Pi

Koenkk, 2024-08-01:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-01 by Koenkk

MartinNeurol, 2024-08-04:
You are The Master, thank you very much!