Koenkk/zigbee2mqtt #23467: Lidl Switch with power meter model HG08673 only reports 0 current and 0 power

evilmumi, 2024-08-01:
### What happened?

Same as https://github.com/Koenkk/zigbee2mqtt/issues/20716
[image]
[image]
Current and power keep being zero.

Only the total used sum is calculated towards the energy property.

### What did you expect to happen?

I expected the power and current consumption to also be reported, as they are supposed to be supported if I look at the compatibility page.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

docker container

evilmumi, 2024-08-02:
still same in 1.39.1

EnterTheMaelstrom, 2024-08-02:
Hi, I had the same problem in 1.38.0. 
Completely deleting the device from z2m and repairing did the trick. Although I also updated to firmware 20240710, so you might try that too.

DeadWalkingDeath, 2024-08-04:
I've got the same problem.
Zigbee2MQTT version 1.39.1
The firmware is updated to the latest version 192.
Deleting the device and repairing does bring the reporting back, at least for a while. But after some time it will again fail to report the power.

DeadWalkingDeath, 2024-08-05:
Yes, I meant re-pair.
I don't know if its a coincidence though.
Yesterday the reporting started working again on it self and worked until this morning. But now I do get 0 values again for power and current.

benjaminburzan, 2024-08-07:
I'm experiencing the same issue with Zigbee2MQTT version 1.39.1. My firmware is updated to the latest version 192. Deleting and re-pairing the device didn't resolve the problem.

benjaminburzan, 2024-08-30:
Is there anything I can do to help fixing this issue? @Koenkk

Koenkk, 2024-08-31:
When attaching a load and reading the `activePower` from the dev console, does it return a non zero value? If it returns 0, I'm afraid the plug doesn't support it or is broken:

[image]

DeadWalkingDeath, 2024-09-01:
Yes, it does return non-zero values.

[image]

benjaminburzan, 2024-09-02:
Same here.
`Info 2024-09-02 13:32:20zhc:tz: Read result of 'haElectricalMeasurement': {"activePower":56}
`

evilmumi, 2024-09-02:
[image]
0w could not be because of running alexa echo whole time. I think the device is not the best for monitoring

Koenkk, 2024-09-03:
- Does the configure succeed? You can trigger it by pressing the yellow refresh icon on the frontend -> device page
- Could you provide a debug log when attaching a load to this device?

xdedek, 2024-09-04:
I have a same problem with Zigbee2MQTT version 1.40.0.
After last update Zigbee2MQTT show this:
[image]

Koenkk, 2024-09-04:
@xdedek interesting, could you try to factory reset the device and configure again?

xdedek, 2024-09-05:
> @xdedek interesting, could you try to factory reset the device and configure again?

Yes, after Last One factory reset : 
[image]
[image]

Koenkk, 2024-09-05:
@xdedek does it report power now? And did the configure succeed?

xdedek, 2024-09-12:
@Koenkk power now reporing, but not show "energy" (kWh) and current sometimes not show and sometimes show bad number (**180 - 400 A**) -- maybe is **m**A. Voltage is OK.

xdedek, 2024-12-11:
hello
after the last update of z2m to version 1.42.0 everything has moved forward considerably, the values ​​look much better, but I haven't had the opportunity to make an accurate measurement yet, however "power", "voltage" and now also "energy" seem to work perfectly fine, only the current value is a bit strange, but maybe that will be resolved or explained.
[image]

evilmumi, 2025-02-11:
so it is HW related?

closed 2025-04-21 by github-actions[bot]