Koenkk/zigbee2mqtt #23469: [New device support]:  TS0601_TZE200_w6n8jeuu Magic Smart Temperature and Humidity sensor with Backlight Display

Rust-Shch, 2024-08-01:
### Link

https://www.aliexpress.com/item/1005006345091426.html?spm=a2g0o.productlist.main.51.145d8MQT8MQTjY&algo_pvid=5bb60871-caeb-45bc-a7b3-6b65b4514725&algo_exp_id=5bb60871-caeb-45bc-a7b3-6b65b4514725-25&pdp_npi=4%40dis%21EUR%2119.12%2117.92%21%21%2120.22%2118.95%21%402103856417225285931415775e36fa%2112000037978600160%21sea%21LV%210%21ABX&curPageLogUid=Z1zNwCoEQY30&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":72,"type":"EndDevice","ieeeAddr":"0xa4c138a4accbb346","nwkAddr":54885,"manufId":4417,"manufName":"_TZE200_w6n8jeuu","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001d�>.\u0019\u001d�>.\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722528565453}

### Comments

Please help me to use this device.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_w6n8jeuu',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Rust-Shch, 2024-08-01:
I have very similar device, but with battery power - it is supported from the box. Also they have very similar DB entries at begining, but manufacturer codes are different:

{"id":72,"type":"EndDevice","ieeeAddr":"0xa4c138a4accbb346","nwkAddr":54885,"manufId":4417,"manufName":"_TZE200_w6n8jeuu","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001d�>.\u0019\u001d�>.\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722528565453}

{"id":73,"type":"EndDevice","ieeeAddr":"0xa4c138038ba05766","nwkAddr":60999,"manufId":4417,"manufName":"_TZE200_vvmbj46n","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i\u0018\u0000\u0000\u0000i","65506":56,"65508":1,"65534":0,"appVersion":72,"modelId":"TS0601","manufacturerName":"_TZE200_vvmbj46n","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1013852590},"lastSeen":1722527854752}

Koenkk, 2024-08-03:
What you can try is to stop z2m, change the `manufName` of the unsupported one to the supported one, start z2m and see if it works now.

closed 2024-12-28 by github-actions[bot]

niw87, 2025-01-20:
Hi there, just got the same device. Any possibility of adding this device? Thanks.

> What you can try is to stop z2m, change the `manufName` of the unsupported one to the supported one, start z2m and see if it works now.

latez, 2025-02-28:
Same here, just got the same device - would be great to add it to the database - @Koenkk  will try making the modifications today that you suggested to see if they work.

latez, 2025-02-28:
I can confirm this worked! - 

[image]

[image]

Koenkk, 2025-03-01:
Thanks, added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.