Koenkk/zigbee2mqtt #23476: [Wrong device]: _TZE204_myd45weu recognized as TS0601_soil not QT-07S, and soil_moisture is only 0% or 100%, nothing between

MilesTEG1, 2024-08-02:
### Link

https://fr.aliexpress.com/item/1005005210965804.html

### Model

QT-07S

### Description

Soil sensor: humidity, temperature, illuminance

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/QT-07S.png

### Database entry

{"id":67,"type":"EndDevice","ieeeAddr":"0xa4c1385abf571524","nwkAddr":997,"manufId":4417,"manufName":"_TZE204_myd45weu","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_myd45weu","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1722579663746}

### Notes

I use the latest Z2M 1.39.1-1, HAOS Addon.

I try to remove batteries for more than 5 minutes, force suppression, and repair. But it still recognized as TS0601_soil, not QT-07S.
And humidity is only these two values: 0% or 100%. Nothing between.

I think the humidity problem should be solved by having the right device recognition: QT-07S

dfsx1, 2024-08-03:
I returned my `QT-07S` for refund 2 days ago since I thought the device was faulty.
I regret now as this is probably not sellers fault but Z2M issue.
2 commits are merged recently:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7593/commits/c3092bd180897e122c99154f24e7ea5636eccd19

Adding `_TZE204_myd45weu` as fingerprint for `TS0601_soil`.
This commit was probably not properly tested by its author @micwiel, since we get wrong data (always 100% moisture).

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7811/commits/b7a2ae24b15ff641f54f17890efb4e4de50cea47

Adding `_TZE204_myd45weu` as whitelabel for `TS0222`/`QT-07S`.
This commit does not delete the `_TZE204_myd45weu` fingerprint from the above commit for `TS0601_soil`.
My QT-07S didnt have a illuminance sensor so I don't know if it should be added here in first place.
Also confusing to mix-up humidity with soil_moisture for both devices.

dfsx1, 2024-08-03:
I returned my `QT-07S` for refund 2 days ago since I thought the device was faulty.
I regret now as this is probably not sellers fault but Z2M issue.
2 commits are merged recently:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7593

Adding `_TZE204_myd45weu` as fingerprint for `TS0601_soil`.
This commit was probably not properly tested by its author @micwiel, since we get wrong data (always 100% moisture).

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7811

Adding `_TZE204_myd45weu` as whitelabel for `TS0222`/`QT-07S`.
This commit does not delete the `_TZE204_myd45weu` fingerprint from the above commit for `TS0601_soil`.
My QT-07S didnt have a illuminance sensor so I don't know if it should be added here in first place.
Also confusing to mix-up humidity with soil_moisture for both devices.

MilesTEG1, 2024-08-03:
I don't really know if this is really a faulty device...
I'll have to return mine too.

dfsx1, 2024-08-03:
Try to remove the `_TZE204_myd45weu` fingerprint from:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7529e1f0db30e1d87dbd13113f1dc997eaff65ac/src/devices/tuya.ts#L1929

And add it to:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7529e1f0db30e1d87dbd13113f1dc997eaff65ac/src/devices/tuya.ts#L6295

dfsx1, 2024-08-03:
Try to remove `_TZE204_myd45weu` fingerprint from:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7529e1f0db30e1d87dbd13113f1dc997eaff65ac/src/devices/tuya.ts#L1929

Add it to:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7529e1f0db30e1d87dbd13113f1dc997eaff65ac/src/devices/tuya.ts#L6295

Re-pair the device.

micwiel, 2024-08-03:
After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

dfsx1, 2024-08-03:
> After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

Thanks for the clarification.
I must admit that I got correct moisture readings around 20-30% on first setup on 1.39.0.
But after watering it went up to 100% and stayed at 100% for almost a week (only 90% once after a re-pair).

I remember I switched Z2M from 1.39.0 to 1.93.0-dev that same day, which could have also caused the issue.
Do you still get correct moisture with latest version 1.39.1?

I had a Xiaomi MiFlora sensor in the same soil for comparison.
Temperature readings were also not accurate, so I already returned my device for a refund.
[image]

dfsx1, 2024-08-03:
> After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

@micwiel thanks for the clarification.
I must admit that I got correct moisture around 20-30% on first setup on 1.39.0.
But after watering it went up to 100% and stayed at 100% for almost a week (only 90% once after a re-pair).
I remember I switched Z2M from 1.39.0 to 1.93.0-dev that same day, which could have also caused the issue.
Do you still get correct moisture with latest version 1.39.1?

I had a Xiaomi MiFlora sensor in the same soil for comparison.
Temperature readings were also not accurate, so I already returned my device for a refund.
[image]

dfsx1, 2024-08-03:
> After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

@micwiel thanks for the clarification.
I must admit that I got correct moisture around 20-30% on first setup on 1.39.0.
But after watering it went up to 100% and stayed at 100% for almost a week (only 90% once after a re-pair).
I remember I switched Z2M from 1.39.0 to 1.93.0-dev that same day, which could have also caused the issue.
Do you still get correct moisture with latest version 1.39.1?

I had a Xiaomi MiFlora sensor in the same soil for comparison.
Temperature readings were also not very accurate, so I already returned my device for a refund.
[image]

dfsx1, 2024-08-03:
> After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

@micwiel thanks for the clarification.
I must admit that I got correct moisture around 20-30% on first setup on 1.39.0.
But after watering it went up to 100% and stayed at 100% for almost a week (only 90% once after a re-pair).
I remember I switched Z2M from 1.39.0 to 1.39.0-dev that same day, which could have also caused the issue.
Do you still get correct moisture with latest version 1.39.1?

I had a Xiaomi MiFlora sensor in the same soil for comparison.
Temperature readings were also not very accurate, so I already returned my device for a refund.
[image]

dfsx1, 2024-08-03:
> After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

@micwiel thanks for the clarification.
I must admit that I got correct moisture around 20-30% on first setup on 1.39.0.
I switched Z2M from 1.39.0 to 1.39.0-dev that same day, which could have also caused the issue.
After watering it went up to 100% and stayed at 100% for almost a week (only 90% once after a re-pair).
Do you still get correct moisture with latest version 1.39.1?

I had a Xiaomi MiFlora sensor in the same soil for comparison.
Temperature readings were also not very accurate, so I already returned my device for a refund.
[image]

dfsx1, 2024-08-03:
> After merge I didn't observe issues with my devices. All parameters seem to be properly exposed and interpreted. I see various values for moisture, not just 100%. But if you are sure it should be recognized as different model I'm happy to perform more tests.

@micwiel thanks for the clarification.
I must admit that I got correct moisture around 20-30% on first setup on 1.39.0.
I switched Z2M to 1.39.0-dev that same day, which could have also caused the issue.
After watering it went up to 100% and stayed at 100% for almost a week (only 90% once after a re-pair).
Do you still get correct moisture with latest version 1.39.1?

I had a Xiaomi MiFlora sensor in the same soil for comparison.
Temperature readings were also not very accurate, so I already returned my device for a refund.
[image]

micwiel, 2024-08-06:
@dfsx1 I've upgraded z2m to 1.39.1 yesterday and devices still report various values however I don't have any other device this type from different vendor to compare if they are really accurate. I'm monitoring changes in moisture only so particular levels are not important for me, they seem to do this job well enough. An example of moisture statistics from my 2 devices over last week in the attached screenshot.

[image]

bHilz, 2024-09-02:
I'm having the same issue. Didn't work for me with 1.39.1 and also not with 1.40.0.  Anyone here who could fix this? It seems that the QT-07S does not work with z2m at all but with zha it does. I think a lot of people are having this problem.

closed 2024-12-28 by github-actions[bot]

10der, 2025-05-30:
1) always offline
2) strange temperature
[image]

[image]