Koenkk/zigbee2mqtt #23478: Tuya ZG-205Z presence sensor - no luminance entity

Ramotny, 2024-08-02:
### What happened?

In "exposed" list there is a "luminance" (numeric) but there is no access (entity) to it. Worked fine in native ZHA/Zigbee enviroinment.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

0122052017

### Adapter

Sonoff Dongle Plus P

### Setup

Lenovo MiniPC

Mikl-h, 2024-08-02:
You have to enable Illuminance entity in device settings:
[image]

uniquePWD, 2024-08-02:
Illuminance broke with the latest update for me too. It's stuck at 29lx

Zigbee Model: TS0225
Device Model: [ZG-205Z/A](https://www.zigbee2mqtt.io/devices/ZG-205Z_A.html#tuya-zg-205z%252Fa)

VoltX, 2024-08-02:
> Illuminance broke with the latest update for me too. It's stuck at 29lx
> 
> Zigbee Model: TS0225 Device Model: [ZG-205Z/A](https://www.zigbee2mqtt.io/devices/ZG-205Z_A.html#tuya-zg-205z%252Fa)

Same problem for me, on HA you can setup specific parameters and fix multiplier
[image]

uniquePWD, 2024-08-02:
> Same problem for me, on HA you can setup specific parameters and fix multiplier [...]

Thanks, the 10000 multiplier fixed it. It doesn't seem as responsive now, but I'll test for a bit and find out for sure.

closed 2024-08-02 by Ramotny