Koenkk/zigbee2mqtt #23483: Invalide state tuya shutter since v1.39.00

ElCaplar, 2024-08-02:
Hello everyone, 

Following an update to Zigbee2MQTT 1.39.00 , I have noticed strange behavior with some of my roller shutters (specifically those with modules recognized as TUYA-TS130F. I have 2 other MOES modules [MS-108ZR] that are working correctly).

When I fully raise the shutter, the status visible in Home Assistant is "closed" even though the shutter is open. Below is the trace from the development menu:

```
state : closed
current_position: 100
backlight_mode: null
calibration: "OFF"
calibration_time: 32
indicator_mode: "off"
linkquality: 255
motor_reversal: "OFF"
moving: STOP
position: 100
friendly_name: volet-salon
supported_features: 15
```

On zigbeemqtt interface here is : 

`{
    "calibration": "OFF",
    "calibration_time": 32,
    "indicator_mode": "off",
    "linkquality": 255,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 100,
    "state": "OPEN"
}`

The shutter appears closed while it is actually open. When I click on "STOP," the status updates correctly

I am experiencing exactly the same strange behavior when I close the shutter, as it appears to be in open mode. When I use intermediate positions, there is no issue.

When i downgrade on 1.38.00 version, all is ok.

Do you have any possible solutions or workarounds for this behavior?

_Originally posted by @ElCaplar in https://github.com/Koenkk/zigbee2mqtt/discussions/23480_

matwojcik, 2024-08-03:
Similar issue was described here: https://github.com/Koenkk/zigbee2mqtt/issues/23277 but there they mention HA as a buggy, once you say that downgrade of Z2M helps.

ElCaplar, 2024-08-16:
Update for this problem. Since the last version i have updated : 
Core 2024.8.1
Supervisor 2024.08.0
Operating System 13.0
Interface utilisateur 20240809.0
Version de Zigbee2MQTT [1.39.1]

Everything seems to be ok.

platte-73, 2024-08-19:
I have the same problem after update z2m

eerguen, 2024-10-02:
I have the same problem with two Moes TS130F (_TZ3000_1dd0d5yi) devices. Shutters are open - HA shows closed and vice versa. Connection correct - hardware buttons work. Two are partially not working and two without any issues. Sometimes I have to press the switch three times for the shutter to close / open completely. It behaves very strangely and is annoying. 

`{
    "calibration": "ON",
    "calibration_time": 19,
    "linkquality": 69,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 0,
    "state": "CLOSE",
    "update": {
        "installed_version": 71,
        "latest_version": 71,
        "state": "idle"
    }
}`

Koenkk, 2024-10-03:
Are they open or closed with `"position": 0, "state": "CLOSE"`?

eerguen, 2024-10-04:
> Are they open or closed with `"position": 0, "state": "CLOSE"`?

In this state, the roller shutter is completely open. In the “open” state, the roller shutter is closed. However, the switches function correctly.

Koenkk, 2024-10-05:
@eerguen fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Suschibomber, 2024-11-01:
Hey... After the newest update my cover status is shown incorrectly :( before the update my moes roller shutters works perfectly fine.

crazyserver, 2024-11-01:
Same happen to me, after I updated today's version.

Model
    TS130F
Manufacturer
    _TZ3000_1dd0d5yi

rogervdh, 2024-11-02:
I have the same issue after updating to Z2M 1.41.0. It was always working perfectly with the Moes MS-108ZR and Z2M 1.39.0 (and all earlier versions) and since the update to 1.41.0 the cover status is incorrect. I seems that only the MS-108ZR shutter switches are affected. Other TS130F shutter switches in my setup are working correctly (also after the update to 1.41.0).

[image]

aqwserf, 2024-11-02:
Well, same issue on my side with 4 MS-108ZR. Back to normal when reverting back to 1.40.2-1

afharo, 2024-11-03:
Same here for `_TZ3000_1dd0d5yi`. They used to work as expected. 

After the update, I had to enable the invert mode for them work as expected again. However, now my physical controls are inverted and I will need to rewire 😞.

afharo, 2024-11-03:
> Other TS130F shutter switches in my setup are still working correctly (also after the update to 1.41.0).

@rogervdh, can you share the "Zigbee manufacturer" for the shutters that don't work and the ones that work?

I have the suspicion that `_TZ3000_1dd0d5yi` are broken with the update but others might be ok.

rogervdh, 2024-11-03:
> @rogervdh, can you share the "Zigbee manufacturer" for the shutters that don't work and the ones that work?
>
> I have the suspicion that `_TZ3000_1dd0d5yi` are broken with the update but others might [...]

@afharo,
TS130F Moes switch which doesn't work correctly since 1.41.0: 

**Zigbee Model** TS130F
**Zigbee Manufacturer** _TZ3000_1dd0d5yi
**Description**  Zigbee + RF curtain switch module

Other TS130F switches which are still working correctly:

**Zigbee Model**  TS130F
**Zigbee Manufacturer**  _TZ3000_ltiqubue
**Description**  Curtain/blind switch

**Zigbee Model**  TS130F
**Zigbee Manufacturer**  _TZ3000_qqdbccb3
**Description**  Curtain/blind switch

Koenkk, 2024-11-03:
Can you provide the data/database.db entry of `_TZ3000_1dd0d5yi` and indicated wether the state is inverted with 1.41.0 or not?

aqwserf, 2024-11-04:
@Koenkk FWIW, with my `_TZ3000_1dd0d5yi` fully open on 1.41.0:
```
$ cat database.db | jq 'select(.ieeeAddr == "0x540f57fffe896eab")'
{
  "id": 28,
  "type": "Router",
  "ieeeAddr": "0x540f57fffe896eab",
  "nwkAddr": 29337,
  "manufId": 4098,
  "manufName": "_TZ3000_1dd0d5yi",
  "powerSource": "Mains (single phase)",
  "modelId": "TS130F",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 515,
      "inClusterList": [
        0,
        4,
        5,
        258,
        6
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "65503": "�\u0010�.\u0016",
            "65506": 31,
            "65508": 0,
            "modelId": "TS130F",
            "manufacturerName": "_TZ3000_1dd0d5yi",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 71,
            "stackVersion": 0,
            "hwVersion": 1,
            "dateCode": ""
          }
        },
        "closuresWindowCovering": {
          "attributes": {
            "moesCalibrationTime": 280,
            "tuyaMovingState": 1,
            "currentPositionLiftPercentage": 100,
            "tuyaMotorReversal": 0,
            "tuyaCalibration": 0
          }
        },
        "genOnOff": {
          "attributes": {
            "20480": 0,
            "tuyaBacklightSwitch": 0
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 71,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1730708760400
}

$ cat state.json | jq '."0x540f57fffe896eab"'
{
  "calibration_time": 28,
  "moving": "STOP",
  "position": 100,
  "motor_reversal": "OFF",
  "linkquality": 255,
  "state": "OPEN",
  "update": {
    "state": "idle",
    "installed_version": 71,
    "latest_version": 71
  },
  "update_available": false,
  "calibration": "ON"
}
$
```

Reverting back to 1.40.2-1 (without moving the cover manually or through HA). Show as `open` in HA:
```
$ cat database.db | jq 'select(.ieeeAddr == "0x540f57fffe896eab")' | diff - 1410_db_open
73c73
<   "lastSeen": 1730709188731
---
>   "lastSeen": 1730708760400

$ cat state.json | jq '."0x540f57fffe896eab"' | diff - 1410_state_open 
4c4
<   "position": 0,
---
>   "position": 100,
7c7
<   "state": "CLOSE",
---
>   "state": "OPEN",
```

After pressing `open` in Z2MQTT (did nothing as the cover was already opened):
```
$ cat state.json | jq '."0x540f57fffe896eab"' | diff - 1410_state_open
$
```

Murada99, 2024-11-04:
Same for me, after the update: 
When it's closed, I get the following states:
{
    "calibration": "ON",
    "calibration_time": 47,
    "last_seen": "2024-11-04T16:13:41.251Z",
    "linkquality": 43,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 100,
    "state": "OPEN"
}
I had to enable the invert mode for them work as expected again but I need to change all my automations.

0rni1, 2024-11-04:
Hello, I’m also having the same issue with the 1.41.0-1 update. The cover’s state appears inverted; when it’s closed, it shows as open. The open and close buttons work correctly.
I have the Moes MS-108ZR model. I have returned to the previous version of zigbee 2 Mqtt and everything is correct.

Murada99, 2024-11-05:
Seems to be the same issue as this one #23277

devilvan33, 2024-11-06:
Hello, in the latest version 1.41 the MOES MS-108ZR device does not work correctly. When it is closed it appears open and vice versa. I have gone back to the previous version and it works fine again.

Koenkk, 2024-11-07:
I think it was fixed in https://github.com/Koenkk/zigbee2mqtt/issues/23483 already, can you check if everything is ok with the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

afharo, 2024-11-07:
I guess you meant #24609, updating the converters with this changelog: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8259 🧡

theyort, 2024-11-10:
The malfunction is still present. When the shutter finishes opening, it appears as closed, and conversely, when the shutter closes, it then appears as open. I reverted to version 1.39.1 to restore normal functionality.

Koenkk, 2024-11-10:
@theyort did you also check the[dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

afharo, 2024-11-11:
> I think it was fixed in #23483 already, can you check if everything is ok with the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

I just tested the docker tag `latest-dev` and it works for me as intended. I only own `_TZ3000_1dd0d5yi`, though. So maybe owners of the other models can confirm that it still works for them in those ones.

Suschibomber, 2024-11-18:
@Koenkk Will it be fixed again in the next update? I would like to update.

UsernetFR, 2024-11-18:
After the 1.41.0 update, this morning my shutters were all reversed. It was clearly mentioned in the changelog (_#23483 Fix Moes MS-108ZR cover inverted_). Question: will there be further changes again?

afharo, 2024-11-18:
@Suschibomber, @UsernetFR, as per [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/23483#issuecomment-2463063247), it should be fixed in the dev branch. 

I tested it on my end and it is fixed it for me. Do you mind testing as well on your end to confirm?

These steps should help you switch to the dev version: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

RealSpyKitty, 2024-11-25:
Updated yesterday from v1.40.2-1 to  1.41.0-1, the controller at 100% meant its open and 0% for completely closed now its completely flipped and it messed up my automations and how I am used to controlling it.

I could flip the motor direction but then I will need to open every single controller to flip the wires.

I am glad I have a backup and will revert to v1.40.2-1 until its fixed.

Model: MS-108ZR
Manufacturer: Moes
Zigbee Model: TS130F
Zigbee Manufacturer: _TZ3000_1dd0d5yi

afharo, 2024-11-25:
@RealSpyKitty, can you try with the dev build: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html?

The version in dev works for me and we're waiting for more people to confirm that it fixed it on their end.

isibizi, 2024-12-08:
I am using the latest build (1.42), but the issue still persists. I switched from ZHA to Z2M yesterday. With ZHA, everything worked fine. Additionally, I am missing the option to change the calibration time for opening and closing. ZHA provided this functionality.

Zigbee Model: TS130F
Zigbee Manufacturer: _TZ3000_5iixzdo7

l1q1d, 2024-12-14:
Hi still see problem for 1.42.
With this version when I click open/close nothing append, if I use the radio remote it change the state on the web interface but still no control.

Zigbee Model:
TS130F
Zigbee Manufacturer: 
_TZ3000_1dd0d5yi

afharo, 2024-12-14:
I'd like to confirm that 1.42 has been working correctly for me since the release. Thanks @Koenkk!

l1q1d, 2024-12-15:
@afharo so with 1.42  you are able to control from web interface and you didn't need to rewire to make the remote work?
Have you done something special or just update? Did you need to revert?

afharo, 2024-12-15:
@l1q1d, this was my state: https://github.com/Koenkk/zigbee2mqtt/issues/23483#issuecomment-2453327294

- <= 1.40 - All worked as expected (I guess I wired everything based on that behavior)
- 1.41 inverted the controls, so physical buttons still worked according to my wiring, but the "smart" controls were inverted. Using the "invert" option fixed the smart controls but broke the wiring
- 1.42 reverted the behavior to 1.40 for me. All good on my end.

Suschibomber, 2024-12-19:
Will the Moes MS-108ZR now work again with the latest Zigbee version as before?

l1q1d, 2024-12-21:
Hi  @afharo I've used a backup of the docker config directory used with 1.40 and now works, so @Suschibomber if you move  directly from 1.40 to latest you should have a working zigbee2mqtt setup.
No rewire, remote and automations work.

Suschibomber, 2024-12-21:
@l1q1d Thank you for your answer. Will try it out tomorrow 👌🏼

l1q1d, 2024-12-27:
new update, after some time they don't move anymore, getting back to 1.40 and now I keep this version for sometime to verify the stability.

aqwserf, 2024-12-31:
Using Moes MS-108ZR `_TZ3000_1dd0d5yi`, I can confirm that on my side, everything is back to normal on `1.42.0-2`. Thanks @Koenkk !

Suschibomber, 2025-01-01:
For me, everything works normally again. Thank you @Koenkk

closed 2025-01-01 by Koenkk

l1q1d, 2025-01-01:
Hi All,
sotto to jump in again, I today I switch back to 1.42 and after 6 hours blinds stopped working. I went to the web interface and try to move on of them but I see no errors. I switch back to 1.40 and worked.
How can I do more debugging?

Kind regards