Koenkk/zigbee2mqtt #23485: TS0601 Производитель Zigbee _TZE204_fhvdgeuhonst    Умный автоматический выключатель Tuya

ia1gerebec, 2024-08-02:
### Link

https://aliexpress.ru/item/1005005984673866.html?spm=a2g2w.orderdetail.0.0.42154aa6JzNYVB&sku_id=12000038280422974

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0xa4c1382762a38dd1","nwkAddr":58462,"manufId":4417,"manufName":"_TZE204_fhvdgeuh","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"&�?.i��?.i��?.i��?.i��?.i��?.i��?.i","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_fhvdgeuh","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722618429404}

### Comments

Устройство  -Умный автоматический выключатель Tuya
https://aliexpress.ru/item/1005005984673866.html?spm=a2g2w.orderdetail.0.0.42154aa6JzNYVB&sku_id=12000038280422974

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_fhvdgeuh',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

SBozhyi, 2024-08-03:
I'm waiting for support for this device too.
There is Custom quirk for this device available for ZHA but I can't convert it to Z2M (lack of skills).
But may be it will help for someone skilled
[_TZE204_fhvdgeuh](https://github.com/zigpy/zha-device-handlers/issues/3140)

peekayn84, 2024-08-03:
Datapoint from tuya cloud:
{"1":"Total forward energy","6":"Phase A","9":"Fault","11":"Switch prepayment","12":"Clear energy","13":"Balance energy (kWh)","14":"charge energy","15":"Leakage current","16":"Switch","17":"Alarm set1","18":"Alarm set2","101":"合闸恢复时间（弃用）","102":"reclosing allowed times","103":"Temp Current","104":"reclosing enable","105":"Timer","106":"Cycle Schedule","107":"reclose recover seconds","108":"Random Timing","109":"Switch Inching","119":"Power-on delay power-on time","124":"Overcurrent Event Threshold Time","125":"Time Threshold of Lost Flow Event","126":"alarm set3","127":"status","134":"relay status for power on"}

francisPiragua, 2024-08-05:
Another one waiting too. Thanks.

peekayn84, 2024-08-05:
I create this one, but how parse this data i don't know, can someone help, please:
Alarm set1 {"0":4,"1":1,"2":0,"3":50,"4":5,"5":1,"6":0,"7":80}
Alarm set2 {"0":1,"1":1,"2":2,"3":118,"4":3,"5":1,"6":0,"7":252,"8":4,"9":1,"10":0,"11":210}
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tz_settings = {
    key: ['reclose_recover_seconds','clear_energy'],
    convertSet: async (entity, key, value, meta) => {
        const settingType = {'trip': 2, 'alarm': 1, 'closed': 0};
        // const typeswitch = {'ON':true, 'OFF':false}
        let dp_key = -1; // value if not found
        // identify dp.
        const tuyadatapoints_list = meta.mapped.meta.tuyaDatapoints
        for (let i = 0; i < tuyadatapoints_list.length; i++) {
            if (tuyadatapoints_list[i][1] === key) {
                dp_key = tuyadatapoints_list[i][0];
                break;
            }
        }
        switch (key) {
        case 'reclose_recover_seconds': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'clear_energy': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        default: // Unknown key
            meta.logger.warn(`Unhandled key ${key}`);
        }
    },
}

const customTrueFalseConverter = {
    key: ['state'],
    convertSet: (entity, key, value, meta) => {
      // Your conversion logic
      const presence = value === 1 ? true : false;
      return {state: {presence}};
    },
    convertGet: (entity, key, meta) => {
      // Convert the state value from Home Assistant to the appropriate Zigbee value
      const zigbeeValue = meta.state.presence ? 1 : 0;
      return {state: zigbeeValue};
    },
  };

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_fhvdgeuh',
        },
    ],
    model: 'TZE204_fhvdgeuh',
    vendor: 'Tuya',
    description: 'Tuya multi-function energy monitor',
    fromZigbee: [
        tuya.fz.datapoints,
        // Дополнительные преобразователи, если необходимо
    ],
    toZigbee: [tuya.tz.datapoints,tz_settings],
    onEvent: tuya.onEvent({respondToMcuVersionResponse:false, queryIntervalSeconds: 3 * 60}),
    configure: tuya.configureMagicPacket,
[... 41 lines cut ...]
            [16, 'state', tuya.valueConverter.onOff],
            [17, 'alarm_set1', tuya.valueConverter.raw],
            [18, 'alarm_set2', tuya.valueConverter.raw],
            [101, 'restore_time', tuya.valueConverter.raw], // Deprecated
            [102, 'reclosing_allowed_times', tuya.valueConverter.raw],
            [103, 'temperature', tuya.valueConverter.raw],
            [104, 'reclosing_enable', tuya.valueConverter.onOff],
            [105, 'timer', tuya.valueConverter.raw],
            [106, 'cycle_schedule', tuya.valueConverter.raw],
            [107, 'reclose_recover_seconds', tuya.valueConverter.raw],
            [108, 'random_timing', tuya.valueConverter.raw],
            [109, 'switch_inching', tuya.valueConverter.raw],
            [119, 'power_on_delay', tuya.valueConverter.raw],
            [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw],
            [125, 'time_threshold_lost_flow', tuya.valueConverter.raw],
            [126, 'alarm_set3', tuya.valueConverter.raw],
            //[127, 'status', tuya.valueConverter.raw],
            [127, 'status', tuya.valueConverter.onOff],
            [134, 'relay_status_power_on', tuya.valueConverter.raw],
            //[134, 'relay_status_power_on', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;`

peekayn84, 2024-08-05:
I find already supported _TZ3000_cayepv1a, and if i right understand it's same, but how connect id to this device i don't khow

peekayn84, 2024-08-05:
@Koenkk , hi can i ask you to add _TZE204_fhvdgeuh to already realized _TZ3000_cayepv1a or i can do it by myself?

Koenkk, 2024-08-05:
If you stop z2m, in the `data/database.db` change `_TZ3000_cayepv1a` to `_TZE204_fhvdgeuh` and start z2m, does it work?

peekayn84, 2024-08-06:
> _TZE204_fhvdgeuh
Unfortunately no, but I almost got a normal converter, but no data is sent, please can you tell me why
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

    const fzLocal = {
        value: {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                    payload = {};
                    for (const dpValue of msg.data.dpValues) {
                            const dp = dpValue.dp;
                            switch (dp) {
                                case 18:
                                    {
                                        payload['under_voltage'] = legacy.getDataValue(dpValue)[11];
                                        payload['over_voltage'] = legacy.getDataValue(dpValue)[7];
                                        return payload;
                                    };
                            };
                    };
            },
        },
    };

const tz_settings = {
    key: ['reclose_recover_seconds','clear_energy','under_voltage','over_voltage'],
    convertSet: async (entity, key, value, meta) => {
        const settingType = {'trip': 2, 'alarm': 1, 'closed': 0};
        // const typeswitch = {'ON':true, 'OFF':false}
        let dp_key = -1; // value if not found
        // identify dp.
        const tuyadatapoints_list = meta.mapped.meta.tuyaDatapoints
        for (let i = 0; i < tuyadatapoints_list.length; i++) {
            if (tuyadatapoints_list[i][1] === key) {
                dp_key = tuyadatapoints_list[i][0];
                break;
            }
        }
        switch (key) {
        case 'reclose_recover_seconds': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'clear_energy': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'over_voltage': {
            //Data not updated
            const value_send = meta.state['alarm2'];
            value_send['7']=value;
            await tuya.sendDataPointValue(entity, 18, value_send);
            break;
        }
        case 'under_voltage': {
            //Data not updated
            const value_send = meta.state['alarm2'];
            value_send['11']=value;
            await tuya.sendDataPointValue(entity, 18, value_send);
            break;
        }
        default: // Unknown key
[... 82 lines cut ...]
            [17, 'alarm_set1', tuya.valueConverter.raw],
            [18, 'alarm2', tuya.valueConverter.raw],
            [101, 'restore_time', tuya.valueConverter.raw], // Deprecated
            [102, 'reclosing_allowed_times', tuya.valueConverter.raw],
            [103, 'temperature', tuya.valueConverter.raw],
            [104, 'reclosing_enable', tuya.valueConverter.onOff],
            [105, 'timer', tuya.valueConverter.raw],
            [106, 'cycle_schedule', tuya.valueConverter.raw],
            [107, 'reclose_recover_seconds', tuya.valueConverter.raw],
            [108, 'random_timing', tuya.valueConverter.raw],
            [109, 'switch_inching', tuya.valueConverter.raw],
            [119, 'power_on_delay', tuya.valueConverter.raw],
            [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw],
            [125, 'time_threshold_lost_flow', tuya.valueConverter.raw],
            [126, 'alarm_set3', tuya.valueConverter.raw],
            [127, 'status', tuya.valueConverter.raw],
            //[127, 'status', customTrueFalseConverter],
            [134, 'relay_status_power_on', tuya.valueConverter.raw],
            //[134, 'relay_status_power_on', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

peekayn84, 2024-08-06:
I created something like this, not all function add, but almost needed
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

    const fzLocal = {
        value: {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                    payload = {};
                    for (const dpValue of msg.data.dpValues) {
                            const dp = dpValue.dp;
                            switch (dp) {
                                case 17:
                                    {
                                        payload['over_leakage_current'] = legacy.getDataValue(dpValue)[3];
                                        payload['over_temperature'] = legacy.getDataValue(dpValue)[7];
                                        return payload;
                                    };
                                case 18:
                                    {
                                        payload['under_voltage'] = legacy.getDataValue(dpValue)[11];
                                        payload['over_voltage'] = legacy.getDataValue(dpValue)[7];
                                        overcurrent=legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                                        payload['over_current']=overcurrent/10.0;
                                        return payload;
                                    };
                            };
                    };
            },
        },
    };

const tz_settings = {
    key: ['reclose_recover_seconds','clear_energy','under_voltage','over_voltage','reclosing_allowed_times','over_leakage_current','over_temperature','over_current'],
    convertSet: async (entity, key, value, meta) => {
        const settingType = {'trip': 2, 'alarm': 1, 'closed': 0};
        // const typeswitch = {'ON':true, 'OFF':false}
        let dp_key = -1; // value if not found
        // identify dp.
        const tuyadatapoints_list = meta.mapped.meta.tuyaDatapoints
        for (let i = 0; i < tuyadatapoints_list.length; i++) {
            if (tuyadatapoints_list[i][1] === key) {
                dp_key = tuyadatapoints_list[i][0];
                break;
            }
        }
        switch (key) {
        case 'reclose_recover_seconds': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'reclosing_allowed_times': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'clear_energy': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'over_leakage_current': {
            const value_send = meta.state['alarm1'];
            value_send['3']=value;
            const payload = [value_send['0'],value_send['1'],value_send['2'],value_send['3'],value_send['4'],value_send['5'],value_send['6'],value_send['7']];
[... 119 lines cut ...]
            [17, 'alarm1', tuya.valueConverter.raw],
            [18, 'alarm2', tuya.valueConverter.raw],
            [101, 'restore_time', tuya.valueConverter.raw], // Deprecated
            [102, 'reclosing_allowed_times', tuya.valueConverter.raw],
            [103, 'temperature', tuya.valueConverter.raw],
            [104, 'reclosing_enable', tuya.valueConverter.onOff],
            [105, 'timer', tuya.valueConverter.raw],
            [106, 'cycle_schedule', tuya.valueConverter.raw],
            [107, 'reclose_recover_seconds', tuya.valueConverter.raw],
            [108, 'random_timing', tuya.valueConverter.raw],
            [109, 'switch_inching', tuya.valueConverter.raw],
            [119, 'power_on_delay', tuya.valueConverter.raw],
            [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw],
            [125, 'time_threshold_lost_flow', tuya.valueConverter.raw],
            [126, 'alarm_set3', tuya.valueConverter.raw],
            [127, 'status', tuya.valueConverter.raw],
            //[127, 'status', customTrueFalseConverter],
            [134, 'relay_status_power_on', tuya.valueConverter.raw],
            //[134, 'relay_status_power_on', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

peekayn84, 2024-08-06:
Bug Fix
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

    const fzLocal = {
        value: {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                    payload = {};
                    for (const dpValue of msg.data.dpValues) {
                            const dp = dpValue.dp;
                            switch (dp) {
                                case 17:
                                    {
                                        payload['over_leakage_current'] = legacy.getDataValue(dpValue)[3];
                                        payload['over_temperature'] = legacy.getDataValue(dpValue)[7];
                                        return payload;
                                    };
                                case 18:
                                    {
                                        payload['under_voltage'] = legacy.getDataValue(dpValue)[11];
                                        overvoltage=legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7]
                                        payload['over_voltage'] = overvoltage;
                                        overcurrent=legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                                        payload['over_current']=overcurrent/10.0;
                                        return payload;
                                    };
                            };
                    };
            },
        },
    };

const tz_settings = {
    key: ['reclose_recover_seconds','clear_energy','under_voltage','over_voltage','reclosing_allowed_times','over_leakage_current','over_temperature','over_current'],
    convertSet: async (entity, key, value, meta) => {
        const settingType = {'trip': 2, 'alarm': 1, 'closed': 0};
        // const typeswitch = {'ON':true, 'OFF':false}
        let dp_key = -1; // value if not found
        // identify dp.
        const tuyadatapoints_list = meta.mapped.meta.tuyaDatapoints
        for (let i = 0; i < tuyadatapoints_list.length; i++) {
            if (tuyadatapoints_list[i][1] === key) {
                dp_key = tuyadatapoints_list[i][0];
                break;
            }
        }
        switch (key) {
        case 'reclose_recover_seconds': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'reclosing_allowed_times': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'clear_energy': {
            await tuya.sendDataPointBool(entity, dp_key, value === 'ON');
            break;
        }
        case 'over_leakage_current': {
            const value_send = meta.state['alarm1'];
            value_send['3']=value;
[... 121 lines cut ...]
            [17, 'alarm1', tuya.valueConverter.raw],
            [18, 'alarm2', tuya.valueConverter.raw],
            [101, 'restore_time', tuya.valueConverter.raw], // Deprecated
            [102, 'reclosing_allowed_times', tuya.valueConverter.raw],
            [103, 'temperature', tuya.valueConverter.raw],
            [104, 'reclosing_enable', tuya.valueConverter.onOff],
            [105, 'timer', tuya.valueConverter.raw],
            [106, 'cycle_schedule', tuya.valueConverter.raw],
            [107, 'reclose_recover_seconds', tuya.valueConverter.raw],
            [108, 'random_timing', tuya.valueConverter.raw],
            [109, 'switch_inching', tuya.valueConverter.raw],
            [119, 'power_on_delay', tuya.valueConverter.raw],
            [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw],
            [125, 'time_threshold_lost_flow', tuya.valueConverter.raw],
            [126, 'alarm_set3', tuya.valueConverter.raw],
            [127, 'status', tuya.valueConverter.raw],
            //[127, 'status', customTrueFalseConverter],
            [134, 'relay_status_power_on', tuya.valueConverter.raw],
            //[134, 'relay_status_power_on', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

peekayn84, 2024-08-06:
Hot Fix
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

    const fzLocal = {
        value: {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                    payload = {};
                    for (const dpValue of msg.data.dpValues) {
                            const dp = dpValue.dp;
                            switch (dp) {
                                case 17:
                                    {
                                        payload['over_leakage_current'] = legacy.getDataValue(dpValue)[3];
                                        payload['over_temperature'] = legacy.getDataValue(dpValue)[7];
                                        return payload;
                                    };
                                case 18:
                                    {
                                        payload['under_voltage'] = legacy.getDataValue(dpValue)[11];
                                        overvoltage=legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7]
                                        payload['over_voltage'] = overvoltage;
                                        overcurrent=legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                                        payload['over_current']=overcurrent/10.0;
                                        return payload;
                                    };
                            };
                    };
            },
        },
    };

const tz_settings = {
    key: ['reclose_recover_seconds','clear_energy','under_voltage','over_voltage','reclosing_allowed_times','over_leakage_current','over_temperature','over_current'],
    convertSet: async (entity, key, value, meta) => {
        const settingType = {'trip': 2, 'alarm': 1, 'closed': 0};
        // const typeswitch = {'ON':true, 'OFF':false}
        let dp_key = -1; // value if not found
        // identify dp.
        const tuyadatapoints_list = meta.mapped.meta.tuyaDatapoints
        for (let i = 0; i < tuyadatapoints_list.length; i++) {
            if (tuyadatapoints_list[i][1] === key) {
                dp_key = tuyadatapoints_list[i][0];
                break;
            }
        }
        switch (key) {
        case 'reclose_recover_seconds': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'reclosing_allowed_times': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'clear_energy': {
            await tuya.sendDataPointBool(entity, dp_key, value === 'ON');
            break;
        }
        case 'over_leakage_current': {
            const value_send = meta.state['alarm1'];
            value_send['3']=value;
[... 121 lines cut ...]
            [17, 'alarm1', tuya.valueConverter.raw],
            [18, 'alarm2', tuya.valueConverter.raw],
            [101, 'restore_time', tuya.valueConverter.raw], // Deprecated
            [102, 'reclosing_allowed_times', tuya.valueConverter.raw],
            [103, 'temperature', tuya.valueConverter.raw],
            [104, 'reclosing_enable', tuya.valueConverter.onOff],
            [105, 'timer', tuya.valueConverter.raw],
            [106, 'cycle_schedule', tuya.valueConverter.raw],
            [107, 'reclose_recover_seconds', tuya.valueConverter.raw],
            [108, 'random_timing', tuya.valueConverter.raw],
            [109, 'switch_inching', tuya.valueConverter.raw],
            [119, 'power_on_delay', tuya.valueConverter.raw],
            [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw],
            [125, 'time_threshold_lost_flow', tuya.valueConverter.raw],
            [126, 'alarm_set3', tuya.valueConverter.raw],
            [127, 'status', tuya.valueConverter.raw],
            //[127, 'status', customTrueFalseConverter],
            [134, 'relay_status_power_on', tuya.valueConverter.raw],
            //[134, 'relay_status_power_on', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

francisPiragua, 2024-08-06:
Thank you very much @peekayn84, I'm trying it and it works, but I don't receive data on the energy.

peekayn84, 2024-08-06:
I don't know what mean at energy, but interface realised it and i show it too, but parametr at down "Power" work correcty, if i understand you right this is what you need

> Thank you very much @peekayn84, I'm trying it and it works, but I don't receive data on the energy.

francisPiragua, 2024-08-06:
I talk about this paramater:

[image]

peekayn84, 2024-08-06:
I will check it, i have same problem

peekayn84, 2024-08-06:
> I talk about this paramater:
> 
> ![imagen](https://private-user-images.githubusercontent.com/177526759/355558222-20cc4a8a-0273-496c-9c4b-01bb356668b8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI5Nzk4ODUsIm5iZiI6MTcyMjk3OTU4NSwicGF0aCI6Ii8xNzc1MjY3NTkvMzU1NTU4MjIyLTIwY2M0YThhLTAyNzMtNDk2Yy05YzRiLTAxYmIzNTY2NjhiOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNlQyMTI2MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZmQ1MDM3Y2YxOTUwMTdjYmQ3OWRjZTVkNmFhZGYyZjQ2YjdmNDFlMmI4NjJkN2JkODhiMmJmMWJiMjMzYWVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SJNj6SnsjhW508UuqkIffOith9fY8eTkyF_ho9jZ9uw)

Fixed

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

    const fzLocal = {
        value: {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                    payload = {};
                    for (const dpValue of msg.data.dpValues) {
                            const dp = dpValue.dp;
                            switch (dp) {
                                case 17:
                                    {
                                        payload['over_leakage_current'] = legacy.getDataValue(dpValue)[3];
                                        payload['over_temperature'] = legacy.getDataValue(dpValue)[7];
                                        return payload;
                                    };
                                case 18:
                                    {
                                        payload['under_voltage'] = legacy.getDataValue(dpValue)[11];
                                        overvoltage=legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7]
                                        payload['over_voltage'] = overvoltage;
                                        overcurrent=legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                                        payload['over_current']=overcurrent/10.0;
                                        return payload;
                                    };
                            };
                    };
            },
        },
    };

const tz_settings = {
    key: ['reclose_recover_seconds','clear_energy','under_voltage','over_voltage','reclosing_allowed_times','over_leakage_current','over_temperature','over_current'],
    convertSet: async (entity, key, value, meta) => {
        const settingType = {'trip': 2, 'alarm': 1, 'closed': 0};
        // const typeswitch = {'ON':true, 'OFF':false}
        let dp_key = -1; // value if not found
        // identify dp.
        const tuyadatapoints_list = meta.mapped.meta.tuyaDatapoints
        for (let i = 0; i < tuyadatapoints_list.length; i++) {
            if (tuyadatapoints_list[i][1] === key) {
                dp_key = tuyadatapoints_list[i][0];
                break;
            }
        }
        switch (key) {
        case 'reclose_recover_seconds': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'reclosing_allowed_times': {
            await tuya.sendDataPointValue(entity, dp_key, value);
            break;
        }
        case 'clear_energy': {
            await tuya.sendDataPointBool(entity, dp_key, value === 'ON');
[... 126 lines cut ...]
            [17, 'alarm1', tuya.valueConverter.raw],
            [18, 'alarm2', tuya.valueConverter.raw],
            [101, 'restore_time', tuya.valueConverter.raw], // Deprecated
            [102, 'reclosing_allowed_times', tuya.valueConverter.raw],
            [103, 'temperature', tuya.valueConverter.raw],
            [104, 'reclosing_enable', tuya.valueConverter.onOff],
            [105, 'timer', tuya.valueConverter.raw],
            [106, 'cycle_schedule', tuya.valueConverter.raw],
            [107, 'reclose_recover_seconds', tuya.valueConverter.raw],
            [108, 'random_timing', tuya.valueConverter.raw],
            [109, 'switch_inching', tuya.valueConverter.raw],
            [119, 'power_on_delay', tuya.valueConverter.raw],
            [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw],
            [125, 'time_threshold_lost_flow', tuya.valueConverter.raw],
            [126, 'alarm_set3', tuya.valueConverter.raw],
            [127, 'status', tuya.valueConverter.raw],
            //[127, 'status', customTrueFalseConverter],
            [134, 'relay_status_power_on', tuya.valueConverter.raw],
            //[134, 'relay_status_power_on', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

francisPiragua, 2024-08-07:
Excellent work, thank you very much.

peekayn84, 2024-08-07:
And all function not realized in this version here. Must have as i think it's clear data energy button:
[image]
I tuya app need write pass, and i real life too. But i don't know where and how. Cause wipe "Clear energy" button not work and as i think it cause whitout pass.
PS I applologize for my english. Ukraine my native language.

peekayn84, 2024-08-07:
Ok, i check it, and as i understand right, it can't reset Energy remotly, not in tuya app, not through zigbee2mqtt

peekayn84, 2024-08-07:
So it's maybe final version of this

Koenkk, 2024-08-08:
@peekayn84 the converter is not yet in a state that I can integrate it. We should e.g. get rid of the `tz_settings`. I guess you redefined it because some keys are missing. I've fixed that now in https://github.com/Koenkk/zigbee-herdsman-converters/commit/36e7440a4fa34b6757ce534771408d5fa0360db5 . Can you check with the latest dev?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

peekayn84, 2024-08-08:
I use it cause i don't know how change complex value like over_voltage, it part of 18 datapoit, i delete tz_setting and get this error
[image]

skinkie, 2024-08-10:
> @peekayn84 the converter is not yet in a state that I can integrate it. We should e.g. get rid of the `tz_settings`. I guess you redefined it because some keys are missing. I've fixed that now in [Koenkk/zigbee-herdsman-converters@36e7440](https://github.com/Koenkk/zigbee-herdsman-converters/commit/36e7440a4fa34b6757ce534771408d5fa0360db5) . Can you check with the latest dev?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Should it be removed and added again? Or should this already have worked out of the box with switching to dev (and npm i, and restart).

[image]

skinkie, 2024-08-10:
@peekayn84 misunderstood @Koenkk, I tried your variant as external converter. Notice that all values remain 'Null'.

[image]

Koenkk, 2024-08-12:
@peekayn84 is this on the latest dev branch? For more complex value converters, see e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/b3169739e07ee6ffa90da93d21eefe78bb046d07/src/lib/tuya.ts#L587

DmytroKozlovskyi, 2024-09-07:
Seems fixed by [TZE204_fhvdgeuh definition](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7932)

ia1gerebec, 2024-09-07:
новое обновление1.40.1-1 решило но есть проблемы по отображению вольт и ампер в пользовательской карточке
[image]

closed 2024-12-28 by github-actions[bot]