Koenkk/zigbee2mqtt #23486: Third Reality Smart Switch Gen 3 (3RSS009Z) reporting much lower battery level after 1.39.1 update

loafbutts, 2024-08-02:
### What happened?

Hello! I have four Third Reality 3RSS009Z switches set up in Z2M. Yesterday, all four were reporting battery status in the 40-50% range. After updating to 1.39.1, all are now reading 12-16%. The switches typically start flashing their physical low battery lights in the low 20s, and none of them are currently doing so. I also tried swapping a completely fresh set of batteries into one of the switches, and it only showed 16%. In addition to the sudden drop, these factors lead me to believe it is a change in reporting rather than the batteries actually being that low. I only have one other battery-powered device on my network, and its Z2M battery status did not change like the 3RSS009Z ones did. Thanks!

### What did you expect to happen?

Battery status for 3RSS009Z switches would be the same as it was before the 1.39.1 update.

### How to reproduce it (minimal and precise)

Happened immediately after updating to 1.39.1.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20221226

### Adapter

Itead Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-P)

### Setup

Add-on on Home Assistant OS, on Odroid N2+

maekse, 2024-10-14:
seeing something similar here. dropped from 100 to 42, but a while after the z2m update.
state:
{
    "battery": 42,
    "linkquality": 220,
    "state": "OFF",
    "update": {
        "installed_version": 29,
        "latest_version": 29,
        "state": "idle"
    },
    "update_available": false,
    "voltage": 2900
}

maekse, 2024-10-14:
seems it's a problem converting: zhc:tz: Read result of 'genPowerCfg': {"batteryPercentageRemaining":139}

Koenkk, 2024-10-14:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. (wait 24 hours before the sensor reports a new battery % value)

loafbutts, 2024-11-10:
This is back to normal with the latest Z2M update. Thank you very much!

closed 2024-11-10 by Koenkk