Koenkk/zigbee2mqtt #23490: Using DNS Entry in Kubernetes causes getaddrinfo ENOTFOUND during service start

rummens, 2024-08-03:
### What happened?

Hi there,
I am setting up Z2M on my new OpenShift SNO and I ran into a small issue resolving the MQTT service. It seems no matter what I do the node.js app of Z2M is not able to resolve the K8s internal DNS record. I tried with/without port, prefix and a bunch of permutations but nothing except a straight IP seems to be accepted. This is of course not ideal in the highly flexible nature of K8s. 

Hope the outputs below help, as I am no expert in node I wasn't really sure if I can tell it somehow to use another way of resolving DNS. None of the standard search methods gave me any useful hints unfortunately.

Appreciate any help
Marcel

1. Z2M Helm Chart: https://github.com/Koenkk/zigbee2mqtt-chart
2. MQTT Helm Chart: https://truecharts.org/charts/stable/mosquitto/

Helm Chart Config:

```
[...]
  zigbee2mqtt:
    homeassistant:
      enabled: true
    permit_join: true
    timezone: "Europe/Berlin"
    mqtt:
      server: "mqtt://172.30.185.244"  # working
      server: "mqtt://mosquitto.mosquitto.svc.cluster.local"  # not working
```

nslookup in the Pod confirms that DNS is correctly working:
```
/app # nslookup mosquitto.mosquitto.svc.cluster.local
Server:         172.30.0.10
Address:        172.30.0.10:53

Name:   mosquitto.mosquitto.svc.cluster.local
Address: 172.30.185.244
```

The Pods output (error can be seen at the end):
```
Using '/app/data' as data directory
[2024-08-03 09:13:23] info: z2m: Logging to console
[2024-08-03 09:13:23] info: z2m: Starting Zigbee2MQTT version 1.37.1 (commit #ea39d86)
[2024-08-03 09:13:23] info: z2m: Starting zigbee-herdsman (0.46.6)
[2024-08-03 09:13:24] info: zh:ember: ======== Ember Adapter Starting ========
[2024-08-03 09:13:24] info: zh:ember:ezsp: ======== EZSP starting ========
[2024-08-03 09:13:24] info: zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-08-03 09:13:24] info: zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-08-03 09:13:24] info: zh:ember:uart:ash: Serial port opened
[2024-08-03 09:13:24] info: zh:ember:uart:ash: ======== ASH starting ========
[2024-08-03 09:13:25] info: zh:ember:uart:ash: ======== ASH connected ========
[2024-08-03 09:13:25] info: zh:ember:uart:ash: ======== ASH started ========
[2024-08-03 09:13:25] info: zh:ember:ezsp: ======== EZSP started ========
[2024-08-03 09:13:25] warning: zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-08-03 09:13:25] info: zh:ember: [STACK STATUS] Network up.
[2024-08-03 09:13:25] info: zh:ember: [INIT TC] NCP network matches config.
[2024-08-03 09:13:25] info: zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-08-03 09:13:25] info: z2m: zigbee-herdsman started (resumed)
[2024-08-03 09:13:25] info: z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":3,"revision":"7.4.3 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-08-03 09:13:25] info: z2m: Set transmit power to '5'
[2024-08-03 09:13:25] info: z2m: Currently 0 devices are joined:
[2024-08-03 09:13:25] warning: z2m: `permit_join` set to `true` in configuration.yaml.
[2024-08-03 09:13:25] warning: z2m: Allowing new devices to join.
[2024-08-03 09:13:25] info: zh:ember: [STACK STATUS] Network opened.
[2024-08-03 09:13:25] info: z2m: Connecting to MQTT server at mqtt://mosquitto.mosquitto.svc.cluster.local:1883
[2024-08-03 09:13:25] error: z2m: MQTT error: getaddrinfo ENOTFOUND mosquitto.mosquitto.svc.cluster.local
[2024-08-03 09:13:25] error: z2m: MQTT failed to connect, exiting...
```

Resolve conf looks good:

```
cat /etc/resolv.conf 
search zigbee2mqtt.svc.cluster.local svc.cluster.local cluster.local ocp.local.<redacted>
nameserver 172.30.0.10
options ndots:5
```

### What did you expect to happen?

Application uses Pod DNS Config correctly.

### How to reproduce it (minimal and precise)

Specify Cluster internal DNS entry for the MQTT Server.

### Zigbee2MQTT version

1.37.1 commit: ea39d86

### Adapter firmware version

7.4.3 [GA]

### Adapter

ember

### Setup

OpenShift SNO with Helm Chart

### Debug log

[zigbee2mqtt.txt](https://github.com/user-attachments/files/16478699/zigbee2mqtt.txt)

lux73, 2024-08-04:
didn't you think that's a Container (configuration) issue?

from my sight of view z2m is unable to fix your problem

rummens, 2024-08-04:
If it would be the DNS resolution from within the container should not work but it clear does. Thats the thing that confuses me.
Is there any layer between the "OS" part of the container and the NodeJS Application that would need additional config?

closed 2024-12-20 by rummens