Koenkk/zigbee2mqtt #23496: Wrong illuminance reported (TS0225) after update to 1.39.1-1

hagier, 2024-08-03:
### What happened?

After change #19803 illuminance reported is too small (with turned on light I get only 23 lx).

### What did you expect to happen?

It was reporting well until change #19803.

### How to reproduce it (minimal and precise)

Using presence detector TS0225 (_TZE200_2aaelwxk) and update Z2M to newest version.

[image]

### Zigbee2MQTT version

1.39.1-1

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle-P

### Setup

HA Add-on

trackhacs, 2024-08-03:
i have the same issue. repairing did not fix it. 

after moving it to ZHA it works again as expected. 

First trouble with ikea lamp, now this.  ZHA uses differnt Sonoff dongle though.

Koenkk, 2024-08-04:
@kkondyl can you confirm it reports the correct illuminance in your case now? I probably have to revert https://github.com/Koenkk/zigbee-herdsman-converters/commit/1a923cf8c359483446873f5445e2607a208808c6 because of this.

kkondyl, 2024-08-04:
> @kkondyl can you confirm it reports the correct illuminance in your case now? I probably have to revert [Koenkk/zigbee-herdsman-converters@1a923cf](https://github.com/Koenkk/zigbee-herdsman-converters/commit/1a923cf8c359483446873f5445e2607a208808c6) because of this.

I confirm that after update to 1.39.1 all my presence sensors ZG-205Z (_TZE200_2aaelwxk) firmware version 0122052017 are reporting correct illuminance!
My sensors are like the one in below link:
https://smarthomescene.com/reviews/tuya-zigbee-presence-sensor-zg-205z-review/

ViktorAkselrod, 2024-08-05:
I have the same issue. 
After update 1.39.1  the sensor ZG-205Z/A (_TZE200_2aaelwxk) shows absolutely underestimated values. 
Max 28lx after update (2400lx before)

felipejgribeiro, 2024-08-05:
Same problem here... :( 
My Illuminance is showing 27lx for _TZE200_2aaelwxk (TS0225)

bmmmm, 2024-08-06:
My illuminance values were completely wrong and "low" as well.
I downgraded to version 1.38.0-1 and it works again for me as expected

Koenkk, 2024-08-07:
@kkondyl  and the others, could you provide the data/database.db entry of your device?

kkondyl, 2024-08-07:
> @kkondyl and the others, could you provide the data/database.db entry of your device?

please check below one of my 4x active devices:
{"id":36,"type":"Router","ieeeAddr":"0xa4c1380e100c5e50","nwkAddr":41289,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[3,57346,61184,60928,57344],"clusters":{"genBasic":{"attributes":{"modelId":"TS0225","manufacturerName":"_TZE200_2aaelwxk","powerSource":4,"zclVersion":3,"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffea7bcf7","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723059635728}

ViktorAkselrod, 2024-08-07:
`{"id":22,"type":"Router","ieeeAddr":"0xa4c1387471f15cad","nwkAddr":31848,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[3,57346,61184,60928,57344],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffe99847c","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723065573439}`

matriphe, 2024-08-08:
mine:

```json
{
  "id": 4,
  "type": "Router",
  "ieeeAddr": "0xa4c138d723b0376b",
  "nwkAddr": 16902,
  "manufId": 4417,
  "manufName": "_TZE200_2aaelwxk",
  "powerSource": "DC Source",
  "modelId": "TS0225",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        3,
        1280,
        57346,
        61184,
        60928,
        57344,
        1024
      ],
      "outClusterList": [
        3,
        57346,
        61184,
        60928,
        57344
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "zclVersion": 3
          }
        },
        "ssIasZone": {
          "attributes": {
            "iasCieAddr": "0x90395efffe3ed413",
            "zoneState": 0
          }
        },
        "msIlluminanceMeasurement": {
          "attributes": {
            "measuredValue": 31954
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 131,
  "stackVersion": 2,
  "hwVersion": 1,
  "swBuildId": "0122052017",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1723132196691
}
```

hagier, 2024-08-11:
> @kkondyl and the others, could you provide the data/database.db entry of your device?

{"id":52,"type":"Router","ieeeAddr":"0xa4c138d51ff3f611","nwkAddr":57334,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[3,57346,61184,60928,57344],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":24314}},"genBasic":{"attributes":{"modelId":"TS0225","manufacturerName":"_TZE200_2aaelwxk","powerSource":4,"zclVersion":3,"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0029dbfef6","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723398730507}

PTR-AA, 2024-08-11:
{"id":33,"type":"Router","ieeeAddr":"0xa4c13806db2c0815","nwkAddr":60580,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[3,57346,61184,60928,57344],"clusters":{"genBasic":{"attributes":{"zclVersion":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff066efb","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723404208193}

Koenkk, 2024-08-12:
Since all the attributes are the same `swBuildId`, `appVersion`, `stackVersion` are the same, we cannot distinguish  these devices so I have to revert https://github.com/Koenkk/zigbee-herdsman-converters/commit/1a923cf8c359483446873f5445e2607a208808c6 (since it seems to be wrong for more people now than before). 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow.

closed 2024-08-12 by Koenkk

hagier, 2024-08-12:
@Koenkk I bought light meter for measuring and calibrating.

Now sensor is correct for low values (~10lx) but when exposed to light it show wrong:

[image]

[image]

I tried to assign "Illuminance calibration" (x1000%) to multiply values but now it's not correct in low light:

[image]

[image]

Will reverting #19803 change calculations or just multiple value?

Koenkk, 2024-08-13:
@hagier it will just multiply the values, it seems the illuminance sensor is not very reliable?

originalcom, 2024-08-16:
I have the same problem. These work surprigingly well as a low cost LUX sensor. I have one in a clear box outside and use it for dawn / dusk and various other  functions. Is it possible to add a manual definition to overide the inbuilt one for this or do I have to restore the previous version?