Koenkk/zigbee2mqtt #23497: Sonoff SNZB-05P always unvailable short after pairing

cpuks, 2024-08-03:
### What happened?

I cannot get Sonoff water leak sensor to work with z2m 1.39/1.39.1 - device is recognized during pairing stage but one it's paired it's never interviewd / pooled again nor water leakage doesn't trigger leak state. Device is about 8 m away from my coordinator.
Found out that deconz (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7843) had to use custom pooling interval.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240308

### Adapter

UZG-01 (CC2652P7)

### Setup

docker

cpuks, 2024-08-11:
Anything I can do to properly solve that device?

Mikeinnc, 2024-10-19:
I am seeing exactly the same result. A new SNZB05P is recognised by zigbee2mqtt but immediately after pairing, the 'water leak' and 'battery_low' signals become 'N/A'. A SNZ03P and IKEA Vallhorn presence sensor on the same system work perfectly. Why? The water Leak sensor is supposed to be fully integrated with zigbee2mqtt, but there's clearly a problem. It's one I'd love to solve!!

Mikeinnc, 2024-10-19:
Oops! Perhaps I've spoken too soon. I've checked a couple of other channels, and I've just 're-paired' my sensor WITHOUT deleting it from the z2m link - and it is working! It's currently 'dry' and if I put a wet finger across the terminals, it flashes red and 'leaking' in my HA overview. So maybe, this is the answer? Try re-pairing the device a second time and DON'T delete the original. It seems to work!

cpuks, 2024-10-19:
Let me investigate with mine - so pair it and then re-pair by just repeating repairing process on device and not touching anything Z2M - I remember it was working for some period of time then it was stale - so keep an eye if it's responsive after few hours or day-two.

cpuks, 2024-10-19:
Yeah, unfortunately it's unresponsive few minutes after pairing, once you re-pair it, it works then leave it for few minutes try to simulate water leak by connecting two pogo pins with your finger, and you'll see it has no effect, after few hours it'll drop out from network.

Mikeinnc, 2024-10-19:
It's been active for two hours now, and is still showing in HA. I haven't
tried 'wetting' it again yet - I'm in Australia, so I'll try it later (the
sensor is upstairs in the ensuite bathroom). It hasn't dropped off, though.
Strange? I'd spent many, many hours trying to get this sensor to work. If a
simple 're-pairing' is all that is needed, you'd have to ask 'why?'

cpuks, 2024-10-20:
Just try to trigger leak scenario please as it's in network - ok but doesn't work as intended at least in my case, I did exactly what you did - paired brand new sensor to with my network, then without removing it from network initiated pairing process again and OK it was responsive shortly after second pairing process but now even when I put in into water bucket it doesn't send that there's a leak or something.
Can you also please check what firmware version you have in yours? I'm on 1.0.1
Thanks

Mikeinnc, 2024-10-20:
Apologies for the delay in replying! Monday morning in Australia and I've
just seen this. I certainly did try the 'wet finger test' across the sensor
pins yesterday, and it reflected 'wet' and as soon as I took the finger
away, it returned to 'dry'. I've checked the firmware version and it is
1.0.2. Important? I'm not sure. I'm using a Sonoff USB co-ordinator with
Zigbee2MQTT on HA running on a RPi5 version 2024.10.3 with OS version 13.2.
Any of this help? I hope so!

cpuks, 2024-10-24:
Yeah mine is at 1.0.1 firmware so I'll have to update somehow to 1.0.2 and then try once more.

Mikeinnc, 2024-10-25:
I realise mine is at 1.0.2 - I can't recall manually updating it? Isn't
there an option for it to be automatically updated? The date of the
firmware is shown as 20240524 which I guess is 24 MAY 2024. I suppose this
would help - but I'm not sure if it was the 'defining' issue? Certainly
worth a try at updating, though - it can't hurt! I wish you all the best of
luck. I know how irritating these things can be!!

Kind regards

Mike

cpuks, 2024-10-25:
Well no option to do OTA via Z2M so I had to create issue https://github.com/Koenkk/zigbee2mqtt/issues/24475

cpuks, 2024-10-29:
A note for anyone that has / have timeout problems or ZCL OnOff and similar - I'm on UZG01 running 20240710, recently bought Sonoff SNZB-05P water leak sensor and I never had them working - my previous firmware from 20240316 was not working with Sonoff water leak too, recently had one of my tuya 10W spotlight failing so wanted to use spares as I had some - another problem - like light cannot finish pairing, and very unresponsive - OK found some problems on Z2M github - let's try something that works every time - IKEA T2035 lamp that's supported for long, long time - again had timeouts, errors during pairing - nothing but troubles, downgraded Z2M to 1.40.1 - no luck, went to 1.39.2 - same - problems, in mean time I went back to 20240316 firmware and well same problems, network is stable, but I cannot pair any new devices - IKEA Sonoff tuya - so well let's get back to 20230507 - and finally - Sonoff water leak works for first time, tuya works, IKEA works! I'm so happy now - even started looking at interference, but I'm on 25 for zigbee and 1/6 for Wi-Fi... so well 20230507 it rock solid!

**So flashing 20230507 firmware solved issues with CC2652P7**