Koenkk/zigbee2mqtt #23498: [New device support]: EGLO HERRORA-Z ceiling light model 99639

huibuh420, 2024-08-03:
### Link

https://www.eglo.com/uk/ceiling-light-herrora-z-99639.html

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0xa4c138e1956c6cad","nwkAddr":56868,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","appVersion":0}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":2,"colorTemperature":153}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1722691977133}

### Comments

I failed with the instructions to push the changes to github and to create the pull request. 
The light can change the color, aswell it is dimmable. Please include this device.
[image]

### External definition

```shell
const {deviceEndpoints, identify, light, commandsOnOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TLSR82xx'],
    model: '99639',
    vendor: 'AwoX',
    description: 'EGLO HERRORA-Z ceiling light',
    extend: [deviceEndpoints({"endpoints":{"1":1,"3":3}}), identify(), light({"colorTemp":{"range":[153,370]}}), commandsOnOff()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Koenkk, 2024-08-05:
Does it support both color (rgb) and color temp or only color temp?

huibuh420, 2024-08-06:
Hi,
only white color temp

Koenkk, 2024-08-07:
Without the external definition, how is it detected in z2m 1.39.1? (can you post a screenshot of the about tab)

huibuh420, 2024-08-07:
[image]
[image]

huibuh420, 2024-08-07:
I hope that german is fine for you, otherwise I can change the language to english.

Koenkk, 2024-08-08:
33957 supports the color temp, so I guess this is only a cosmetic issue? (wrong picture/description which cannot be fixed since all these devices use the same model id)

closed 2024-12-28 by github-actions[bot]