Koenkk/zigbee2mqtt #23502: [New device support]:

unwiredtech, 2024-08-03:
### Link

https://www.amazon.com.au/Intelligent-Photovoltaic-Bidirectional-Automatic-Compatible/dp/B0CYPQHVYR

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c138fa6becd2bf","nwkAddr":22993,"manufId":4417,"manufName":"_TZE204_lb0fsvba","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[>

### Comments

I tested this external definition but for some reason the Z2M Frontend could not see the Exposed sensors.

### External definition

```shell
//Device type Router
//Zigbee Model - TS0601
//Zigbee Manufacturer - _TZE204_lb0fsvba

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Identifies the device using fingerprint instead of zigbeeModel for Tuya devices
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_lb0fsvba',
        },
    ],
    model: 'ZBN-DJ-63',
    vendor: 'Tuya',
    description: 'Tuya Zigbee Smart Circuit Breaker with energy monitoring and protection settings',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, tuya.fz.datapoints],
    toZigbee: [tz.on_off, tz.electrical_measurement, tz.metering, tuya.tz.datapoints],
    exposes: [
        e.switch().setAccess('state', ea.ALL).withDescription('Switch the circuit breaker on/off'),
        e.energy().withAccess(ea.STATE).withDescription('Energy usage in kWh'),
        e.power().withAccess(ea.STATE).withDescription('Power usage in W'),
        e.voltage().withAccess(ea.STATE).withDescription('Voltage in V'),
        e.current().withAccess(ea.STATE).withDescription('Current in A'),

        exposes.enum('fault', ea.STATE, ['clear', 'over_current', 'over_power', 'over_voltage', 'under_voltage', 'leakage'])
            .withDescription('Fault status (clear = no faults)'),

        exposes.numeric('temperature_threshold', ea.ALL)
            .withUnit('°C')
            .withDescription('Temperature threshold for triggering protection')
            .withValueMin(10)
            .withValueMax(85),

        exposes.numeric('power_threshold', ea.ALL)
            .withUnit('W')
            .withDescription('Power threshold for triggering protection')
            .withValueMin(0)
            .withValueMax(9999),

        exposes.numeric('over_voltage_threshold', ea.ALL)
            .withUnit('V')
            .withDescription('Over voltage threshold for protection')
            .withValueMin(120)
            .withValueMax(300),

        exposes.numeric('under_voltage_threshold', ea.ALL)
            .withUnit('V')
            .withDescription('Under voltage threshold for protection')
            .withValueMin(80)
            .withValueMax(210),

        exposes.numeric('over_current_threshold', ea.ALL)
            .withUnit('A')
            .withDescription('Over current threshold for protection')
            .withValueMin(1)
            .withValueMax(63),

        exposes.numeric('leakage_current_threshold', ea.ALL)
            .withUnit('mA')
            .withDescription('Leakage current threshold for protection')
            .withValueMin(10)
            .withValueMax(99),

        exposes.binary('power_outage_memory', ea.ALL, 'ON', 'OFF')
            .withDescription('Retain state after power outage (ON - remembers last state)'),

        exposes.enum('indicator_mode', ea.ALL, ['off', 'normal', 'inverse'])
            .withDescription('LED indicator mode'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [6, null, tuya.valueConverter.phaseVariant2], // Handles voltage, current, power
            [10, 'fault', tuya.valueConverterBasic.lookup({
                'clear': 0,
                'over_current': 1,
                'over_power': 2,
                'over_voltage': 4,
                'under_voltage': 9,
                'leakage': 10,
            })],
            [17, 'temperature_threshold', tuya.valueConverter.raw],
            [18, 'meter_id', tuya.valueConverter.raw],
            [19, 'power_threshold', tuya.valueConverter.divideBy10],
            [20, 'clear_fault', tuya.valueConverter.onOff],
            [21, 'leakage_current_threshold', tuya.valueConverter.divideBy100],
            [22, 'indicator_mode', tuya.valueConverterBasic.lookup({
                'off': 0,
                'normal': 1,
                'inverse': 2,
            })],
            [23, 'power_outage_memory', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

unwiredtech, 2024-08-03:
[image]

unwiredtech, 2024-08-03:
[image]
Product Image

unwiredtech, 2024-08-03:
Product Spec:

-  Name: Circuit Breaker 
-  Material: PC 
-  Rated vottage: 230VAC 
-  Operation voltage: AC80V-400V(single phase) 
-  Rated frequency: 50/60Hz 
-  Electric current: (>A) 1-63A 

 
 Setting Ranges: 

-  Under voltage protecting: 80V-210V adjustable 
-  Overvoltage protecting: 120-300V adjustable 
-  Overcurrent protecting:1.0A-63A adjustable 
-  Reconnection time (delay time): 10s-99s adjustable 
-  Overcurrent and leakage: 1-30 off adjustable 
-  Leakage protecting: 10-99mA adjustable 
-  Over temperature: 10-85 adjustable 
-  Over current delay time: 1s-15s adjustable 
-  Loss of current delay time: 1s-15s adjustable

droserHA, 2024-08-06:
вот что набросал я. частично работает

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {

        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lb0fsvba']),
        model: 'TS0601_din_2',
        vendor: 'TuYa',
        description: 'Zigbee DIN energy meter',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        whiteLabel: [{vendor: 'XOCA', model: 'DAC2161C'}],
        exposes: [tuya.exposes.switch(), e.energy(), e.power(), e.voltage(), e.current(),
            exposes.enum('fault', ea.STATE, ['clear', 'over_current_threshold', 'over_power_threshold',
                'over_voltage threshold', 'wrong_frequency_threshold']).withDescription('Fault status of the device (clear = nothing)'),
            exposes.enum('threshold_1', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
                .withDescription('State of threshold_1'),
            exposes.binary('threshold_1_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
            exposes.numeric('threshold_1_value', ea.STATE_SET)
                .withDescription('Can be in Volt or Ampere depending on threshold setting. Setup the value on the device'),
            exposes.enum('threshold_2', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
                .withDescription('State of threshold_2'),
            exposes.binary('threshold_2_protection', ea.STATE, 'ON', 'OFF')
                .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
            exposes.numeric('threshold_2_value', ea.STATE)
                .withDescription('Setup value on the device'),
			exposes.numeric('reverse_energy', ea.STATE).withUnit('kWh')
                .withDescription('Reverse Energy Total'),
			exposes.numeric('temperature', ea.STATE).withUnit('°C')
                .withDescription('Temperature'),
            exposes.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Turn ON to clear last the fault'),
            exposes.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'energy', tuya.valueConverter.divideBy100],
                [3, null, null], // Monthly, but sends data only after request
                [4, null, null], // Dayly, but sends data only after request
                [6, null, tuya.valueConverter.phaseVariant2], // voltage and current
                [9, 'fault', tuya.valueConverterBasic.lookup({'clear': 0, 'over_current_threshold': 1,
                    'over_power_threshold': 2, 'over_voltage_threshold': 4, 'wrong_frequency_threshold': 8})],
                [11, null, null], // Frozen - strange function, in native app - nothing is clear
                [16, 'state', tuya.valueConverter.onOff],
                [17, null, tuya.valueConverter.threshold], // It's settable, but can't write converter
                [103, 'temperature', tuya.valueConverter.raw],
                [20, 'clear_fault', tuya.valueConverter.onOff], // Clear fault
                [13, null, null], // Balance energy
                [110, 'reverse_energy', tuya.valueConverter.divideBy100], // Reverse Energy Total
                [23, null, null], // Forward Energy T3 - don't know what this
                [24, null, null], // Forward Energy T4 - don't know what this
            ],
        },

};

module.exports = definition;

unwiredtech, 2024-08-13:
Apologies for late reply.

I tested the definition and seems to work, but as you said some features are not working but this is better than before. 

Thanks!
[image]

slayer, 2024-08-14:
Hey, thanks for sketch, I have the same device

I've tried it, but my circuit breaker fails with `wrong_frequency_threshold` error (power is off), and I don't know how to change this threshold (or even is it real `wrong_frequency_threshold` error, because multimeter says frequency in socket is 50.00 Hz)

[image]

[image]

git-greatguy, 2024-08-15:
Hi, i'm interested in tring this device to get "negative" apparent power injected to the grid from my solar setup.
Is it working well / stable enough for this purpose ?
thanks !

slayer, 2024-08-15:
@git-greatguy do you want to measure DC or AC? 
I believe this module is able to measure AC only. for DC there are other devices

git-greatguy, 2024-08-15:
Sure, it's for measurin AC flow between grid and my inverter. 👍 
my current zigbee module (TO-Q-SY1-JZT), send accurate values but values are not "signed". It's a bit annoying to know if i inject or pull from the grid ! This device looks promising to clear this issue.

slayer, 2024-08-15:
@git-greatguy I would recommend for this another [device](https://www.aliexpress.com/item/1005005770299928.html), it works good with in/out measurement

slayer, 2024-08-23:
Any ideas how to fix 

> I've tried it, but my circuit breaker fails with `wrong_frequency_threshold`

?

pana800, 2024-12-09:
> ### Link
> https://www.amazon.com.au/Intelligent-Photovoltaic-Bidirectional-Automatic-Compatible/dp/B0CYPQHVYR
>
> ### Database entry [...]

there are 2 syntax mistake in this file I guest 
No comma after value: 10 in this datapoint ('leakage;: 10)
and similar for datapoint 22: no ',' after 'inverse': 2

     [10, 'fault', tuya.valueConverterBasic.lookup({
>                 'clear': 0,
>                 'over_current': 1,
>                 'over_power': 2,
>                 'over_voltage': 4,
>                 'under_voltage': 9,
>                 'leakage': 10,
>             })],

sudipta26889, 2024-12-28:
I managed to add my ZBN-J-63 using
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const { Buffer } = require('buffer'); // Ensure Buffer is available

// Check if getDataValue function exists, if not define it
if (typeof tuya.getDataValue !== 'function') {
    tuya.getDataValue = (dpValue) => {
        // Extract data value from dpValue
        return dpValue.data;
    };
}

const tzLocal = {
    state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value === 'ON');
        },
    },
};

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE204_xxxxxxxx' }],
    model: 'ZBN-J-63',
    vendor: 'Tuya',
    description: 'Tuya Zigbee Smart Circuit Breaker with energy monitoring and protection settings',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, tuya.fz.datapoints],
    toZigbee: [tzLocal.state, tuya.tz.datapoints],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.switch().setAccess('state', ea.STATE_SET).withDescription('Switch the circuit breaker on/off'),
        e.energy().withAccess(ea.STATE).withDescription('Energy usage in kWh'),
        e.power().withAccess(ea.STATE).withDescription('Power usage in W'),
        e.voltage().withAccess(ea.STATE).withDescription('Voltage in V'),
        e.current().withAccess(ea.STATE).withDescription('Current in A'),
        exposes.enum('fault', ea.STATE, ['clear', 'over_current', 'over_power', 'over_voltage', 'under_voltage', 'leakage'])
            .withDescription('Fault status (clear = no faults)'),
        exposes.numeric('temperature_threshold', ea.ALL)
            .withUnit('°C')
            .withDescription('Temperature threshold for triggering protection')
            .withValueMin(10)
            .withValueMax(85),
        exposes.numeric('power_threshold', ea.ALL)
            .withUnit('W')
            .withDescription('Power threshold for triggering protection')
            .withValueMin(0)
            .withValueMax(9999),
        exposes.numeric('over_voltage_threshold', ea.ALL)
            .withUnit('V')
            .withDescription('Over voltage threshold for protection')
            .withValueMin(120)
            .withValueMax(300),
        exposes.numeric('under_voltage_threshold', ea.ALL)
            .withUnit('V')
            .withDescription('Under voltage threshold for protection')
            .withValueMin(80)
            .withValueMax(210),
        exposes.numeric('over_current_threshold', ea.ALL)
            .withUnit('A')
            .withDescription('Over current threshold for protection')
            .withValueMin(1)
            .withValueMax(63),
        exposes.numeric('leakage_current_threshold', ea.ALL)
            .withUnit('mA')
            .withDescription('Leakage current threshold for protection')
            .withValueMin(10)
            .withValueMax(99),
        exposes.binary('power_outage_memory', ea.ALL, 'ON', 'OFF')
            .withDescription('Retain state after power outage (ON - remembers last state)'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'normal', 'inverse'])
            .withDescription('LED indicator mode'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [6, null, tuya.valueConverter.phaseVariant2],
            [10, 'fault', tuya.valueConverterBasic.lookup({
                'clear': 0,
                'over_current': 1,
                'over_power': 2,
                'over_voltage': 4,
                'under_voltage': 9,
                'leakage': 10
            })],
            [17, 'temperature_threshold', tuya.valueConverter.raw],
            [18, 'meter_id', tuya.valueConverter.raw],
            [19, 'power_threshold', tuya.valueConverter.divideBy10],
            [20, 'clear_fault', tuya.valueConverter.onOff],
            [21, 'leakage_current_threshold', tuya.valueConverter.divideBy100],
            [22, 'indicator_mode', tuya.valueConverterBasic.lookup({
                'off': 0,
                'normal': 1,
                'inverse': 2
            })],
            [23, 'power_outage_memory', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

droserHA, 2024-12-29:
Your converter didn't work for me. Here's the updated code, but the datapoints still need to be tested. Again, this converter works for me, but not all the functionality is implemented.

 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');

const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');

const reporting = require('zigbee-herdsman-converters/lib/reporting');

const e = exposes.presets;

const ea = exposes.access;

const tuya = require('zigbee-herdsman-converters/lib/tuya');

const { Buffer } = require('buffer'); // Ensure Buffer is available

 

// Check if getDataValue function exists, if not define it

if (typeof tuya.getDataValue !== 'function') {

    tuya.getDataValue = (dpValue) => {

        // Extract data value from dpValue

        return dpValue.data;

    };

}

 

const tzLocal = {

    state: {

        key: ['state'],

        convertSet: async (entity, key, value, meta) => {

            await tuya.sendDataPointBool(entity, 16, value === 'ON');

        },

    },

};

 

const definition = {

    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lb0fsvba']), // Используем рабочий fingerprint

    model: 'ZBN-DJ-63',

    vendor: 'TuYa',

    description: 'Tuya Zigbee Smart Circuit Breaker with energy monitoring and protection settings',

    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, tuya.fz.datapoints],

    toZigbee: [tzLocal.state, tuya.tz.datapoints],

[... 150 lines cut ...]
            [19, 'power_threshold', tuya.valueConverter.divideBy10],

            [21, 'leakage_current_threshold', tuya.valueConverter.divideBy100],

            [22, 'indicator_mode', tuya.valueConverterBasic.lookup({

                'off': 0,

                'normal': 1,

                'inverse': 2

            })],

            [23, 'power_outage_memory', tuya.valueConverter.onOff],

        ],

    },

};

 

module.exports = definition;

closed 2025-03-07 by github-actions[bot]

LightMoon, 2025-05-05:
I'm looking into using the ZBN-DJ-63 smart circuit breaker, which supports Zigbee and claims to offer bidirectional metering for photovoltaic (PV) setups (i.e., both grid import and solar export monitoring).

Before proceeding, I’d like to ask:

- Is this model currently supported by Zigbee2MQTT?
- If supported, does Zigbee2MQTT expose both import and export energy values through MQTT?

M66B, 2025-05-05:
The short answer is that the device isn't supported. Better purchase something else.

M66B, 2025-05-05:
It's the same reason why I bought this device. If you can live without the over/under voltage/current options, you can define an external converter:

```
// https://github.com/Koenkk/zigbee2mqtt/issues/23502

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_lb0fsvba', // ZBN-DJ-63
        },
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_kpg9hpb0', // ZBN-JLS-63
        },
    ],
    model: 'ZBN-DJ/JLS-63',
    vendor: 'NoName',
    description: 'Bidirectional energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            tuya.exposes.switch(),
            e.energy(),
            e.voltage(),
            e.power(),
            e.current(),
            e.produced_energy(),
            exposes.numeric('temperature', ea.STATE)
                .withUnit('°C')
                .withDescription('Temperature'),
            exposes.enum('indicator_mode', ea.ALL, ['off', 'normal', 'inverse'])
                .withDescription('LED indicator mode'),
            exposes.binary('power_outage_memory', ea.ALL, 'ON', 'OFF')
                .withDescription('Retain state after power outage (ON - remembers last state)'),
    ],
    meta: {
            tuyaDatapoints: [
                [1, 'energy', tuya.valueConverter.divideBy100], // Forward energy total
                [6, null, tuya.valueConverter.phaseVariant2], // Handles voltage, current, power
                [16, 'state', tuya.valueConverter.onOff],
                [22, 'indicator_mode', tuya.valueConverterBasic.lookup({'off': 0, 'normal': 1, 'inverse': 2})],
                [23, 'power_outage_memory', tuya.valueConverter.onOff],
                [103, 'temperature', tuya.valueConverter.raw],
                [110, 'produced_energy', tuya.valueConverter.divideBy100], // Reverse Energy Total
            ],
    },
    extend: [
    ],
};

module.exports = definition;
```

LightMoon, 2025-05-05:
@M66B, just wondering—any particular reason we couldn’t get it to work?
@Sudipta26889, how’s everything going with your converter?

M66B, 2025-05-05:
For me, voltage, current, power, forward and reverse energy are enough, and I didn't put effort into other options.

rbb-dev, 2025-06-15:
I managed to make it work okay, although I'm still missing a few alarms, such as loss_of_flow_alarm and options. This can be expanded later.
save is as in  zigbee2mqtt/external_converters/zbn-dj63.js
only tested with Z2M 2.4.0

```const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const reporting = require('zigbee-herdsman-converters/lib/reporting');

module.exports = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lb0fsvba']),
    model: 'ZBN-DJ-63',
    vendor: 'Tuya',
    description: 'Tuya smart circuit breaker with energy monitoring and comprehensive alarm control',

    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
    },

    exposes: [
        e.switch().setAccess('state', ea.STATE_SET).withDescription('Circuit Breaker Switch'),
        e.energy().withDescription('Total Forward Energy (kWh)').withAccess(ea.STATE),
        exposes.numeric('reverse_energy', ea.STATE).withUnit('kWh').withDescription('Total Reverse Energy (kWh)'),
        e.power().withDescription('Active Power (W)').withAccess(ea.STATE),
        e.voltage().withDescription('Phase A Voltage (V)').withAccess(ea.STATE),
        e.current().withDescription('Phase A Current (A)').withAccess(ea.STATE),
        exposes.numeric('temperature', ea.STATE).withUnit('°C').withDescription('Current Temperature'),

        exposes.binary('switch_prepayment', ea.STATE_SET, 'ON', 'OFF').withDescription('Prepaid Function Switch'),
        exposes.binary('clear_energy', ea.SET, 'ON', 'OFF').withDescription('Clear Remaining Available Energy'),
        exposes.numeric('balance_energy', ea.STATE).withUnit('kWh').withDescription('Remaining Available Energy Display (kWh)'),
        exposes.numeric('charge_energy', ea.SET).withValueMin(0).withValueMax(999999).withUnit('kWh').withDescription('Energy Recharge'),

        exposes.binary('over_current_breaker', ea.ALL, 'ON','OFF').withDescription('Overcurrent Alarm Enabled'),
        exposes.numeric('over_current_threshold', ea.ALL).withValueMin(100).withValueMax(630).withUnit('mA').withDescription('Overcurrent Threshold'),
        exposes.binary('over_voltage_breaker', ea.ALL, 'ON','OFF').withDescription('Overvoltage Alarm Enabled'),
        exposes.numeric('over_voltage_threshold', ea.ALL).withValueMin(0).withValueMax(999).withUnit('V').withDescription('Overvoltage Threshold'),
        exposes.binary('under_voltage_breaker', ea.ALL, 'ON','OFF').withDescription('Undervoltage Alarm Enabled'),
        exposes.numeric('under_voltage_threshold', ea.ALL).withValueMin(0).withValueMax(999).withUnit('V').withDescription('Undervoltage Threshold'),
        exposes.numeric('leakage_current', ea.STATE).withUnit('mA').withDescription('Residual Current Display'),
        exposes.binary('leakage_breaker', ea.ALL, 'ON','OFF').withDescription('Leakage Current Alarm Enabled'),
        exposes.numeric('leakage_delay', ea.STATE_SET).withValueMin(1).withValueMax(9999).withUnit('s').withDescription('Leakage Delay'),
        exposes.numeric('leakage_threshold', ea.ALL).withUnit('mA').withDescription('Leakage Current Threshold'),
        exposes.binary('high_temperature_breaker', ea.ALL, 'ON','OFF').withDescription('Temperature Alarm Enabled'),
        exposes.numeric('high_temperature_threshold', ea.ALL).withUnit('°C').withDescription('Temperature Threshold'),

        exposes.numeric('countdown', ea.STATE_SET).withValueMin(0).withValueMax(43200).withUnit('s').withDescription('Countdown Timer'),
        exposes.numeric('power_on_delay', ea.STATE_SET).withValueMin(1).withValueMax(9999).withUnit('s').withDescription('Power-On Delay Energization Time'),
        exposes.numeric('recover_count', ea.STATE_SET).withValueMin(0).withValueMax(999).withDescription('Reclosing Attempt Limit'),
        exposes.binary('recover_enable', ea.STATE_SET, 'ON', 'OFF').withDescription('Reclosing Enable'),

        exposes.enum('fault', ea.STATE, [
            'clear', 'short_circuit_alarm', 'surge_alarm', 'overload_alarm', 'leakagecurr_alarm',
            'temp_dif_fault', 'fire_alarm', 'high_power_alarm', 'self_test_alarm', 'ov_cr',
            'unbalance_alarm', 'ov_vol', 'undervoltage_alarm', 'miss_phase_alarm', 'outage_alarm',
            'magnetism_alarm', 'credit_alarm', 'no_balance_alarm'
        ]).withDescription('Fault Alarm'),

        exposes.enum('relay_power_on_state', ea.STATE_SET, ['Off', 'On', 'Restore']).withDescription('Relay Power-On State Setting'),
    ],

    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [6, null, tuya.valueConverter.phaseVariant2],
[... 19 lines cut ...]
            [18, null, tuya.valueConverter.threshold_3],
            [18, "over_current_threshold", tuya.valueConverter.threshold_3],
            [18, "over_current_breaker", tuya.valueConverter.threshold_3],
            [18, "over_voltage_threshold", tuya.valueConverter.threshold_3],
            [18, "over_voltage_breaker", tuya.valueConverter.threshold_3],
            [18, "under_voltage_threshold", tuya.valueConverter.threshold_3],
            [18, "under_voltage_breaker", tuya.valueConverter.threshold_3],
[... cut ...]
//            [109, 'switch_inching',tuya.valueConverter.raw],
            [110, 'reverse_energy',tuya.valueConverter.divideBy100],
            [119, 'power_on_delay',tuya.valueConverter.raw],
            [124, 'alarm_over_current_count',tuya.valueConverter.raw],
            [125, 'alarm_low_current_count',tuya.valueConverter.raw],
            [127, 'status',tuya.valueConverter.raw],
            [134, 'relay_power_on_state', tuya.valueConverterBasic.lookup({Off: 0, On: 1, Restore: 2})],
        ],
    },
};
```

PR2022, 2026-04-08:
I have no idea if everything is correct, but I ran richardsg307's code through the AI, and after several attempts, I was finally able to set the overvoltage and undervoltage thresholds as well as enable and disable them. Thanks to everyone.

Here's my code

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const presets = exposes.presets;
const access = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lb0fsvba']),
    model: 'ZBN-DJ-63',
    vendor: 'TuYa',
    description: 'Zigbee Bi-directional DIN energy meter and breaker',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        presets.switch(),
        presets.power(),
        presets.energy(),
        presets.voltage(),
        presets.current(),
        exposes.numeric('over_current_threshold', access.STATE_SET).withUnit('A').withValueMin(1).withValueMax(63),
        exposes.numeric('over_voltage_threshold', access.STATE_SET).withUnit('V').withValueMin(130).withValueMax(300),
        exposes.numeric('under_voltage_threshold', access.STATE_SET).withUnit('V').withValueMin(65).withValueMax(200),
        exposes.binary('over_current_breaker', access.STATE_SET, 'ON', 'OFF'),
        exposes.binary('over_voltage_breaker', access.STATE_SET, 'ON', 'OFF'),
        exposes.binary('under_voltage_breaker', access.STATE_SET, 'ON', 'OFF'),
        exposes.numeric('temperature', access.STATE).withUnit('°C').withDescription('Temperature'),
        exposes.numeric('temperature_limit', access.STATE_SET).withUnit('°C').withDescription('Thought to be Temperature in °C'),
        exposes.numeric('leakage_limit', access.STATE_SET).withDescription('Thought to be Leakage Current in mA'),
        exposes.enum('fault', access.STATE, ['clear', 'over_current', 'leakage_trip', 'over_voltage']).withDescription('Fault status of the device (clear = nothing)'),
        exposes.binary('clear_fault', access.STATE_SET, 'ON', 'OFF').withDescription('Turn ON to clear last the fault'),
        exposes.binary('recover_enable', access.STATE_SET, 'ON', 'OFF').withDescription('Enable auto-reclose count after trip. Off = infinite retries'),
        exposes.numeric('recover_count', access.STATE_SET).withDescription('Number of reclose attempts. 0 = infinite retries'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [6, null, tuya.valueConverter.phaseVariant2],
            [16, 'state', tuya.valueConverter.onOff],
            [103, 'temperature', tuya.valueConverter.raw],

            // FROM device: liest DP 18 und verteilt alle Werte
            // Array-Format: [1, oc_b, HB, LB, 3, ov_b, 0, ov, 4, uv_b, 0, uv]
            // Index:         0   1    2   3   4   5    6   7  8   9   10  11
            [18, 'over_current_threshold_obj', {
                from: (value) => {
                    return {
                        over_current_threshold: (value[2] << 8 | value[3]) / 10, // ✅ beide Bytes
                        over_voltage_threshold: value[7],
                        under_voltage_threshold: value[11],
                        over_current_breaker: value[1] === 1 ? 'ON' : 'OFF',
                        over_voltage_breaker: value[5] === 1 ? 'ON' : 'OFF',   // ✅ Index 5
                        under_voltage_breaker: value[9] === 1 ? 'ON' : 'OFF',  // ✅ Index 9
                    };
                }
            }],

            // TO device: over_current_threshold ändern
            [18, 'over_current_threshold', {
                to: (value, meta) => {
                    const oc = Math.round(value * 10);
                    const oc_b = (meta.state.over_current_breaker === 'ON' ? 1 : 0);
                    const ov = (meta.state.over_voltage_threshold || 275);
                    const ov_b = (meta.state.over_voltage_breaker === 'ON' ? 1 : 0);
                    const uv = (meta.state.under_voltage_threshold || 175);
[... 47 lines cut ...]
                    const ov_b = (value === 'ON' ? 1 : 0);
                    const ov = (meta.state.over_voltage_threshold || 275);
                    const uv = (meta.state.under_voltage_threshold || 175);
                    const uv_b = (meta.state.under_voltage_breaker === 'ON' ? 1 : 0);
                    return [1, oc_b, (oc >> 8) & 0xFF, oc & 0xFF, 3, ov_b, 0, ov, 4, uv_b, 0, uv];
                }
            }],

            // TO device: under_voltage_breaker umschalten
            [18, 'under_voltage_breaker', {
                to: (value, meta) => {
                    const oc = Math.round((meta.state.over_current_threshold || 63) * 10);
                    const oc_b = (meta.state.over_current_breaker === 'ON' ? 1 : 0);
                    const ov = (meta.state.over_voltage_threshold || 275);
                    const ov_b = (meta.state.over_voltage_breaker === 'ON' ? 1 : 0);
                    const uv_b = (value === 'ON' ? 1 : 0);
                    const uv = (meta.state.under_voltage_threshold || 175);
                    return [1, oc_b, (oc >> 8) & 0xFF, oc & 0xFF, 3, ov_b, 0, ov, 4, uv_b, 0, uv];
                }
            }],
        ],
    },
};

module.exports = definition;`