Koenkk/zigbee2mqtt #23505: [New device support]: Support for Ikea Tretakt E2217 (Australian Tretakt) requested

Elephantoplasty, 2024-08-04:
### Link

https://www.ikea.com/au/en/p/tretakt-plug-smart-10556517/

### Database entry

{"id":4,"type":"Unknown","ieeeAddr":"0x94b216fffec684a3","nwkAddr":45848,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1722743756080}

### Comments

Trying to get the Australian version (E2217) of the Tretakt connected.  The EU one is supported, but not the Australian one.

Multiple attempts at pairing fail, always with a reported support status as 'Null'. and manufacturer as 'Unsupported'.

I had a go at creating a new device, but when I looked for what it exposes, it just says 'Empty exposes definition'.

The External definition is empty.

This is about the limit of my abilities, so I'd appreciate someone with more skills having a go at getting this supported.

Much appreciated.

### External definition

```shell
const definition = {
    zigbeeModel: [''],
    model: '',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-08-04:
The device failed to pair, try to re-pair it.

Elephantoplasty, 2024-08-04:
I've tried about 20 times.

Different batteries (alkaline and charged NiMH), moved to beside coordinator (SLZB-06), rebooted coordinator.

Always the same result.

MaxRower, 2024-08-06:
> I've tried about 20 times.
> 
> Different batteries (alkaline and charged NiMH), moved to beside coordinator (SLZB-06), rebooted coordinator.
> 
> Always the same result.

The TRETAKT plug itself should not have any batteries at all. Are sure sure, you don't mix up devices? Maybe it's the remote your talking about?

Koenkk, 2024-08-06:
> No matter what I did, always the same result - unsupported.

I expect it to be detected as unsupported, but does the interview also always fail? (from your db entry: `"interviewCompleted":false`)

Elephantoplasty, 2024-08-07:
Yes, interview completed comes up as false.

Koenkk, 2024-08-07:
could you provide the debug log of the interview?

Elephantoplasty, 2024-08-07:
I got a mate to have a go at the Tretakt, and he had it setup fine.  He is using the most recent version of Z2M.

Once I work out how to update Z2M in my Homeseer, I'll have a go.  Z2M is updated only moderately regularly by Homeseer, and is a couple of versions behind.  But there is a manual way of updating it posted somewhere......

Elephantoplasty, 2024-08-08:
Downloaded Z2M 1.39.1 - no change despite a few attempts.

I have debug logging level on under Advanced, but all I can find in the logs is below:

error 2024-08-08 11:35:15zh:controller: Interview failed for '0x94b216fffec684a3 with error 'Error: Interview failed because can not get node descriptor ('0x94b216fffec684a3')'
error 2024-08-08 11:35:15z2m: Failed to interview '0x94b216fffec684a3', device has not successfully been paired

Thanks for your help.

Dave

Elephantoplasty, 2024-08-08:
Ah, got it thanks.  Had set logging to debug level, but not to publish to front end....

Running Z2M version 1.39.1

Log had multiple attempts like below:
[2024-08-09 07:56:30] debug: 	zh:controller:device: Interview - quirks check for 'undefined'-'undefined'-'Unknown'
[2024-08-09 07:56:30] debug: 	zh:controller:device: Interview - quirks did not match
[2024-08-09 07:56:30] debug: 	zh:controller:device: Interview - node descriptor request failed for '0x94b216fffec684a3', attempt 1

Interview failed message is:
[2024-08-09 07:57:17] debug: 	zh:controller:device: Interview - failed for device '0x94b216fffec684a3' with error 'Error: Interview failed because can not get node descriptor ('0x94b216fffec684a3')
    at Device.interviewInternal (C:\Program Files (x86)\Homeseer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\device.ts:874:19)
    at Device.interview (C:\Program Files (x86)\Homeseer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\device.ts:755:13)
    at Controller.onDeviceJoined (C:\Program Files (x86)\Homeseer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:644:17)'

and:
[2024-08-09 07:57:17] error: 	zh:controller: Interview failed for '0x94b216fffec684a3 with error 'Error: Interview failed because can not get node descriptor ('0x94b216fffec684a3')'
[2024-08-09 07:57:17] error: 	z2m: Failed to interview '0x94b216fffec684a3', device has not successfully been paired

That is all I can find in the log file relating to the interview.

Thanks,

Dave

Koenkk, 2024-08-12:
It seems the device does not stay in the network for some reason. One thing that may help is to try to pair the device closer to the coordinator. Otherwise we need to start sniffing to see what is going on: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Elephantoplasty, 2024-08-12:
I've tried on multiple times to pair it right beside the coordinator - exactly the same result.   

I don't have the hardware or skills to undertake or understand the sniffing I'm afraid.

Thanks,  Dave

HZ-41, 2024-08-14:
@Elephantoplasty I got two of these today. For me they connected to Z2M in an instant however Z2M identifies it as E2204 even though the physical device states E2217. I turned it on and off a few times and it works (you can see that at the end of the logs below). Here is the output of the logs (sorry there is a bunch of data from my other devices, I dont know how to separate it):

```
2024-08-14 18:13:42z2m: Zigbee: allowing new devices to join.
info 2024-08-14 18:13:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"434k3-1"}'
info 2024-08-14 18:13:50z2m: Zigbee: disabling joining new devices.
info 2024-08-14 18:13:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"434k3-2"}'
info 2024-08-14 18:13:52z2m: Zigbee: allowing new devices to join.
info 2024-08-14 18:13:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"434k3-3"}'
info 2024-08-14 18:13:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Front Light 3', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":370,"linkquality":69,"power_on_behavior":null,"state":"ON","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":null}'
info 2024-08-14 18:14:03zh:controller: Interview for '0x94b216fffec685a2' started
info 2024-08-14 18:14:03z2m: Device '0x94b216fffec685a2' joined
info 2024-08-14 18:14:03z2m: Starting interview of '0x94b216fffec685a2'
info 2024-08-14 18:14:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x94b216fffec685a2","ieee_address":"0x94b216fffec685a2"},"type":"device_joined"}'
info 2024-08-14 18:14:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x94b216fffec685a2","ieee_address":"0x94b216fffec685a2","status":"started"},"type":"device_interview"}'
info 2024-08-14 18:14:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x94b216fffec685a2","ieee_address":"0x94b216fffec685a2"},"type":"device_announce"}'
info 2024-08-14 18:14:04zh:controller: Succesfully interviewed '0x94b216fffec685a2'
info 2024-08-14 18:14:04z2m: Successfully interviewed '0x94b216fffec685a2', device has successfully been paired
info 2024-08-14 18:14:04z2m: Device '0x94b216fffec685a2' is supported, identified as: IKEA TRETAKT smart plug (E2204)
info 2024-08-14 18:14:04z2m: Configuring '0x94b216fffec685a2'
info 2024-08-14 18:14:04z2m:mqtt: MQTT publish: topic 'homeassistant/switch/0x94b216fffec685a2/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x94b216fffec685a2/set","device":{"identifiers":["zigbee2mqtt_0x94b216fffec685a2"],"manufacturer":"IKEA","model":"TRETAKT smart plug (E2204)","name":"0x94b216fffec685a2","sw_version":"2.4.4","via_device":"zigbee2mqtt_bridge_0x00124b002e1ea10d"},"name":null,"object_id":"0x94b216fffec685a2","origin":{"name":"Zigbee2MQTT","sw":"1.39.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x94b216fffec685a2","unique_id":"0x94b216fffec685a2_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info 2024-08-14 18:14:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRETAKT smart plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"category":"config","description":"Controls the behavior when the device is powered on after power loss. If you get an `UNSUPPORTED_ATTRIBUTE` error, the device does not support it.","label":"Power-on behavior","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","on","toggle","previous"]},{"access":2,"category":"config","description":"Initiate device identification","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2204","options":[{"access":2,"description":"Sets the duration of the identification procedure in seconds (i.e., how long the device would flash).The value ranges from 1 to 30 seconds (default: 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x94b216fffec685a2","ieee_address":"0x94b216fffec685a2","status":"successful","supported":true},"type":"device_interview"}'
[... cut ...]
info 2024-08-14 18:15:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"434k3-4"}'
info 2024-08-14 18:16:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x94b216fffec685a2', payload '{"linkquality":153,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":33816580,"latest_version":33816613,"state":"available"},"update_available":null}'
info 2024-08-14 18:16:11z2m: Update available for '0x94b216fffec685a2'
```

Screenshot from Devices tab:

[image]

Screenshot of map:

[image]

psycoticninja, 2024-09-16:
Having same problem here - just purchased one of these not knowing it is a E2217 and I am seeing no attempt to connect... any ideas?

closed 2024-12-28 by github-actions[bot]