Koenkk/zigbee2mqtt #23506: [New device support]: Tuya Iluminance Sensor

runjmc, 2024-08-04:
### Link

https://expo.tuya.com/product/752043

### Database entry

{"id":87,"type":"Router","ieeeAddr":"0xa4c138914e7f06be","nwkAddr":57729,"manufId":4417,"manufName":"_TZE204_khx7nnka","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"HlA.iIlA.iIlA.iIlA.iJlA.i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_khx7nnka","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1722749646876}

### Comments

I tried to follow the supporting new device page I was able to see the device in the logs and it changes with the light change.    I could not figure out how to map the data points.

These are the 2 datapoints I saw in the logs
[2024-08-03 22:49:56] debug: 	zhc:tuya: Datapoint 2 not defined for '_TZE204_khx7nnka' with value 9
[2024-08-03 22:49:56] debug: 	zhc:tuya: Datapoint 1 not defined for '_TZE204_khx7nnka' with value 0

Device seems similar to their other Illuminance sensor but with a newer fingerprint?
{modelID: 'TS0601', manufacturerName: '_TZE200_yi4jtqq1'},
{modelID: 'TS0601', manufacturerName: '_TZE200_khx7nnka'},

Thanks for the assistance.  Let me know what other info is needed.  This is the first time I've tried add a new device for support

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_khx7nnka',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-08-04:
What you can try to do is stop z2, rename `_TZE200_khx7nnka` to `_TZE200_yi4jtqq1`, start z2m and see if it works now.

runjmc, 2024-08-04:
Which file should I be doing the change in?

Koenkk, 2024-08-05:
In the `data/database.db` file.

runjmc, 2024-08-06:
When I changed it to _TZE200_khx7nnka it started working.

Koenkk, 2024-08-06:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-06 by Koenkk