Koenkk/zigbee2mqtt #23511: Xiaomi LYWSD03MMC-z failed to configure an OTA does not work for all

z-master42, 2024-08-04:
### What happened?

The error message below appears in my Z2M logs for my [Xiaomi LYWSD03MMC-z](https://www.zigbee2mqtt.io/devices/LYWSD03MMC-z.htmlhttps://www.zigbee2mqtt.io/devices/LYWSD03MMC-z.html). As far as I can tell, however, they are working. Temperature and humidity are displayed.

I have seven of these devices in use.
Curiously, however, OTA does not work on all of them. [pvvx](https://github.com/pvvx/ZigbeeTLc/tree/master/bin) has now released version 0122, which could only be installed on one of the devices. There was a similar problem with the previous version 0121. This could only be installed on three devices. An update is offered for a fourth device, but this does not work. So I only have four devices with 0120, two devices with 0121 and one device with 0122.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Add-On on Home Assistant OS

### Debug log

z2m: Failed to configure 'TF Treppenflur', attempt 2 (Error: Bind 0xa4c138ffffd0b1da/1 msTemperatureMeasurement from '0x00212effff060f38/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:885:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:13) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:109:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:155:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:190:21) at Configure.configure (/app/lib/extension/configure.ts:126:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

closed 2024-08-04 by z-master42

hayvan96, 2024-09-06:
Hi, how have you solved this issue? 0.123 is out and I have the same problem.

z-master42, 2024-09-11:
Hello,

I have downloaded the correct and current [firmware version](https://github.com/pvvx/ZigbeeTLc/tree/master/bin) (.zigbee) and saved it in the `config/zigbee2mqtt/ota` directory.

At the end of the Z2M `configuration.yaml` I have added 
```
ota:
  zigbee_ota_override_index_location: my_index.json
```
supplemented.

And in the `config/zigbee2mqtt` folder is the `my_index.json` with the following content:
```
[
    {
        "url": "ota/1141-020a-01223001-Z03MMC.zigbee",
        "force": true
    }
]
```
here still with the 0122 version.

For some thermometers, a _new version is available_ message was sent directly. For others, I had to perform a manual check. This worked for all of them by removing the button cell, then reinserting it and clicking directly on the _Check for new updates_ button. A similar procedure was then followed when carrying out the firmware update.

When I looked at the OTA page in Z2M now, I just saw that new firmware is now available for all themometers. So the problem seems to solve itself afterwards.

hayvan96, 2024-09-11:
Thank you, it's useful. What's not clear for me : that bin directory clearly contains the 0.123 firmware, but when searching for OTA update, Z2M fails to find the new one. Does it require a converter/z2M update (will be include in next update) or is it supposed to check directly in pvvx's repository?

z-master42, 2024-09-11:
Yes, a Z2M update is required for this, as the update process only finds OTA updates that are in this file: https://github.com/Koenkk/zigbee-OTA/blob/master/index.json. The firmware files for this are then also in the Z2M Github.

hayvan96, 2024-09-11:
Thanks you, it's very clear! I'll wait for the OTA to be commited then

pavel-trubitsyn, 2025-07-15:
> Yes, a Z2M update is required for this, as the update process only finds OTA updates that are in this file: https://github.com/Koenkk/zigbee-OTA/blob/master/index.json. The firmware files for this are then also in the Z2M Github.

Could you describe how to update it. Still not clear for me. Thanks!

z-master42, 2025-07-15:
What exactly do you mean?

z-master42, 2025-07-15:
Which version of Z2M are you currently using?

pavel-trubitsyn, 2025-07-15:
Unfortunately, normal OTA does not work for this device

[image]
> Update of 'ДК2' failed (Device didn't respond to OTA request)

z-master42, 2025-07-15:
Did you remove and reinsert the battery and then immediately press the button?