Koenkk/zigbee2mqtt #23518: Update to 3.9 Problem

knozturk, 2024-08-04:
### What happened?

Every time I update to 3.9. All my current devices becomes unavailable and it creates second devices as _2 at the end. 

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230507

### Adapter

CC2652

### Setup

Add on to Home Assistant

Koenkk, 2024-08-04:
Try removing the `base_topic` from your Z2M HA addon configuration file.

knozturk, 2024-08-04:
Do I do this before the update or after? Because everytime I update, I recover from my backup. Currently, I am still on 3.8. Thanks,Kaan Ozturk

knozturk, 2024-08-05:
After removing base_topic, it has worked.

closed 2024-08-05 by knozturk