Koenkk/zigbee2mqtt #23522: Sonoff SNZB-03P always showing Battery Low as N/A

irata-coder, 2024-08-05:
### What happened?

"
[image]

[image]

"Battery low" therefore "Power" status is not supplied by the device

`{
    "battery": 100,
    "illumination": "bright",
    "last_seen": "2024-08-05T13:13:47+01:00",
    "linkquality": 248,
    "motion_timeout": 15,
    "occupancy": false,
    "update": {
        "installed_version": 8705,
        "latest_version": 8705,
        "state": "idle"
    },
    "voltage": 3100
}`

Therefore, power is always show in the frontend as ? on all front end screens.  Waiting 24 hours or more has no improvment.

### What did you expect to happen?

The battery percentage should be used to evaluate battery power and show correct battery icon based upon percentage.

### How to reproduce it (minimal and precise)

Add a Sonoff SNZB-03P, power is imediately shown as ? and remains so.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.3

### Adapter

Zigbee Dongle-E

### Setup

plain install on x86.64 debian

Koenkk, 2024-08-06:
This was a mistake in the definition, removed it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-06 by Koenkk

pezzy-o, 2024-09-19:
@irata-coder is this fixed for you? I’m still seeing a question mark for my battery state (although battery is reported as 100%).

irata-coder, 2024-09-19:
I'm seeing the same, the battery percentage is now reported correctly but the power source is showing as a question mark.  The device does not appear to report its powerSource, which can only be battery.

pezzy-o, 2024-09-19:
That's right - looks like powerSource isn't reported. Technically a different issue to the one you reported, but same outcome I think.

kingp0dd, 2024-11-01:
Are you guys getting `illumination` value from this device? Mine in Z2M always reports `dim` value.

pezzy-o, 2024-11-02:
> Are you guys getting `illumination` value from this device? Mine in Z2M always reports `dim` value.

Seems to be working for me. Note the comment on the config page: "Only updated when occupancy is detected"

kingp0dd, 2024-11-02:
Confirmed my 2nd unit is working, 1st one seems to be defective

pezzy-o, 2024-11-09:
> That's right - looks like powerSource isn't reported. Technically a different issue to the one you reported, but same outcome I think.

This has been raised in [https://github.com/Koenkk/zigbee-herdsman-converters/issues/8098](url)