Koenkk/zigbee2mqtt #23523: EARU EAKCB-T-M TS011F _TZ3000_lepzuhto DIN Power Meter - No Energy Consumption

VincenzoMaletta, 2024-08-05:
### What happened?

What happened?
I have 3 of these DIN power meters. They are paired and they show as TS011F _TZ3000_lepzuhto / [TS011F_plug_1].
They report Power, Current and Voltage but no Energy (appears as "null").

### What did you expect to happen?

Have energy metering.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240315

### Adapter

SMLight SLZB06

### Setup

Plain on docker container

### Debug log

info 2024-08-05 16:17:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ClimaDisimpegnoEnergy', payload '{"ac_frequency":null,"current":0.96,"device":{"applicationVersion":69,"dateCode":"","friendlyName":"ClimaDisimpegnoEnergy","hardwareVersion":1,"ieeeAddr":"0xdc8e95fffe81872e","manufacturerID":4098,"manufacturerName":"_TZ3000_lepzuhto","model":"EAKCB-T-M-Z","networkAddress":46457,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":null,"indicator_mode":"off/on","linkquality":87,"over_current_breaker":"ON","over_current_threshold":80,"over_voltage_breaker":"ON","over_voltage_threshold":265,"power":179,"power_apparent":null,"power_breaker":"ON","power_factor":null,"power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":36.8,"temperature_breaker":"ON","temperature_threshold":100,"under_voltage_breaker":"ON","under_voltage_threshold":90,"update":{"installed_version":69,"latest_version":69,"state":"idle"},"update_available":null,"voltage":235}'

Koenkk, 2024-08-06:
- Could you provide the debug log when trying to do a configure from the z2m frontend (yellow refresh icon)
- Does it provide a value when reading it from the dev console?

[image]

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

VincenzoMaletta, 2024-08-06:
I first tryied reading the value from the dev console and it gave back a value; at least i could see a message on the z2m console reporting some (hexadecimal ?) value.
Then I run the configure command from the z2m frontend and this activated the energy metering.
Now, if I try to read again from the dev console I can't get any message; the log reports this:

error 2024-08-07 00:04:31z2m: Publish 'set' 'read' to 'ClimaDisimpegnoEnergy' failed: 'Error: ZCL command 0xdc8e95fffe81872e/1 seMetering.read(["currentSummReceived"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Anyway the energy meter is working now.

Thank you for your kind assistance.

closed 2024-08-06 by VincenzoMaletta