Koenkk/zigbee2mqtt #23529: New device Zigbee tuya Za03 Smart siren alarm

AntonBeelen, 2024-08-06:
### Link

https://nl.aliexpress.com/item/1005007187593170.html?

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0xa4c1383f44db880e","nwkAddr":59906,"manufId":4417,"manufName":"_TZE204_hcxvyxa5","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000b\u0000\u0000\u0000f\f\u0000\u0000\u0000\u0012\f\u0000\u0000\u0000e","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_hcxvyxa5","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722929563200}

### Comments

When adding the device i got unsupported status, has the same features as NAS-AB02B2

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_hcxvyxa5',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

msherry, 2024-08-13:
I'm not the original reporter, but I have the same device, and I can confirm that making this change gives me an Exposes tab that lets me successfully turn the alarm on and off and change the melody.

AntonBeelen, 2024-08-13:
@Koenkk Thank you can confirm this works now.

languy99, 2024-08-15:
Yeah it shows up but I can’t get it to alert it anything. It throws errors when trying to configure the alarm.

alexboor, 2024-08-29:
> Can you, stop z2m, in the `data/database.db` change all occurrences of `_TZE204_hcxvyxa5` to `_TZE200_t1blo2bj`, start z2m and see if it works now?

Confirm the same issue.
And after the suggested changes, it works and is operable.

Koenkk, 2024-08-30:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-30 by Koenkk