Koenkk/zigbee2mqtt #23532: New Device: Zigbee Solar Rain Sensor

cloudbr34k84, 2024-08-06:
### Link

https://www.aliexpress.com/item/1005007269624052.html?spm=a2g0o.order_list.order_list_main.16.49731802i8AvVZ

### Database entry

{"id":172,"type":"EndDevice","ieeeAddr":"0xa4c1386b5af8658b","nwkAddr":28526,"manufId":4417,"manufName":"_TZ3210_tgvtvdoc","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,4,5,1,1280,61184],"outClusterList":[3,4,6,4096,10,25],"clusters":{"genBasic":{"attributes":{"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2ee669","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":148}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ee669","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":67,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":543112114},"lastSeen":1722943531629}

### Comments

When i pair the device its been picked up as 

- TS0207
- _TZ3210_tgvtvdoc
- Description: Water leak detector

I have tried to construct the converter, but its not sticking and im not sure if its because its already identified.

here is my converter

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const rbSrain01 = {
    fingerprint: [{modelID: 'RB-SRAIN01', manufacturerName: '_TZ3210_tgvtvdoc'}],
    model: 'RB-SRAIN01',
    vendor: 'Tuya',
    description: 'Solar Rain Sensor',
    fromZigbee: [fz.tuya_data_point],
    toZigbee: [],
    exposes: [
        e.binary('droplet_detection_state', ea.STATE, 'ON', 'OFF').withDescription('Droplet Detection State'),
        e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
        e.numeric('light_intensity', ea.STATE).withUnit('lx').withDescription('Light intensity'),
        e.numeric('average_light_intensity_20mins', ea.STATE).withUnit('lx').withDescription('20 mins average light intensity'),
        e.numeric('todays_max_light_intensity', ea.STATE).withUnit('lx').withDescription("Today's max light intensity"),
        e.binary('cleaning_reminder', ea.STATE, 'ON', 'OFF').withDescription('Cleaning reminder'),
        e.numeric('rain_sensor_voltage', ea.STATE).withUnit('V').withDescription('Rain sensor voltage'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'droplet_detection_state', tuya.valueConverter.trueFalse],
            [4, 'battery', tuya.valueConverter.batteryPercentage],
            [101, 'light_intensity', tuya.valueConverter.raw],
            [102, 'average_light_intensity_20mins', tuya.valueConverter.raw],
            [103, 'todays_max_light_intensity', tuya.valueConverter.raw],
            [104, 'cleaning_reminder', tuya.valueConverter.trueFalse],
            [105, 'rain_sensor_voltage', tuya.valueConverter.raw],
        ],
    },
    extend: [],
};

module.exports = rbSrain01;
```

### External definition

```shell
const {battery, iasZoneAlarm, commandsOnOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0207'],
    model: 'TS0207',
    vendor: '_TZ3210_tgvtvdoc',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), commandsOnOff()],
    meta: {},
};

module.exports = definition;
```

kiselio, 2024-08-09:
Here's a fully working converter for this device, [found here](https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/):

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0207',
            manufacturerName: '_TZ3210_tgvtvdoc',
        },
    ],
    model: 'TS0207_rain_sensor',
    vendor: 'Tuya',
    description: 'Solar Rain Sensor RB-SRAIN01',
    fromZigbee: [tuya.fz.datapoints, fz.battery, fz.ias_water_leak_alarm_1],
    toZigbee: [],
    exposes: [
        e.water_leak('raining').withDescription('Droplet Detection (Raining)'), 
        e.numeric('illuminance', ea.STATE).withUnit('mV').withDescription('Illuminance level'),
        e.numeric('illuminance_average_20min', ea.STATE).withUnit('mV').withDescription('Illuminance average for the last 20 minutes'),
        e.numeric('illuminance_maximum_today', ea.STATE).withUnit('mV').withDescription('Illuminance maximum for the last 24 hours'),
        e.binary('cleaning_reminder', ea.STATE, 'ON', 'OFF').withDescription('Cleaning reminder'),
        e.numeric('rain_intensity', ea.STATE).withUnit('mV').withDescription('Rainfall intensity'),
        e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
        e.battery_low(),        
    ],
    meta: {
        tuyaDatapoints: [
            [4, 'battery', tuya.valueConverter.batteryPercentage],
            [101, 'illuminance', tuya.valueConverter.raw],
            [102, 'illuminance_average_20min', tuya.valueConverter.raw],
            [103, 'illuminance_maximum_today', tuya.valueConverter.raw],
            [104, 'cleaning_reminder', tuya.valueConverter.trueFalse],
            [105, 'rain_intensity', tuya.valueConverter.raw],
        ],
    },
    extend: [],
};

module.exports = definition;`

cloudbr34k84, 2024-08-09:
Yeah I saw this yesterday as well

closed 2024-08-09 by cloudbr34k84

Koenkk, 2024-08-12:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `RB-SRAIN01.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

kiselio, 2024-08-12:
Added the image @Koenkk

BogdanNikolas, 2024-08-31:
Please, help me change the picture on this device in Z2M. I can't find clear instructions.

budcalloway, 2024-09-01:
Hi !
In my case, I do not have any illuminance reported. Still at 0 mV under sunlight. Anyone having this problem too ?

`{
    "battery": 73,
    "battery_low": false,
    "illuminance": 0,
    "illuminance_average_20min": 0,
    "illuminance_maximum_today": 11,
    "linkquality": 149,
    "rain_intensity": 4,
    "tamper": false,
    "water_leak": false
}`

rdorys, 2024-09-02:
It seems that water leak is missing since the last release,
https://github.com/Koenkk/zigbee2mqtt/issues/23678#issuecomment-2324770434

I created a PR to put in back in the converter. 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/7931

RubenKelevra, 2024-09-02:
> Please, help me change the picture on this device in Z2M. I can't find clear instructions.

Instructions are above your comment:

https://github.com/Koenkk/zigbee2mqtt/issues/23532#issuecomment-2283649487

rdorys, 2024-09-13:
Is this device stable on your network? Mine seems to stop sending data for several hours at a time until I move it significantly closer to a router. What’s odd is that I don’t have to re-pair it. Once I bring it near the router (within just a few centimeters), the sensor starts working normally again, sending data every minute as expected. After that, I can put it back in its original position, and it continues to function properly for a while, but then the issue comes back.

RubenKelevra, 2024-09-13:
Can you try covering the solar panel if that happens, before you move it, so we can rule out a low power situation?

rdorys, 2024-09-13:
Ok I will try it next time, but it really happens randomly. Note that my battery is still 97%, so I don't know if the device will trigger any low power behaviour.

RubenKelevra, 2024-09-13:
> Ok I will try it next time, but it really happens randomly. Note that my battery is still 97%, so I don't know if the device will trigger any low power behaviour.

Battery percentage reporting is non-optimal currently in Z2M, as most devices don't like usually considered pretty high voltage levels.

As this one got a rechargeable battery the conversion from voltage to percent might actually be completely wrong.

Would you mind graphing the battery voltage as well in HA to see if the battery just drops to a certain voltage when the device disappears?

Thanks!

RubenKelevra, 2024-09-13:
Ah okay. Doesn't sound like a power issue in that case. 🤔

Have you confirmed that the signal strength is fine? Maybe it just falls off the network because it gets wet and thus the signal strength drops?

Btw: Does the device report a battery voltage?

rdorys, 2024-09-13:
Thx for your time. The battery is reported but I needed to manually set the reporting in the dedicated tab.

c4m0-lab, 2024-09-23:
Hi :) 

I also get no battery informations 

[image]

is there any solution?

If I try to add it like that i get an error:
[image]

rdorys, 2024-09-23:
@c4m0-lab did you try to wake the sensor (clicking shortly the pairing button) just before sending the configuration?

c4m0-lab, 2024-09-23:
no but I will do it now :) 

it worked - YEAH THANKS!!!

knuti1960, 2024-09-23:
> did you try to wake the sensor (clicking shortly the pairing button) just before sending the configuration?

That helped me too, **thank you so much!**

aston668334, 2024-09-25:
> Is this device stable on your network? Mine seems to stop sending data for several hours at a time until I move it significantly closer to a router. What’s odd is that I don’t have to re-pair it. Once I bring it near the router (within just a few centimeters), the sensor starts working normally again, sending data every minute as expected. After that, I can put it back in its original position, and it continues to function properly for a while, but then the issue comes back.
> 
> Edit: It seems that it needs to reach directly the coordinator to be back again on the network (bringing it close to a router seemed to be close enought to send the info to the coordinator).

Same issue, is there any solution to fix it, thk.

rdorys, 2024-09-25:
@aston668334, I positioned a router close to a window. When the window is opened, sometime it starts to send data for a while again, but when the windows is closed, eventually it stops sending data. I didn’t have the time to sniff the network and I didn’t received a new antenna for my coordinator.

knuti1960, 2024-09-25:
In my case, it automatically connected to the nearest router ([https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1](url)), about 2.5 metres away, with a house wall in between. The connection is consistently good and stable. So far, I have only been using the device since yesterday.

aston668334, 2024-09-25:
> @aston668334, I positioned a router close to a window. When the window is opened, sometime it starts to send data for a while again, but when the windows is closed, eventually it stops sending data. I didn’t have the time to sniff the network and I didn’t received a new antenna for my coordinator.

Thk, I will try.

RubenKelevra, 2024-09-26:
> @aston668334, I positioned a router close to a window. When the window is opened, sometime it starts to send data for a while again, but when the windows is closed, eventually it stops sending data. I didn’t have the time to sniff the network and I didn’t received a new antenna for my coordinator.

Do you got fairly new windows?

Because some of the higher tier ones get a metal coating for better insulation, but this blocks radio frequencies from passing through, easily.

rdorys, 2024-09-26:
Yes I have, but I’m not going to change them for this device 😅😅

RubenKelevra, 2024-09-26:
@rdorys ha you don't have to. You just need to put a not-battery-powered zigbee device outside. For example a [zigbee module for switching lights](https://www.zigbee2mqtt.io/devices/ZBMINI.html#sonoff-zbmini), which can be placed inside a power outlet outside.

Another option are USB powered zigbee sticks or plugs, which are often called "Zigbee repeaters".

[Ikea](https://www.zigbee2mqtt.io/devices/E1746.html) got one for example.

Most non battery powered devices act as routers inside the zigbee network. You just need to have a router outside so the lamp can connect to it. 

The router outside will then transmit either through the wall or a window the signal to the next router. 

Routers are less focused on conserving power and thus can establish more reliable links.

Hope this helps :)

aston668334, 2024-09-27:
> Thx for your time. The battery is reported but I needed to manually set the reporting in the dedicated tab. [...]

After setting this up and moving my Zigbee coordinator near the window, the sensor has become more stable.

c4m0-lab, 2024-09-27:
I can confirm it. The main problem with stability for me was the distance from the nearest device (socket). I brought that one meter closer and it has stabilized, the breaks are no longer there.

RubenKelevra, 2024-09-27:
So I guess we should add some info about this on the wiki page, right @Koenkk? 🤔

rdorys, 2024-09-27:
And how to deal with the need of manually configuring the battery reporting for some users ?
Can we do it right into the converter ?

lluisd, 2024-09-27:
I have the same problem. I have it outside and it stops working after a while. If I move inside it stars working again. I have other zigbee aqara devices near that outside and they work perfectly.

RubenKelevra, 2024-09-28:
@lluisd is your Aqara device a router? 🤔

RubenKelevra, 2024-09-28:
> And how to deal with the need of manually configuring the battery reporting for some users ? Can we do it right into the converter ?

I think it makes sense to create a new bug for this, to explain the issue dedicated from this one, as the issue seems well defined

lluisd, 2024-09-28:
> @lluisd is your Aqara device a router? 🤔

no, but I will try to move a little bit to check if it has signal. The only difference between these 2 devices is that the aqara sensor it´s not behind the balcony wall where I placed the rain-solar sensor 

[image]

rdorys, 2025-01-20:
@marine1988 you are right it can be triggered by morning humidity.

marine1988, 2025-01-21:
Yes I tought that I never find a working sensor for rain that works

rdorys, 2025-01-21:
There is one https://shop.ecowitt.com/en-fr/products/wittboy that works in home assistant, but not for 15$ 😀

RubenKelevra, 2025-01-21:
> Yes I tought that I never find a working sensor for rain that works

I was kinda pragmatic 4 years ago and just put a leak detection sensor in the rain gutter. If there's more than just a few drops of rain it will trigger very reliably. :)

[image]

Clusters, 2025-04-21:
I want to detect if the sun is shining, or if there is overcast. But the sensor always show the same lx values in daylight, regardless the sun intensity. Sometimes the values are higher if there's an overcast. The values are around 5500lx to 6000lx. Which is way below any daylight situation.
Has someone else the same problem?