Koenkk/zigbee2mqtt #23542: Tuya is correct but lacking one more socket entity

reylinux, 2024-08-07:
### What happened?

Correct device however only able to control one socket instead of two. I already removed and re added this more than 3 times and still got the same result
[image]

### What did you expect to happen?

Previously was working fine with ZHA. It's detected with the same model - TS011F but doesn't allow me to control two sockets like I used to with ZHA.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add On with HA Green

crembz, 2024-08-09:
I am having the same issue with this device. Only one of the two plugs show in z2m

Koenkk, 2024-08-12:
Could you try to: stop z2m, in the `data/database.db` change `_TZ3000_qlmnxmac` to `_TZ3000_bep7ccew`, start z2m and check if it works?

reylinux, 2024-08-12:
Hi thanks for the reply.

It exposes correctly however I still can't get them to work as expected.

[image]

[image]

reylinux, 2024-08-13:
> Could you try to: stop z2m, in the `data/database.db` change `_TZ3000_qlmnxmac` to `_TZ3000_bep7ccew`, start z2m and check if it works?

Hi following this issue, I've changed the database to _TZ3000_8nyaanzb and it seems to be working ok now
[image]

Koenkk, 2024-08-15:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-15 by Koenkk

crembz, 2024-08-16:
> Hi following this issue, I've changed the database to _TZ3000_8nyaanzb and it seems to be working ok now [...]

I'm new to all this. Confirmed this worked for me last night. Can you please help me understand how you found what device code to use?

reylinux, 2024-08-16:
> Added, thanks!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Hi, thanks for the support. I was wondering if _TZ3000_8nyaanzb also supports power/energy monitoring just like _TZ3000_qlmnxmac? If I remember correctly, this device supports energy monitoring in ZHA (for both sockets)

[image]

reylinux, 2024-08-16:
> I'm new to all this. Confirmed this worked for me last night. Can you please help me understand how you found what device code to use?

Hi, I just got lucky as I have another device at home that was purchased from a different supplier and paired with _TZ3000_8nyaanzb. From there I then tried to change the _TZ3000_qlmnxmac  to this one and confirmed to work.

crembz, 2024-08-16:
> Hi, thanks for the support. I was wondering if _TZ3000_8nyaanzb also supports power/energy monitoring just like _TZ3000_qlmnxmac? If I remember correctly, this device supports energy monitoring in [...]

Just looking at it now in zha and it only reports current and power for the device not for each socket.

reylinux, 2024-08-16:
> Just looking at it now in zha and it only reports current and power for the device not for each socket.

Thanks for confirming. I don't have access for ZHA anymore. Yeah it's a shame otherwise it's good for tracking

crembz, 2024-08-22:
Just to clarify ... ZHA supports power monitoring for this device as a total (not per socket). Currently z2m does not seem to expose power monitoring at all

[image]

Any chance this can be added?

crembz, 2024-08-28:
@reylinux can you try the device code _TZ3000_dd8wwzcy. It seems to be correctly reporting the power values as well as control for both sockets

reylinux, 2024-08-30:
> @reylinux can you try the device code _TZ3000_dd8wwzcy. It seems to be correctly reporting the power values as well as control for both sockets

Hi Crembz thanks. How do you think you could figure out this time?