Koenkk/zigbee2mqtt #23543: New Tuya Smart Knob

reylinux, 2024-08-07:
### Link

https://a.aliexpress.com/_mLbCmeC

### Database entry

{"id":35,"type":"EndDevice","ieeeAddr":"0xa4c138f0669a8240","nwkAddr":12534,"manufId":4417,"manufName":"_TZ3000_6zvw8ham","powerSource":"Battery","modelId":"TS0203","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[3,4,5,6,8,4096,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0203","manufacturerName":"_TZ3000_6zvw8ham","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002b4888ee","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4888ee","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1463247413},"lastSeen":1723012706672}

### Comments

I believe this is a completely different device than the other one that is supported, link below. 

I tried to click the button and this came up. Please let me know if you need more details. I am very new with Z2M so I'm unsure what to provide.

[image]

[image]

https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html

### External definition

```shell
const {battery, identify, onOff, commandsOnOff, commandsLevelCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_abrsvsou',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), onOff({"powerOnBehavior":false}), commandsOnOff(), commandsLevelCtrl()],
    meta: {},
};

module.exports = definition;
```

reylinux, 2024-08-22:
Hello, is there any plan to integrate this one yet?

Barley194, 2024-10-25:
I got the same one  TS004F  _TZ3000_abrsvsou and it is wrongly detected as ERS-10TZBVK-AA. And it doesn't work as expected. 

changing the modes does nothing, it behaves as it should only in event mode. Command mode doesn't change the behavior

6PATyCb, 2024-11-07:
> I got the same one TS004F _TZ3000_abrsvsou and it is wrongly detected as ERS-10TZBVK-AA. And it doesn't work as expected.
> 
> changing the modes does nothing, it behaves as it should only in event mode. Command mode doesn't change the behavior

I got same too. You can switch the mode by triple clicking. I checked, it works for me.

jamiegalbreath, 2024-11-24:
Also have this problem

Scoopien, 2024-11-29:
Same here _TZ3000_402vrq2i not suported

djurny, 2024-11-29:
There is already some support in zigbee-herdsman-converters, but I know too little about how to get that to work when just using the  zigbee2mqtt docker image.
You can however, use an external converter, like the one below:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS004F',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_402vrq2i',
        },
    ],
    model: 'ERS-10TZBVK-AA',
    vendor: 'Tuya',
    description: 'Smart Knob Dimmer',
    fromZigbee: [
        fz.command_step,
        fz.command_toggle,
        fz.command_move_hue,
        fz.command_step_color_temperature,
        fz.command_stop_move_raw,
        fz.tuya_multi_action,
        fz.tuya_operation_mode,
        fz.battery,
        ],
    toZigbee: [
        tz.tuya_operation_mode,
        ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.action([
            'toggle',
            'brightness_step_up',
            'brightness_step_down',
            'color_temperature_step_up',
            'color_temperature_step_down',
            'saturation_move',
            'hue_move',
            'hue_stop',
            'single',
            'double',
            'hold',
            'rotate_left',
            'rotate_right']),
        e.numeric('action_step_size', ea.STATE).withValueMin(0).withValueMax(255),
        e.numeric('action_transition_time', ea.STATE).withUnit('s'),
        e.numeric('action_rate', ea.STATE).withValueMin(0).withValueMax(255),
        e.battery(),
        e
            .enum('operation_mode', ea.ALL, ['command', 'event'])
            .withDescription('Operation mode: "command" - for group control, "event" - for clicks'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```
Put the file in the same location as your `configuration.yaml`, add the following line to your `configuration.yaml` and restart z2m:
'''
external_converters:
  - Smart_Knob_Dimmer.js
'''

Do note that I do not really know if I did that correctly or not, but it did work for my [Smart Knob Dimmer](https://www.aliexpress.com/item/3256807534367914.html).

Groetjes,

Scoopien, 2024-11-29:
THANKS!!! First time it works perfectly!

jamiegalbreath, 2024-11-29:
This semi works for me! THANK YOU but double tap and hold does come up as an option?! Any ideas?

djurny, 2024-11-29:
> This semi works for me! THANK YOU but double tap and hold doesn't come up as an option?! Any ideas?

Hi @jamiegalbreath,
If you have the "Smart Knob Dimmer" device, you need to change the `operation mode` by quickly triple clicking the button. It will change from `command` to `event` mode. That mode should have a `double click` event.

Instead of doing the triple click, you can also use the z2m frontend, select your device in the `Devices` tab, then `Exposes` to change the `Operation mode` from `command` to `event`. 

Alternatively, you can search for your `Dining room Dimmer Operation mode` in HASS's `Settings`/`Devices and services`/`Entities` and change the value from `command` to `event`. 

If you change in either HASS or z2m, you must perform some button action, as the button has a banana in it's ear when at rest. It will receive/process the updated `command mode` after it has started to communicate.

Tap and hold _should_ work in the `command` mode as well, but you will get a `hue_move` event after holding the button for over 4 seconds. `hue_stop` will be emitted when releasing the button.

Groetjes,

djurny, 2024-12-02:
[1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) supports the [Smart Knob Dimmer](https://www.aliexpress.com/item/3256807534367914.html).
From the release notes:

- [#8257](https://github.com/Koenkk/zigbee-herdsman-converters/issues/8257) Detect _TZ3000_402vrq2i as Tuya ERS-10TZBVK-AA (@Velomeloman)

Remove the `Smart_Knob_Dimmer.js` file and remove the following from your `configuration.yaml` (+ z2m pull/upgrade/restart):
```
external_converters:
  - Smart_Knob_Dimmer.js
```
Groetjes,

closed 2025-02-09 by github-actions[bot]