Koenkk/zigbee2mqtt #23545: Trying to make sense out of Zemismart ZM16E-03/33 tubular motor

danielmmmm, 2024-08-07:
### What happened?

I am one of those people who bought a Zemismart tubular motor without a remote, thinking it could be controlled using Zigbee alone. Unfortunately, I can't seem to set the upper and lower border for my motor.

The manufacturer name is _TZE200_fzo2pocs and the model was set to ZM25TQ. I guess I got one of the many Zemismart products with a wrong identification name, since it is a ZM16E-03/33 (powered from USB, unlike the battery powered ZM16EL-03/33). Since I can't seem to get an external converter to make z2m treat the _TZE200_fzo2pocs as a ZM16E-03/33, I edited the zemismart.js directly, pasting the ZM16EL-03/33 code over the ZM25TQ code:

```
        fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_fzo2pocs' }],
        model: 'ZM25TQ',
        vendor: 'Zemismart',
        description: 'Tubular motor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        options: [exposes.options.invert_cover()],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.cover_position(),
            e.enum('reverse_direction', ea.STATE_SET, ['forward', 'back']).withDescription('Reverse the motor direction'),
            e.enum('border', ea.STATE_SET, ['up', 'down', 'up_delete', 'down_delete', 'remove_top_bottom']),
            e.enum('click_control', ea.STATE_SET, ['up', 'down']).withDescription('Single motor steps'),
            e.binary('motor_fault', ea.STATE, true, false),
        ],
        whiteLabel: [
            tuya.whitelabel('Zemismart', 'ZM16EL-03/33', 'Cover motor', ['_TZE200_fzo2pocs']),
        ],
        meta: {
            // All datapoints go in here
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverterBasic.lookup({ OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2) })],
                [2, 'position', tuya.valueConverter.coverPosition],
                [3, 'position', tuya.valueConverter.raw],
                [5, 'reverse_direction', tuya.valueConverterBasic.lookup({ forward: tuya.enum(0), back: tuya.enum(1) })],
                [12, 'motor_fault', tuya.valueConverter.trueFalse1],
                [13, 'battery', tuya.valueConverter.raw],
                [
                    16,
                    'border',
                    tuya.valueConverterBasic.lookup({
                        up: tuya.enum(0),
                        down: tuya.enum(1),
                        up_delete: tuya.enum(2),
                        down_delete: tuya.enum(3),
                        remove_top_bottom: tuya.enum(4),
                    }),
                ],
                [20, 'click_control', tuya.valueConverterBasic.lookup({ up: tuya.enum(0), down: tuya.enum(1) })],
            ],
        },
```

However, I still can't set the border limits or the position. Looking at the z2m output, I found it doesn't make much sense:

```
MQTT publish: topic 'zigbee2mqtt/0x94b216fffee4afd5', payload '{"border":"up","click_control":"up","last_seen":"2024-08-07T01:31:25+02:00","linkquality":182,"motor_fault":true,"position":12,"reverse_direction":"forward","state":"OPEN"}'
MQTT publish: topic 'zigbee2mqtt/0x94b216fffee4afd5', payload '{"border":"up","click_control":"up","last_seen":"2024-08-07T01:32:36+02:00","linkquality":185,"motor_fault":true,"position":12,"reverse_direction":"forward","state":"OPEN"}'
MQTT publish: topic 'zigbee2mqtt/0x94b216fffee4afd5', payload '{"border":"up","click_control":"up","last_seen":"2024-08-07T01:32:36+02:00","linkquality":185,"motor_fault":true,"position":12,"reverse_direction":"forward","state":"OPEN"}'
```

Firstly, clicking the open button in the z2m web interface triggers sending the payload three times. In fact, any action will send a payload three times in a row.
Secondly, and I suspect this is the real problem, the payload always contains all possible messages at once. Instead of just sending e.g. `"state":"OPEN"`, z2m sends `"border":"up","click_control":"up",...` too. It looks like just moving the roller blinds up or down will always set a new border, and therefore never effectively set any border.

I couldn't figure out how to send a payload through the web interface or how to do that using the PowerShell. But I managed to send messages using mosquitto_pub. Unfortunately, sending any message with mosquitto_pub will trigger the same z2m response of sending a full payload (three times). I tried `mosquitto_pub -t zigbee2mqtt/0x94b216fffee4afd5/set -m "{\"state\": \"STOP\"}"` and `mosquitto_pub -t zigbee2mqtt/0x94b216fffee4afd5/set/state -m "STOP"`, which both make the motor stop, but both make z2m send the full `{"border":"up","click_control":"up",...,"position":12,"reverse_direction":"forward","state":"OPEN"}` payload?

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20221226

### Adapter

zStack3x0

### Setup

plain z2m on Windows 10

cisasteelersfan, 2024-12-22:
I have the same problem. I then bought the RF 1-channel remote, and even with that I can't set the up and down limits. Well, I can "set" them (the motor beeps), but once set, it won't move up or down. "Inching" buttons don't work either.

srolskyi, 2025-01-22:
Hello everyone,

I have a model ZM16EL-03/33 with the same behavior, and yes, I didn’t buy the remote either. Why? Because a month ago, I bought this model without a remote, and it worked without any problems. So, I decided to buy a second one, and here we are…

After some time, I realized that the initial setup needs to be done differently. Since we don’t have proper instructions on how to set it up via z2m, here’s what I did:
	1.	I set the STATE in z2m to the 'STOP' state.
	2.	Then, using the Click Control, I adjusted the position with the UP/DOWN buttons.
	3.	Once I reached the correct top position, I set the Border to 'up', and the motor beeped.
	4.	Next, I adjusted the bottom position and set the Border to 'down', and the motor beeped.

[image]

danielmmmm, 2025-01-27:
How did you set up your motor without editing the _TZE200_fzo2pocs entry in zemismart.js?

I am pretty sure that my modification is incorrect. I see the correct controls, but e.g. the Click control doesn't work for me.

srolskyi, 2025-01-29:
> How did you set up your motor without editing the _TZE200_fzo2pocs entry in zemismart.js?
> 
> I am pretty sure that my modification is incorrect. I see the correct controls, but e.g. the Click control doesn't work for me.

So, for the first roller I received, I didn’t do any customization—I just used the initial configuration. I also had no issues with the second roller (I simply followed the steps I described in my post).

But then… yesterday, I received the third roller, and even my own instructions didn’t work!

After some time, I found an external converter:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoint = {
    ...tuya.tz.datapoints,
    key: ['state', 'position', 'reverse_direction', 'motor_fault', 'battery', 'border', 'click_control'],
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_68nvbio9',
        },
    ],
    model: 'ZM16EL-03/33',
    vendor: 'Zemismart',
    description: 'Smart tubular motor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoint],
    onEvent: tuya.onEventSetTime,
    options: [exposes.options.invert_cover()],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        e.cover_position(),
        exposes.composite('options', 'options', ea.STATE_SET),
        exposes
            .enum('reverse_direction', ea.STATE_SET, ['forward', 'back'])
            .withDescription('Reverse the motor direction'),
        exposes.enum('border', ea.STATE_SET, [
            'up',
            'down',
            'up_delete',
            'down_delete',
            'remove_top_bottom',
        ]),
        exposes
            .enum('click_control', ea.STATE_SET, ['up', 'down'])
            .withDescription('Single motor steps'),
        exposes.binary('fault_alarm', ea.STATE, true, false),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({ OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2) })],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],
            [5, 'reverse_direction', tuya.valueConverterBasic.lookup({ forward: tuya.enum(0), back: tuya.enum(1) })],
            [12, 'motor_fault', tuya.valueConverter.trueFalse1],
            [13, 'battery', tuya.valueConverter.raw],
            [
                16,
                'border',
                tuya.valueConverterBasic.lookup({
                    up: tuya.enum(0),
                    down: tuya.enum(1),
                    up_delete: tuya.enum(2),
                    down_delete: tuya.enum(3),
                    remove_top_bottom: tuya.enum(4),
                }),
            ],
            [20, 'click_control', tuya.valueConverterBasic.lookup({ up: tuya.enum(0), down: tuya.enum(1) })],
        ],
    },
};

module.exports = definition;`

I saved it in the following path:
/homeassistant/zigbee2mqtt/external_converters/zemismart.js
(not sure is that correct configuration, it look like bit different from the basic config, some of buttons won't work...)

After restarting Z2M, I was finally able to operate it using the ‘Control’ buttons.

No idea why… Three rollers, three different setup processes!

danielmmmm, 2025-01-29:
Thanks!

I had to update z2m from 1.42.0 to 2.0.0 to get the external converter to work. Unfortunately, the controls (up/down) don't do anything on my motor.
I guess these Zemismart motors are just garbage because the firmware seems to be random/wonky and they can't be updated or modified easily. For the price of 50 USD, I am very disappointed. Well well, I learned my lesson and am happy with my IKEA Fyrtur.

Still, glad that you contributed and that others might find this thread here helpful.

I guess I should close this issue?

srolskyi, 2025-01-29:
@danielmmmm,
Sure thing!

closed 2025-01-29 by danielmmmm

avdept, 2025-02-17:
> Hello everyone,
> 
> I have a model ZM16EL-03/33 with the same behavior, and yes, I didn’t buy the remote either. Why? Because a month ago, I bought this model without a remote, and it worked without any problems. So, I decided to buy a second one, and here we are…
> 
> After some time, I realized that the initial setup needs to be done differently. Since we don’t have proper instructions on how to set it up via z2m, here’s what I did: 1. I set the STATE in z2m to the 'STOP' state. 2. Then, using the Click Control, I adjusted the position with the UP/DOWN buttons. 3. Once I reached the correct top position, I set the Border to 'up', and the motor beeped. 4. Next, I adjusted the bottom position and set the Border to 'down', and the motor beeped.
> 
> <img alt="Image" width="995" src="https://private-user-images.githubusercontent.com/31102861/405539559-599fc940-0735-4b9f-abf1-8df8793acb9e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzk4MTEzOTEsIm5iZiI6MTczOTgxMTA5MSwicGF0aCI6Ii8zMTEwMjg2MS80MDU1Mzk1NTktNTk5ZmM5NDAtMDczNS00YjlmLWFiZjEtOGRmODc5M2FjYjllLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjE3VDE2NTEzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUxYzc0MDljM2UzY2MyM2E2Y2E1OGU5NjE5NDdiZGVkNDc3NjEzNjNhODBkNjAwOGNhN2E2MDI1NTIxOWZhMjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SApM0MjttmHmMAcLMCsjDwUUi4GXWaVWWF1_VcNaUBg">

Thanks for sharing, this saved my ass too. Ordered without control so it was a way to teach up/down positions