Koenkk/zigbee2mqtt #23551: [Feature request]: option to temporarily change policyDecision for TRUST_CENTER_JOIN_HANDLER

RuneRoven, 2024-08-07:
### Is your feature request related to a problem? Please describe

I tried to interview Nimly door lock multiple times but it failed upon TRUST_CENTER_JOIN_HANDLER
There is probably something in the firmware that not understand that the interview is done so it leaves the network and the device is removed from the database. To go around this i had to shut down the door lock just before it leaves so it is not removed from the database.
`zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=2603], [newNodeEui64=0xf4ce3603e9ecc8e3], [status=DEVICE_LEFT], [policyDecision=NO_ACTION], [parentOfNewNodeId=42643]`

### Describe the solution you'd like

It would be good to have an option to temporarily change the the TRUST_CENTER_JOIN_HANDLER or whatever mechanism that is responsible to not remove it from the database. 

### Describe alternatives you've considered

Maybe an option to immediately turn of allow join when a device is interviewed, this could perhaps also work 

### Additional context

-

Nerivec, 2024-08-11:
`ezspTrustCenterJoinHandler` is only feedback from the adapter's decision (according to Zigbee 3.0 specification), Z2M has no part in it, cannot change it. Also, the interview is Z2M-only and has no influence on that decision (the device is joined near immediately as far as the adapter is concerned, then it does the key exchange, then the adapter decides if the device should stay, the device can also decide to leave on its own after that due to internal firmware logic).

If the device is leaving, something is not going right, or the device is not compliant.

What Z2M/zigbee-herdsman version are you using?
Can you provide the logs surrounding that `ezspTrustCenterJoinHandler`?

RuneRoven, 2024-08-12:
Yes i understand that the leave message comes from the device. I updated to latest stable version to see if the problem persisted and it was the same issue.. i also understand that there is something in the device firmware that is not working properly and i have a support issue with the company. 
`Zigbee2MQTT version 1.39.1 (commit #e132316)
 zigbee-herdsman (0.55.3)
ember: ======== Ember Adapter Starting ========`
 I do not have the logs on the problem since i solved it and the pod is restarted since then. 
The problem i had was that after the interview the device was saved to the database but direct after it decided to leave it was removed from the database which caused the whole process to start over since the device decided to restart the interview. This kept repeating over and over again and i managed to solve it by timing the exact moment the device was stored in the database and removing the batteries. This way it was not deleted from the database and when the device started up again it was already paired and it accepted it..
The feature i would like to have was to temporarily disable the removal of the device from the database if it leaves direct after the interview..

Nerivec, 2024-10-01:
Keeping the device in the database on the Z2M side after it has left the network would result in an orphan device.
That kind of issue, solved by pulling the battery/power before the device has time to leave, works because it interrupts the logic on the device's side, so it stays after a restart.
If you can get a sniff and logs of the process, please open a new issue so we can see if a workaround is possible or not.

closed 2024-10-01 by Nerivec