Koenkk/zigbee2mqtt #23557: Aqara ZNJLBL01LM don´t show position anymore

matze20211, 2024-08-08:
### What happened?

Aqara ZNJLBL01LM don´t show position  anymore 

### What did you expect to happen?

until one of the lasts updates, the position was shown from 0 - 100%

### How to reproduce it (minimal and precise)

changing position slider will not initiate any message about position to mqtt server

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

PI

avehome, 2024-08-08:
Hi, I have exactly the same. Battery status etc. get all updated, but the device does not respond to open/close.

[image]

Cameleon99, 2024-08-16:
Same here, not sure what update changed it, position is always reported as 0 now. Previously it would be reported as 0 after moving, but and if you did a manual GET on the MQTT topic with payload below, this would force the current position to be refreshed, but this no longer appears to do anything. 

GET payload from NodeRed - {"topic":"zigbee2mqtt/Office Blind/get","payload":"{\"position\": \"\"}"}

Also when you hit the 'refresh' button on Position the Z2M devices GUI under the Exposes tab, I can't see a message in the log, nor on the queue using MQTT Explorer. But if I change the position via the slider, a message appears in the log and MTQQ explorer. 

Hitting the 'Battery' refresh icon causes a message in the log and on MQTT Explorer. Hitting 'State' does nothing. 

So something is happening specially with the position reporting and state? Setting position by the slider and calling open/close make the blind move, so they work.

Cameleon99, 2024-08-17:
Does this help from the logs? When I call position (this used to work)

[2024-08-17 15:59:01] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Blind/get' with data '{"position":""}'
[2024-08-17 15:59:01] debug: 	z2m: Publishing get 'get' 'position' to 'Kitchen Blind'
[2024-08-17 15:59:01] debug: 	zhc:lumi: Not reading position of ZNJLBL01LM since device doesn't support it

Having a dig around, I found that if I manually read via the dev console - 'genAnalogOutput' : 'presentValue' I get this in the logs and the position value is updated. Forgive me if there are extra messages mixed in, I'm not 100% sure what I'm doing here! 

2024-08-17 16:25:13] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Blind/1/set' with data '{"read":{"attributes":["presentValue"],"cluster":"genAnalogOutput","options":{}}}'
[2024-08-17 16:25:13] debug: 	z2m: Publishing 'set' 'read' to 'Kitchen Blind'
[2024-08-17 16:25:13] debug: 	zh:controller:endpoint: ZCL command 0x54ef4410006bb58d/1 genAnalogOutput.read(["presentValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-08-17 16:25:13] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x54ef4410006bb58d:47304/1 (0,0,2)
[2024-08-17 16:25:13] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":47304,"destendpoint":1,"srcendpoint":1,"clusterid":13,"transid":149,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,55,0,85,0]}}
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,200,184,1,1,13,0,149,0,30,5,16,55,0,85,0,171]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-08-17 16:25:13] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,149,83]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,149,83]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,149] - 83
[2024-08-17 16:25:13] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":149}
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,32,0,7,151,35,1,0,123,0,10,121,131,0,0,5,24,54,11,0,0,166,104,28,55]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,32,0,7,151,35,1,0,123,0,10,121,131,0,0,5,24,54,11,0,0,166,104,28,55]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,32,0,7,151,35,1,0,123,0,10,121,131,0,0,5,24,54,11,0,0,166,104,28] - 55
[2024-08-17 16:25:13] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":32,"srcaddr":38663,"srcendpoint":35,"dstendpoint":1,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":8616202,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,54,11,0,0]}}
[2024-08-17 16:25:13] debug: 	zh:controller: Received payload: clusterID=32, address=38663, groupID=0, endpoint=35, destinationEndpoint=1, wasBroadcast=false, linkQuality=123, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":54,"commandIdentifier":11},"payload":{"cmdId":0,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: <-- [254,31,68,129,0,0,13,0,200,184,1,1,0,132,0,51,124,131,0,0,11,24,55,1,85,0,0,57,0,0,200,66,187,107,28,224]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,31,68,129,0,0,13,0,200,184,1,1,0,132,0,51,124,131,0,0,11,24,55,1,85,0,0,57,0,0,200,66,187,107,28,224]
[2024-08-17 16:25:13] debug: 	zh:zstack:unpi:parser: --> parsed 31 - 2 - 4 - 129 - [0,0,13,0,200,184,1,1,0,132,0,51,124,131,0,0,11,24,55,1,85,0,0,57,0,0,200,66,187,107,28] - 224
[2024-08-17 16:25:13] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":13,"srcaddr":47304,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":132,"securityuse":0,"timestamp":8617011,"transseqnumber":0,"len":11,"data":{"type":"Buffer","data":[24,55,1,85,0,0,57,0,0,200,66]}}
[2024-08-17 16:25:13] debug: 	zh:controller: Received payload: clusterID=13, address=47304, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=132, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":55,"commandIdentifier":1},"payload":[{"attrId":85,"status":0,"dataType":57,"attrData":100}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusE [... cut ...]
[2024-08-17 16:25:13] debug: 	z2m: Received Zigbee message from 'Kitchen Blind', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":100}' from endpoint 1 with groupID 0
[2024-08-17 16:25:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Blind', payload '{"battery":88,"charging_status":false,"device_temperature":null,"last_seen":"2024-08-17T15:25:13.516Z","linkquality":132,"motor_speed":"high","motor_state":null,"position":100,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
[2024-08-17 16:25:13] info: 	zhc:tz: Read result of 'genAnalogOutput': {"presentValue":100}

Cameleon99, 2024-08-17:
I've also noticed that when changing the position MQTT is not reporting updates to the state, it is always STOP and always has the position as the old position. When I call 'genAnalogOutput' : 'presentValue' from the dev console, I then get the correct position in the logs. 

I can see here - https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/lumi.ts#L4699 - that ZNJLBL01LM has been excluded from polling the position and instead waits for the STOP event, but I don't believe that is coming for some reason and thus position is not updated. 

It's also where the position call is erroring: "Not reading position of ZNJLBL01LM since device doesn't support it" - from here - https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/lumi.ts#L4751 

This may have caused it somehow or be related: https://github.com/Koenkk/zigbee2mqtt/issues/23056

@Koenkk can you help?

Koenkk, 2024-08-18:
Could you provide the debug log when changing the position and it doesn't update?

Cameleon99, 2024-08-18:
@Koenkk - Thanks for looking. Sorry this seems to have a lot of other noise in it but left it so you can see all the messages. I took the blind from ~50% to fully open (100%). You can see in the logs the request to move to position 100 but later a message with position 77 which is what it shows in HA. 

[z2m_log.txt](https://github.com/user-attachments/files/16650312/z2m_log.txt)

Cameleon99, 2024-08-20:
I don't think they ever has worked properly, I've never seen it update the status, or give updates to position. But the manual call to get position on MQTT would work as a workaround. What I was doing was catching the events in NodeRed from HA that were calling the blind to change position or open/close and for the next minute call get position every few seconds so that it looked like it was being updated automatically. With some recent change, we are now getting the error 'zhc:lumi: Not reading position of ZNJLBL01LM since device doesn't support it' which I can see in the code as linked above. I don't know when it happened as largely the blinds are fully automated, I have a light sensor outside and the all open/close with light levels. I just noticed recently that sometimes they wouldn't open or close automatically, and assumed it was an issue with Zigbee network or my coding, but then realised when I manually called get position, it was not returning anything any more. And I checked the position before opening/closing them which when they reported they were already open/closed, it meant the code did nothing. For now I've removed that check (it was there because some old blinds I had if they were open and you tried to open them, they would go full circle and try to close partially again, but these Sonoff ones don't do that)

Maybe there is something else wrong with my setup that means I can't get the state? I'm hesitant to try and update the F/W on the Zigbee stick as everything else works and calling from the dev console to fetch the position also works. Seems its been artificially restricted for some reason? Its a Sonoff ZBDongle-P (CC2652P) as far as I can tell, bought in May 2022.

Koenkk, 2024-08-20:
Can you see if reconfiguring the device via the z2m frontend fixes the issue? (yellow refresh button)

matze20211, 2024-08-20:
I m not shure witch old  z2m worked, i will test it too.

matze

Cameleon99, 2024-08-20:
> Can you see if reconfiguring the device via the z2m frontend fixes the issue? (yellow refresh button)

As requested but hasn't made any difference. 

Below, I run 3 commands (nothing in between), reconfigure, then tell it to open to 50 (which only puts one line in the log when I request it and at position 100), then after its finished moving (and no update), I did a read of  'genAnalogOutput' : 'presentValue' on the dev console and then it responds with 51% 

info 2024-08-20 20:51:44z2m: Successfully configured 'Kitchen Blind'
info 2024-08-20 20:51:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Kitchen Blind"},"status":"ok","transaction":"k4jtk-1"}'

info 2024-08-20 20:52:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Blind', payload '{"battery":84,"charging_status":false,"device_temperature":null,"last_seen":"2024-08-20T19:52:04.618Z","linkquality":126,"motor_speed":"high","motor_state":null,"position":100,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'

info 2024-08-20 20:52:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Blind', payload '{"battery":84,"charging_status":false,"device_temperature":null,"last_seen":"2024-08-20T19:52:59.071Z","linkquality":126,"motor_speed":"high","motor_state":null,"position":51,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'

So it can get the position but we get no update on it moving, state is always STOP and running is null. 

I also tried to go from position 51 to 0 and repeatedly called read of 'genAnalogOutput' : 'presentValue' from Dev Console while it was moving, the logs show the position go from 51 to 0 with no steps and keeps reporting position 0 as its still moving, state is STOP and running is null.

Cameleon99, 2024-08-21:
Is it worth me upgrading the f/w? Been researching it a bit and for some reason I thought I needed a serial connection for it and a lot of poking about but it seems quite easy to just dismantle the stick, connect it to USB while holding the flash button and run the s/w to flash it. 

Seems there are multiple options, what's the easiest/least error prone for CC2652P?

ikeros, 2024-08-21:
Similar problems here. Rollers can be controller, opened and closed as expected, either via frontend or MQTT, no problems there. But all rollers seem to be stuck to the same position as they were before update.

I have similar experience with @Cameleon99. Position reporting has never worked properly. I was also using a workaround and called get state to expose new final location after 5 seconds delay to moving event. I have noticed some "random" state updates (once a day), but you can not rely on them, because it's not happening every time when roller is moving or stopped moving.

I just did an upgrade to the latest version last week. My previous backup was dated 16.6.2023 so probably version 1.31.0 (2023-06-01) was working correctly (with my workaround). Haven't tested anything between.

matze20211, 2024-08-21:
The position reporting has worked for me before, i don t  know witch version.

Koenkk, 2024-08-22:
Could you try to bind all clusters to the coordinator and see if that helps? It can be done from the frontend, example: 

[image]

matze20211, 2024-08-23:
No change, still no Position mrssage.

Cameleon99, 2024-08-23:
Same here, no change. I have 3 of them and they all behave the same.

Koenkk, 2024-08-26:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

Cameleon99, 2024-08-26:
Tried it and no change I'm afraid. 

I couldn't see custom in the name, so took a look in the converter file and saw the description didn't include it, so added it to check it was being loaded and it is. Is the linked definitely for the custom converter?

matze20211, 2024-08-27:
here the same

Koenkk, 2024-08-28:
I updated the description, can you check if it is correctly loaded now?

Cameleon99, 2024-08-28:
Correctly loaded with CUSTOM in description. No change in behaviour or logs, log only updates (in debug) on call to change position/open/close, no update to position. 

What should we be seeing?

Koenkk, 2024-08-30:
Made a mistake, updated https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24, with this the `position` should immediately update when you set it.

Cameleon99, 2024-08-30:
Tried the new one and getting errors:

> [2024-08-30 20:25:49] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Blind/set' with data '{ "position": 11 }'
> [2024-08-30 20:25:49] debug: 	z2m: Publishing 'set' 'position' to 'Kitchen Blind'
> [2024-08-30 20:25:49] error: 	z2m: Publish 'set' 'position' to 'Kitchen Blind' failed: 'ReferenceError: isString is not defined'
> [2024-08-30 20:25:49] debug: 	z2m: ReferenceError: isString is not defined
>     at Object.convertSet (/app/data/extension/externally-loaded.js:48:17)
>     at Publish.onMQTTMessage (/app/lib/extension/publish.ts:277:52)
>     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
>     at EventEmitter.emit (node:events:529:35)
>     at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
>     at MQTT.onMessage (/app/lib/mqtt.ts:147:27)
>     at MqttClient.emit (node:events:517:28)
>     at handlePublish (/app/node_modules/mqtt/src/lib/handlers/publish.ts:172:11)
>     at handle (/app/node_modules/mqtt/src/lib/handlers/index.ts:31:17)
>     at work (/app/node_modules/mqtt/src/lib/client.ts:762:17)
> [2024-08-30 20:25:49] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'position' to 'Kitchen Blind' failed: 'ReferenceError: isString is not defined'","meta":{"friendly_name":"Kitchen Blind"},"type":"zigbee_publish_error"}'

Cameleon99, 2024-08-30:
@Koenkk any ideas why I'm getting the error above?

matze20211, 2024-09-01:
i dont have these error, but the position is not updated with the new external converter

Koenkk, 2024-09-02:
@Cameleon99 updated https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24, can you try again?

Cameleon99, 2024-09-02:
> @Cameleon99 updated https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24, can you try again?

Thanks, solved the previous error but new one snuck in:

> z2m: Publish 'get' 'position' to 'Office Blind' failed: **'ReferenceError: NS is not defined'**
> z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'position' to 'Office Blind' failed: **'ReferenceError: NS is not defined**'","meta":{"friendly_name":"Office Blind"},"type":"zigbee_publish_error"}'

Cameleon99, 2024-09-03:
Did you put the file in the same directory as the Z2M config? Does it now say CUSTOM in the description? Seems like the custom  converter hasn't been loaded.

Cameleon99, 2024-09-03:
@Koenkk - got another error when calling OPEN or CLOSE (have been calling it with move to certain position so far)

> z2m: Publish 'set' 'state' to 'Kitchen Blind' failed: 'ReferenceError: getFromLookup is not defined'

matze20211, 2024-09-03:
Found my failure, CUSTOM is now shown. same eroor:

> [2024-09-03 15:52:36] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/set' with data '{"state":"OPEN"}'
[2024-09-03 15:52:36] debug:    z2m: Publishing 'set' 'state' to 'Rollo'
[2024-09-03 15:52:36] error:    z2m: Publish 'set' 'state' to 'Rollo' failed: 'ReferenceError: getFromLookup is not defined'
[2024-09-03 15:52:36] debug:    z2m: ReferenceError: getFromLookup is not defined
    at Object.convertSet (/opt/zigbee2mqtt/data/extension/externally-loaded.js:49:21)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:278:52)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:147:27)
    at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:125:27)
    at WebSocket.emit (node:events:519:28)
    at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:519:28)

Minek124, 2024-09-04:
Hello, I dig a little in this code https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24 and found soulution. Replace convertGet function with with this:
convertGet: async (entity, key, meta) => {
            if (!Array.isArray(meta.mapped) && ['ZNCLBL01LM'].includes(meta.mapped.model)) {
                await entity.read('closuresWindowCovering', ['currentPositionLiftPercentage']);
            } else {
                await entity.read('genAnalogOutput', [0x0055]);
            }
        },

Now position is set correctly if it is set via home assistant controls, but you need to refresh it manualy if position is set via device physical buttons.

matze20211, 2024-09-04:
no change , same error:

> 2024-09-04 14:54:40] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/set' with data '{">
[2024-09-04 14:54:40] debug:    z2m: Publishing 'set' 'state' to 'Rollo'
[2024-09-04 14:54:40] error:    z2m: Publish 'set' 'state' to 'Rollo' failed: 'ReferenceError: getFromLo>
[2024-09-04 14:54:40] debug:    z2m: ReferenceError: getFromLookup is not defined
    at Object.convertSet (/opt/zigbee2mqtt/data/extension/externally-loaded.js:49:21)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:278:52)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:147:27)
    at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:125:27)
    at WebSocket.emit (node:events:519:28)
    at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:519:28)

matze20211, 2024-09-04:
because of using the rollo, i removed 

> external_converters:
  - ext_converter.js

from configuration.yaml and deleted ext_converter.js

Restart z2m, and still "CUSTOM" and the error:

> [2024-09-04 15:03:50] debug:    z2m: Publishing 'set' 'state' to 'Rollo'
[2024-09-04 15:03:50] error:    z2m: Publish 'set' 'state' to 'Rollo' failed: 'ReferenceError: getFromLo>
[2024-09-04 15:03:50] debug:    z2m: ReferenceError: getFromLookup is not defined
    at Object.convertSet (/opt/zigbee2mqtt/data/extension/externally-loaded.js:49:21)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:278:52)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:147:27)
    at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:125:27)
    at WebSocket.emit (node:events:519:28)
    at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:519:28)

Minek124, 2024-09-04:
Removing 
`external_converters:
  - ext_converter.js
`
will go back to default version.

I would like to mention this code from https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24:
`
if (!Array.isArray(meta.mapped) && 'ZNJLBL01LM' == meta.mapped.model) {
    // https://github.com/Koenkk/zigbee2mqtt/issues/23056
    logger.debug(`Not reading position of ZNJLBL01LM since device doesn't support it`, NS);
} 
`
has bug because NS is not defined and it throws error and also it makes the position to not update.

matze20211, 2024-09-04:
> because of using the rollo, i removed
> 
> > external_converters:
> 
>     * ext_converter.js
> 
> 
> from configuration.yaml and deleted ext_converter.js
> 
> Restart z2m, and still "CUSTOM" in device description and the error:
> 
> > [2024-09-04 15:03:50] debug:    z2m: Publishing 'set' 'state' to 'Rollo'
> > [2024-09-04 15:03:50] error:    z2m: Publish 'set' 'state' to 'Rollo' failed: 'ReferenceError: getFromLo>
> > [2024-09-04 15:03:50] debug:    z2m: ReferenceError: getFromLookup is not defined
> > at Object.convertSet (/opt/zigbee2mqtt/data/extension/externally-loaded.js:49:21)
> > at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:278:52)
> > at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23)
> > at EventEmitter.emit (node:events:531:35)
> > at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22)
> > at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:147:27)
> > at WebSocket. (/opt/zigbee2mqtt/lib/extension/frontend.ts:125:27)
> > at WebSocket.emit (node:events:519:28)
> > at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1220:20)
> > at Receiver.emit (node:events:519:28)
> 
> How to go back to working version, without position message?

restart pi solved this

Koenkk, 2024-09-04:
Updated https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24 , can you try again?

matze20211, 2024-09-05:
now, no error, position jumps to "0" or "100", depends on "close" or "open", no position between will shown. 

I think version 1.36 was one in which the position information worked.

Here the log with new external converter

> [2024-09-05 07:32:57] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/set' with data '{"state": "OPEN"}'
[2024-09-05 07:32:57] debug:    z2m: Publishing 'set' 'state' to 'Rollo'
[2024-09-05 07:32:57] debug:    zh:controller:endpoint: ZCL command 0x54ef4410004454bb/1 genAnalogOutput.write({"presentValue":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultRes>
[2024-09-05 07:32:57] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x54ef4410004454bb:48906/1 (0,0,2)
[2024-09-05 07:32:57] debug:    zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":48906,"destendpoint":1,"srcendpoint":1,"clusterid":13,"transid":42,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data>
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:writer: --> frame [254,20,36,1,10,191,1,1,13,0,42,0,30,10,16,11,2,85,0,57,0,0,200,66,72]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-09-05 07:32:57] debug:    zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,42,236]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,42,236]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,42] - 236
[2024-09-05 07:32:57] debug:    zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":42}
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,41,239]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,41,239]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,41] - 239
[2024-09-05 07:32:57] debug:    zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":41}
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,13,0,10,191,1,1,0,105,0,170,164,229,0,0,4,24,11,4,0,62,48,28,230]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,13,0,10,191,1,1,0,105,0,170,164,229,0,0,4,24,11,4,0,62,48,28,230]
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,13,0,10,191,1,1,0,105,0,170,164,229,0,0,4,24,11,4,0,62,48,28] - 230
[2024-09-05 07:32:57] debug:    zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":13,"srcaddr":48906,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":1>
[2024-09-05 07:32:57] debug:    zh:controller: Received payload: clusterID=13, address=48906, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=105, frame={"header":{"frameControl":{"frame>
[2024-09-05 07:32:57] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:32:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqDD/Rollo', payload '{"battery":100,"charging_status":true,"linkquality":105,"motor_speed":"low","position":100,"state":"STOP","update":{"install>

matze20211, 2024-09-05:
try to set pct to 47 manually (set position from FHEM isn't working anymore, only "open","close" and "stop"):

> [2024-09-05 07:46:21] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/set' with data '{"position":47}'
[2024-09-05 07:46:21] debug:    z2m: Publishing 'set' 'position' to 'Rollo'
[2024-09-05 07:46:21] debug:    zh:controller:endpoint: ZCL command 0x54ef4410004454bb/1 genAnalogOutput.write({"presentValue":47}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResp>
[2024-09-05 07:46:21] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x54ef4410004454bb:48906/1 (0,0,1)
[2024-09-05 07:46:21] debug:    zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":48906,"destendpoint":1,"srcendpoint":1,"clusterid":13,"transid":182,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","dat>
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:writer: --> frame [254,20,36,1,10,191,1,1,13,0,182,0,30,10,16,43,2,85,0,57,0,0,60,66,0]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-09-05 07:46:21] debug:    zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,182,112]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,182,112]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,182] - 112
[2024-09-05 07:46:21] debug:    zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":182}
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,13,0,10,191,1,1,0,109,0,142,140,36,0,0,4,24,43,4,0,62,48,28,15]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,13,0,10,191,1,1,0,109,0,142,140,36,0,0,4,24,43,4,0,62,48,28,15]
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,13,0,10,191,1,1,0,109,0,142,140,36,0,0,4,24,43,4,0,62,48,28] - 15
[2024-09-05 07:46:21] debug:    zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":13,"srcaddr":48906,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":109,"securityuse":0,"timestamp":2>
[2024-09-05 07:46:21] debug:    zh:controller: Received payload: clusterID=13, address=48906, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=109, frame={"header":{"frameControl":{"frame>
[2024-09-05 07:46:21] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-09-05 07:46:21] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqDD/Rollo', payload '{"battery":100,"charging_status":true,"linkquality":109,"motor_speed":"low","position":47,"state":"STOP","update":{"installe>
[2024-09-05 07:46:24] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/get' with data '{"position":""}'
[2024-09-05 07:46:24] debug:    z2m: Publishing get 'get' 'position' to 'Rollo'
[2024-09-05 07:46:24] error:    z2m: Publish 'get' 'position' to 'Rollo' failed: 'ReferenceError: NS is not defined'
[2024-09-05 07:46:24] debug:    z2m: ReferenceError: NS is not defined
    at Object.convertGet (/opt/zigbee2mqtt/data/extension/externally-loaded.js:72:100)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:305:37)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:147:27)
    at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:125:27)
    at WebSocket.emit (node:events:519:28)
    at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:519:28)
[2024-09-05 07:46:24] debug:    zh:zstack:unpi:parser: <-- [254,12,69,202,78,249,159,19,0,52,0,188,21,0,66,169,206]
[2024-09-05 07:46:24] debug:    zh:zstack:unpi:parser: --- parseNext [254,12,69,202,78,249,159,19,0,52,0,188,21,0,66,169,206]
[2024-09-05 07:46:24] debug:    zh:zstack:unpi:parser: --> parsed 12 - 2 - 5 - 202 - [78,249,159,19,0,52,0,188,21,0,66,169] - 206
[2024-09-05 07:46:24] debug:    zh:zstack:znp: AREQ: <-- ZDO - tcDeviceInd - {"nwkaddr":63822,"extaddr":"0x0015bc003400139f","parentaddr":43330}

Cameleon99, 2024-09-05:
> Updated https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24 , can you try again?

This now works for me, with one exception, the code below errors if you try to call to get the position, saying NS 

```
if (!Array.isArray(meta.mapped) && 'ZNJLBL01LM' == meta.mapped.model) {
    // https://github.com/Koenkk/zigbee2mqtt/issues/23056
    logger.debug('Not reading position of ZNJLBL01LM since device doesn't support it', NS);
}  
```

``` 
Error: z2m: Publish 'get' 'position' to 'Office Blind' failed: 'ReferenceError: NS is not defined'
```

Koenkk, 2024-09-05:
Fixed the `NS` error, please try again: https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24

Cameleon99, 2024-09-05:
No more errors, but its still not quite right compared to the manual work around. 

If I move to a position, it correctly updates to that position and reports it, but if I 'close' from 'open' and then 'stop' it, the position shows as 'closed'/0, which it is not. 

When we used to be able to do a get/position, it would report the position. In the instance above that would update it to the correct position. 

If I comment out the new(ish) code, the workaround works again: 

```
        convertGet: async (entity, key, meta) => {
            //if (!Array.isArray(meta.mapped) && 'ZNJLBL01LM' == meta.mapped.model) {
            //    // https://github.com/Koenkk/zigbee2mqtt/issues/23056
            //    logger.debug(`Not reading position of ZNJLBL01LM since device doesn't support it`, NS);
            //} else 
            if (!Array.isArray(meta.mapped) && ['ZNCLBL01LM'].includes(meta.mapped.model)) {
                await entity.read('closuresWindowCovering', ['currentPositionLiftPercentage']);
            } else {
                await entity.read('genAnalogOutput', [0x0055]);
            }
        },
```

matze20211, 2024-09-05:
If i change position "50" via Mqtt (FHEM), this happens:

> [2024-09-05 15:35:26] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/set' with data '{"position": "50"}'
[2024-09-05 15:35:26] debug:    z2m: Publishing 'set' 'position' to 'Rollo'
[2024-09-05 15:35:26] error:    z2m: Publish 'set' 'position' to 'Rollo' failed: 'Error: Value: '50' not found in: [open, close, on, off]'
[2024-09-05 15:35:26] debug:    z2m: Error: Value: '50' not found in: [open, close, on, off]
    at Object.getFromLookup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:668:15)
    at Object.convertSet (/opt/zigbee2mqtt/data/extension/externally-loaded.js:50:35)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:278:52)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:147:27)
    at MqttClient.emit (node:events:519:28)
    at handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/src/lib/handlers/publish.ts:172:11)
    at handle (/opt/zigbee2mqtt/node_modules/mqtt/src/lib/handlers/index.ts:31:17)

If i use the z2m slider in web ui to "33":

> [2024-09-05 15:37:27] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqDD/Rollo/set' with data '{"position":33}'
[2024-09-05 15:37:27] debug:    z2m: Publishing 'set' 'position' to 'Rollo'
[2024-09-05 15:37:27] debug:    zh:controller:endpoint: ZCL command 0x54ef4410004454bb/1 genAnalogOutput.write({"presentValue":33}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"d>
[2024-09-05 15:37:27] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x54ef4410004454bb:48906/1 (0,0,1)

Cameleon99, 2024-09-05:
Hmm odd, fine for me with the latest one: 

Moved position to 94 from Z2M device GUI. 

```
info 2024-09-05 15:00:14z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T14:00:15.951Z","linkquality":51,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 15:00:14z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T14:00:15.951Z","linkquality":51,"motor_speed":"high","motor_state":null,"position":94,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

matze20211, 2024-09-05:
z2m gui worked, but external mqtt server not.

Koenkk, 2024-09-05:
@Cameleon99 updated https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24, is the position now correct when stopping it?

Cameleon99, 2024-09-05:
Better but some weird behaviour. 

1. If CLOSED/OPEN, hit OPEN/CLOSE, then STOP, position reports OK and HA says Open (correct)
2. If OPEN and I hit CLOSE, blind physically closes fully, and position reports as 0 but in HA says Open still (incorrect)
3. If drag slider to 0, position reports as 0% and HA says Closed (correct).
4. If I hit STOP when incorrectly stating OPEN/CLOSED updates correctly.

Logs:

Blind OPEN, request 84% - shows 84% - CORRECT

```
info 2024-09-05 21:23:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:23:23.230Z","linkquality":51,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 21:23:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:23:23.230Z","linkquality":51,"motor_speed":"high","motor_state":null,"position":84,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

Blind OPEN, request CLOSE (leave running)

```
info 2024-09-05 21:24:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:24:36.654Z","linkquality":45,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 21:24:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:24:36.654Z","linkquality":45,"motor_speed":"high","motor_state":null,"position":0,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

Hit STOP while running - position shows 46% - CORRECT

```
info 2024-09-05 21:24:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:24:45.847Z","linkquality":51,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 21:24:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:24:46.420Z","linkquality":48,"motor_speed":"high","motor_state":null,"position":46,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 21:24:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:24:46.420Z","linkquality":48,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

Hit CLOSE - left to fully close - position reported as 0 - HA shows 0% but Open - INCORRECT

```
info 2024-09-05 21:26:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:26:59.747Z","linkquality":51,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 21:26:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:26:59.747Z","linkquality":51,"motor_speed":"high","motor_state":null,"position":0,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

Hit STOP - position shows as 0 and HA shows Closed. - CORRECT
```
info 2024-09-05 21:26:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:26:59.747Z","linkquality":51,"motor_speed":"high","motor_state":null,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
info 2024-09-05 21:26:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Blind', payload '{"battery":42,"charging_status":false,"device_temperature":null,"last_seen":"2024-09-05T20:26:59.747Z","linkquality":51,"motor_speed":"high","motor_state":null,"position":0,"running":null,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
```

Hopefully that makes sense!

Koenkk, 2024-09-05:
Can you try with https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24 ?

matze20211, 2024-09-06:
I don't have HA, but the end-position is shown correctly in FHEM and GUI , state always "STOP" after finishing "CLOSE" or "OPEN"

Cameleon99, 2024-09-06:
> Can you try with https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24 ?

Now from Open/100% (HA) doing CLOSE then STOP, it now reports Closed/0% in HA, position 0, but I can see this error which is probably stopping it getting the actual position. 

```
error 2024-09-06 09:17:36z2m: Exception while calling fromZigbee converter: precisionRound is not defined}
```

Koenkk, 2024-09-06:
@Cameleon99 fixed the error: https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24

lskeet, 2024-10-08:
> @Cameleon99 fixed the error: https://gist.github.com/Koenkk/c9d93a35cd8bfcd984ff138fe8f29e24

Thanks @Koenkk.
I have tested the latest converter and it reports the current position correctly when stopped.
Note scenario 2 & 3 below where it would be ideal if it's possible to report that it's running/opening/closing but it is likely out of scope of this issue?

Tested:

1. OPEN (100) > CLOSE
2. OPEN > CLOSE (with STOP in between) _It reports as Closed @ 0% while closing and updates to the correct position & state once stopped_
3. CLOSE > OPEN (with STOP in between) _It reports as Open @ 100% while closing and updates to the correct position & state once stopped_
4. CLOSE > OPEN
5. Specific position % change

Koenkk, 2024-10-10:
Integrated the fix!

> would be ideal if it's possible to report that it's running/opening/closing 

I see it already exposes a `running` value, doesn't this work?

lskeet, 2024-10-10:
> I see it already exposes a `running` value, doesn't this work?

It's reporting null no matter how I operate the device:
```
{
    "battery": 95,
    "linkquality": 42,
    "motor_speed": "high",
    "position": 0,
    "state": "CLOSE",
    "update": {
        "installed_version": 3870,
        "latest_version": 3870,
        "state": "idle"
    },
    "update_available": false,
    "charging_status": null,
    "device_temperature": null,
    "motor_state": null,
    "running": null
}
```

matze20211, 2024-10-10:
That was my question, before 1.39 (?), the Position was reported from 0 to 100, i used it for animation of Rollers.

Koenkk, 2024-10-10:
@lskeet could you provide the debug log when closing the cover?

@matze20211 could you provide the debug log when the position was updated continuously pre 1.39?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

matze20211, 2024-10-11:
i'm back to 1.36.1 (?)

but "Rollo" shows this, if i try to use it:

`Failed to configure 'Rollo', attempt 1 (Error: Device 0x54ef4410004454bb has no input cluster manuSpecificLumi at getEndpointsWithCluster (/opt/zigbee2mqtt_36/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:28:15) at setupAttributes (/opt/zigbee2mqtt_36/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:55) at configure (/opt/zigbee2mqtt_36/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:104:23) at Object.configure (/opt/zigbee2mqtt_36/node_modules/zigbee-herdsman-converters/src/index.ts:162:27) at Configure.configure (/opt/zigbee2mqtt_36/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt_36/lib/eventBus.ts:174:17))`

matze

matze20211, 2024-10-11:
I cannot install a version lower than 1.31, with 1.31 the position is also not displayed.
but I am now unsure whether there really was a current position message with this device. I also have some TREDANSEN with which this works wonderfully. I may have mixed something up here. I am sorry.

lskeet, 2024-10-12:
@Koenkk is this what you're after? Hope I managed to filter out the noise from other devices correctly
```

[2024-10-12 18:26:57] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Bedroom Shades/set' with data 'CLOSE'
[2024-10-12 18:26:57] debug: 	z2m: Publishing 'set' 'state' to 'Bedroom Shades'
[2024-10-12 18:26:57] debug: 	zh:controller:endpoint: ZCL command 0x54ef4410004a6959/1 genAnalogOutput.write({"presentValue":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2024-10-12 18:26:57] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x54ef4410004a6959:39060/1 (0,0,1)
[2024-10-12 18:26:57] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":39060,"destendpoint":1,"srcendpoint":1,"clusterid":13,"transid":24,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[16,22,2,85,0,57,0,0,0,0]}}
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:writer: --> frame [254,20,36,1,148,152,1,1,13,0,24,0,30,10,16,22,2,85,0,57,0,0,0,0,84]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-10-12 18:26:57] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,24,222]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,24,222]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,24] - 222
[2024-10-12 18:26:57] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":24}
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,13,0,148,152,1,1,0,90,0,122,78,31,0,0,4,24,22,4,0,85,72,28,162]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,13,0,148,152,1,1,0,90,0,122,78,31,0,0,4,24,22,4,0,85,72,28,162]
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,13,0,148,152,1,1,0,90,0,122,78,31,0,0,4,24,22,4,0,85,72,28] - 162
[2024-10-12 18:26:57] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":13,"srcaddr":39060,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":90,"securityuse":0,"timestamp":2051706,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,22,4,0]}}
[2024-10-12 18:26:57] debug: 	zh:controller: Received payload: clusterID=13, address=39060, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=90, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":22,"commandIdentifier":4},"payload":[{"status":0}],"command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}}
[2024-10-12 18:26:57] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-12 18:26:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Shades', payload '{"battery":86,"charging_status":null,"device_temperature":null,"linkquality":90,"motor_speed":"high","motor_state":null,"position":0,"running":null,"state":"OPEN","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
[2024-10-12 18:27:52] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,97,242,1,202,55,235]
[2024-10-12 18:27:52] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,97,242,1,202,55,235]
[2024-10-12 18:27:52] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [97,242,1,202,55] - 235
[2024-10-12 18:27:52] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":62049,"relaycount":1,"relaylist":[14282]}
[2024-10-12 18:27:52] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-12 18:28:04] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,148,152,1,85,72,148]
[2024-10-12 18:28:04] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,148,152,1,85,72,148]
[2024-10-12 18:28:04] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [148,152,1,85,72] - 148
[2024-10-12 18:28:04] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":39060,"relaycount":1,"relaylist":[18517]}
[2024-10-12 18:28:04] debug: 	zh:zstack:unpi:parser: --- parseNext []

```

Koenkk, 2024-10-13:
It seems the device does not publish the `running` state (could be that there are multiple revisions of this device), the device also doesn't publish anything when it stops.

matze20211, 2024-10-14:
debug log with v31
[log_v.31.txt](https://github.com/user-attachments/files/17366679/log_v.31.txt)

[11 more replies]

closed 2025-03-14 by github-actions[bot]