Koenkk/zigbee2mqtt #23559: [New device support]: Air quolity sensor Tuya _TZE204_yvx5lh6k

aliakseika, 2024-08-08:
### Link

https://aliexpress.com/item/1005003428063886.html?spm=a2g2w.orderdetail.0.0.7af54aa66i4YHC&sku_id=12000026982644984

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0xa4c13879c5118c29","nwkAddr":30452,"manufId":4417,"manufName":"_TZE204_yvx5lh6k","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�G.i�G.i��G.i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723118109568}

### Comments

Hello, please add support for the device.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_yvx5lh6k',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

aliakseika, 2024-08-09:
maybe, I have version koenkk/zigbee2mqtt:latest-dev

Pauligrinder, 2024-08-21:
> maybe, I have version koenkk/zigbee2mqtt:latest-dev

Yeah, I'm running release 1.33.

pcrednet, 2024-08-21:
Any news about this?
I have 2 devices that are recognized as follows

_TZE204_yvx5lh6k

appears as unsupported.

What can I do?

pcrednet, 2024-08-21:
is already working with this converter:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const legacy= require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {   modelID: 'TS0601',
            manufacturerName: '_TZE204_yvx5lh6k',
        },
    ],
    model: 'TS0601_air_quality_sensor',
    vendor: 'Tuya',
    description: 'Smart Air Quality Sensor',
    fromZigbee: [legacy.fromZigbee.tuya_air_quality],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature(), e.humidity(), e.co2(), e.voc().withUnit('ppm'), e.formaldehyd(),
    ],
};

module.exports = definition;`

but I see that you are sending a lot of messages per second, is there any way to limit that amount of messages?

thx in advance

aliakseika, 2024-08-22:
It works with this converter, thanks! I have several similar sensors, they all send reports so often. I did not find how to change the interval

pcrednet, 2024-08-22:
to reduce reports interval; add this to this device:

debounce: 10
like this:

```python
devices:
  "0xa4sxxxxxxxx90":
    friendly_name: sonda
    debounce: 10
```

Koenkk, 2024-08-24:
Added the `_TZE204_yvx5lh6k`, thanks @pcrednet 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-24 by Koenkk