Koenkk/zigbee2mqtt #23563: Custom cluster and external converters

grebenyuk51, 2024-08-08:
### What happened?

When I use external converters, I have a problem with customs clusters

My device has one custom cluster 0xfc00 (64512) and two custom attributes 0x0a00 (2560), 0x0a01 (2561), manufacturercode 0x4c44
In frontend I cannot read values of this attributes due to error. But I can write attributes
```
debug: 	z2m: Received Zigbee message from 'S1', type 'readResponse', cluster 'manuSpecificAssaDoorLock', data '{"2560":0}' from endpoint 1 with groupID 0
debug: 	z2m: No converter available for 'My Device' with cluster 'manuSpecificAssaDoorLock' and type 'readResponse' and data '{"2560":0}'
```

As far as I understand, cluster 0xfc00 exists in `cluster.js` in herdsman library with name `manuSpecificAssaDoorLock` and has no 0x0a00/0x0a01 attributes. Furthermore, there is no manufacturer specific code in this cluster!

Q1: Why library processes the request with wrong manufacturer specific code? (my code is 0x4c44, and `manuSpecificAssaDoorLock` doesn't have any code)
Q2: How can I work with custom clusters with external converters without adding them to `cluster.js`

Thanks for reply!

P.S. My code
```
...
modernExtend.enumLookup({
	name: 'switch_mode',
	lookup: {
		'toggle': 0,
		'alternate': 2,
	},
	endpointName: `button_${i + 1}`,
	cluster: 0xfc00,
	attribute: {
		ID: 0x0a00,
		type: 0x30,
	},
	description: 'Operation method of the switch',
	access: 'ALL',
	entityCategory: 'config',
	zigbeeCommandOptions: {
		manufacturerCode: 0x4c44,
	},
}),
...
```

### What did you expect to happen?

Either exception about missing custom cluster in `cluster.js` or successfully processed read attribute request

### Zigbee2MQTT version

1.39.0 commit: 0326926

### Adapter firmware version

20230507

### Adapter

Dongle-P

### Setup

Docker, Pi 4

### Debug log

[2024-08-08 20:59:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/S1', payload '{"action":null,"last_seen":"2024-08-08T20:59:33+03:00","linkquality":61,"switch_events_button_1":null,"switch_events_button_2":null,"switch_events_button_3":null,"switch_events_button_4":null,"switch_mode_button_1":null,"switch_mode_button_2":null,"switch_mode_button_3":null,"switch_mode_button_4":null,"temperature":34.5}'
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-08-08 20:59:33] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,117,179]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,117,179]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,117] - 179
[2024-08-08 20:59:33] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":117}
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-08 20:59:33] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/S1/get' with data '{"switch_events_button_4":""}'
[2024-08-08 20:59:33] debug: 	z2m: Publishing get 'get' 'switch_events' to 'S1'
[2024-08-08 20:59:33] debug: 	zh:controller:endpoint: ZCL command 0x404ccafffe42b350/4 manuSpecificAssaDoorLock.read([2561], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":19524,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,173,3,1,155,99,211]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,173,3,1,155,99,211]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [173,3,1,155,99] - 211
[2024-08-08 20:59:33] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":941,"relaycount":1,"relaylist":[25499]}
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,0,252,173,3,2,1,0,61,0,182,141,165,0,0,10,28,68,76,130,1,0,10,0,48,0,155,99,28,106]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,252,173,3,2,1,0,61,0,182,141,165,0,0,10,28,68,76,130,1,0,10,0,48,0,155,99,28,106]
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,0,252,173,3,2,1,0,61,0,182,141,165,0,0,10,28,68,76,130,1,0,10,0,48,0,155,99,28] - 106
[2024-08-08 20:59:33] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":64512,"srcaddr":941,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":61,"securityuse":0,"timestamp":10849718,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[28,68,76,130,1,0,10,0,48,0]}}
[2024-08-08 20:59:33] debug: 	zh:controller: Received payload: clusterID=64512, address=941, groupID=0, endpoint=2, destinationEndpoint=1, wasBroadcast=false, linkQuality=61, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":19524,"transactionSequenceNumber":130,"commandIdentifier":1},"payload":[{"attrId":2560,"status":0,"dataType":48,"attrData":0}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2024-08-08 20:59:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-08 20:59:33] debug: 	z2m: Received Zigbee message from 'S1', type 'readResponse', cluster 'manuSpecificAssaDoorLock', data '{"2560":0}' from endpoint 2 with groupID 0
[2024-08-08 20:59:33] debug: 	z2m: No converter available for 'Zucchini' with cluster 'manuSpecificAssaDoorLock' and type 'readResponse' and data '{"2560":0}'

sjorge, 2024-08-09:
You probably want to push your custom cluster from your external converter, this is now support and custom cluster are slowly being moved from zh to zhc. Here is an example:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/6e96d847ddf633725108c08ddfcd6088a9fe3a71/src/lib/ikea.ts#L759-L777

grebenyuk51, 2024-08-09:
how to do export function in external converter? is there any examples?

sjorge, 2024-08-09:
I'm assuming you can just import `deviceAddCustomCluster`

```
import { deviceAddCustomCluster } from '../lib/modernExtend';
```

And then just add `deviceAddCustomCluster('manuSpecificXXX', { ... }` directly to the `extend` entry for the device in your external converter.

grebenyuk51, 2024-08-09:
so, I have this code
```
        extend: [
            modernExtend.deviceAddCustomCluster('manuSpecificLlamaDevZucchini', {
                ID: 0xfc00,
                manufacturerCode: 0x4c44,
                attributes: {
                    switch_mode: {ID: 0x0a00, type: Zcl.DataType.ENUM8},
                    switch_events: {ID: 0x0a01, type: Zcl.DataType.ENUM8},            
                },
                commands: {},
                commandsResponse: {},
            }),
            modernExtend.temperature({reporting: {min: '10_SECONDS', max: '1_HOUR', change: 10}}),
            ...(
                () => {
                    const feature = [];

                    for (let i = 0; i < 4; i++) {
                        feature.push(
                            modernExtend.enumLookup({
                                name: 'switch_mode',
                                lookup: {
                                    'toggle': 0,
                                    'alternate': 2,
                                },
                                endpointName: `button_${i + 1}`,
                                cluster: 0xfc00,
                                attribute: {
                                    ID: 0x0a00,
                                    type: 0x30,
                                },
                                description: 'Operation method of the switch',
                                access: 'ALL',
                                entityCategory: 'config',
                                zigbeeCommandOptions: {
                                    manufacturerCode: 0x4c44,
                                },
                            }),
                        );
                        feature.push(
                            modernExtend.enumLookup({
                                name: 'switch_events',
                                lookup: {
                                    'on/off': 6,
                                    'off/on': 5,
                                    'toggle': 3,
                                    'on': 2,
                                    'off': 1,
                                },
                                endpointName: `button_${i + 1}`,
                                cluster: 0xfc00,
                                attribute: {
                                    ID: 0x0a01,
                                    type: 0x30,
                                },
                                description: 'Command to be generated when the switch moves between its two states',
                                access: 'ALL',
                                entityCategory: 'config',
                                zigbeeCommandOptions: {
                                    manufacturerCode: 0x4c44,
                                },
                            }), 
                        );
                    }

                    return feature;
                }
            )(),
            modernExtend.deviceEndpoints({
[... 13 lines cut ...]
[2024-08-09 13:37:46] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":48718,"destendpoint":4,"srcendpoint":1,"clusterid":64512,"transid":14,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[20,68,76,18,0,1,10]}}
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:writer: --> frame [254,17,36,1,78,190,4,1,0,252,14,0,30,7,20,68,76,18,0,1,10,47]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-08-09 13:37:46] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,14,200]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,14,200]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,14] - 200
[2024-08-09 13:37:46] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":14}
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,0,252,78,190,4,1,0,171,0,121,240,163,0,0,10,28,68,76,18,1,1,10,0,48,6,78,190,29,142]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,252,78,190,4,1,0,171,0,121,240,163,0,0,10,28,68,76,18,1,1,10,0,48,6,78,190,29,142]
[2024-08-09 13:37:46] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,0,252,78,190,4,1,0,171,0,121,240,163,0,0,10,28,68,76,18,1,1,10,0,48,6,78,190,29] - 142
[2024-08-09 13:37:46] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":64512,"srcaddr":48718,"srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"linkquality":171,"securityuse":0,"timestamp":10743929,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[28,68,76,18,1,1,10,0,48,6]}}
[2024-08-09 13:37:46] debug: 	zh:controller: Received payload: clusterID=64512, address=48718, groupID=0, endpoint=4, destinationEndpoint=1, wasBroadcast=false, linkQuality=171, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":19524,"transactionSequenceNumber":18 [... cut ...]
[2024-08-09 13:37:46] debug: 	z2m: Received Zigbee message from 'Test Switch', type 'readResponse', cluster 'manuSpecificLlamaDevZucchini', data '{"switch_events":6}' from endpoint 4 with groupID 0
[2024-08-09 13:37:46] debug: 	z2m: No converter available for 'Zucchini' with cluster 'manuSpecificLlamaDevZucchini' and type 'readResponse' and data '{"switch_events":6}'
```

`deviceAddCustomCluster('manuSpecificXXX', { ... }` works, but it can't parse response. Do I need to write custom fromZigbee converter? Or I did something wrong?

Bytheway, thanks for helping me

sjorge, 2024-08-09:
I think you can now do
```
                            modernExtend.enumLookup({
                                name: 'switch_mode',
                                lookup: {
                                    'toggle': 0,
                                    'alternate': 2,
                                },
                                endpointName: `button_${i + 1}`,
                                cluster: 'manuSpecificLlamaDevZucchini',
                                attribute: 'switch_mode',
                                description: 'Operation method of the switch',
                                access: 'ALL',
                                entityCategory: 'config',
                            }),
```

As you have added a definition for it to the device, it should also know to use the manufacturing code because you told it so.

grebenyuk51, 2024-08-09:
works perfectly! I couldn’t have done it without you!

Here is all my code as an example

```
        extend: [
            modernExtend.deviceAddCustomCluster('manuSpecificLlamaDevZucchini', {
                ID: 0xfc00,
                manufacturerCode: 0x4c44,
                attributes: {
                    switch_mode: {ID: 0x0a00, type: Zcl.DataType.ENUM8},
                    switch_events: {ID: 0x0a01, type: Zcl.DataType.ENUM8},            
                },
                commands: {},
                commandsResponse: {},
            }),
            modernExtend.temperature({reporting: {min: '10_SECONDS', max: '1_HOUR', change: 10}}),
            ...(
                () => {
                    const feature = [];

                    for (let i = 0; i < 4; i++) {
                        feature.push(
                            modernExtend.enumLookup({
                                name: 'switch_mode',
                                lookup: {
                                    'toggle': 0,
                                    'alternate': 2,
                                },
                                endpointName: `button_${i + 1}`,
                                cluster: 'manuSpecificLlamaDevZucchini',
                                attribute: 'switch_mode',
                                description: 'Operation method of the switch',
                                access: 'ALL',
                                entityCategory: 'config',
                            }),
                        );
                        feature.push(
                            modernExtend.enumLookup({
                                name: 'switch_events',
                                lookup: {
                                    'on/off': 6,
                                    'off/on': 5,
                                    'toggle': 3,
                                    'on': 2,
                                    'off': 1,
                                },
                                endpointName: `button_${i + 1}`,
                                cluster: 'manuSpecificLlamaDevZucchini',
                                attribute: 'switch_events',
                                description: 'Command to be generated when the switch moves between its two states',
                                access: 'ALL',
                                entityCategory: 'config',
                            }), 
                        );
                    }

                    return feature;
                }
            )(),
            modernExtend.deviceEndpoints({
                endpoints: {'button_1': 1, 'button_2': 2, 'button_3': 3, 'button_4': 4},
                multiEndpointSkip: ['temperature'],
            }),
        ],
```

closed 2024-08-09 by grebenyuk51