Koenkk/zigbee2mqtt #23569: [New device support]: Tuya CO2 Sensor TS0601_TZE204_ogkdpgy2

Yoo-Yohan, 2024-08-09:
### Link

https://aliexpress.com/item/1005005185654474.html?

### Database entry

{"id":51,"type":"Router","ieeeAddr":"0xa4c13865459668c9","nwkAddr":62024,"manufId":4417,"manufName":"_TZE204_ogkdpgy2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_ogkdpgy2","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723176996867}

### Comments

[image]

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_ogkdpgy2',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

nmaster2042, 2024-08-13:
I'm wondering if this new version suffers the zigbee network flood bug, like previous one, any update on this ?

Yoo-Yohan, 2024-08-19:
> I'm wondering if this new version suffers the zigbee network flood bug, like previous one, any update on this ?

This also seems to have same problem, too short reporting cycle, milliseconds

rokiden, 2024-09-07:
I have fixed flood bug of this device with [custom firmware](https://github.com/rokiden/Tuya_CO2_STC8/blob/tuya_co2/user/README.md).

nmaster2042, 2024-09-08:
Very interesting.
Thanks you for this nice dev.

closed 2024-12-23 by github-actions[bot]