Koenkk/zigbee2mqtt #23575: [ember] Zigbee Green Power frames not recognized correctly close to some Philips bulbs

LaurentvdBos, 2024-08-09:
### What happened?

After pairing a Philips Hue Tap [8718696743133](https://www.zigbee2mqtt.io/devices/8718696743133.html#philips-8718696743133), it works *almost* everywhere in my house, except close to my https://www.zigbee2mqtt.io/devices/4090330P9.html#philips-4090330p9 and https://www.zigbee2mqtt.io/devices/8718699703424.html#philips-8718699703424.

After digging a bit deeper, and turning logging on with debug, it seems that frames "far away" from these devices are correctly picked up as Green Power frames:
```
zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=184 Len=33]
zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=undefined], [gpdLink=207], [sequenceNumber=166], [addr={"applicationId":0,"sourceId":4650238,"endpoint":0}], [gpdfSecurityLevel=FC_MIC], [gpdfSecurityKeyType=NWK], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=2726], [gpdCommandId=17], [mic=479921080], [proxyTableIndex=255], [gpdCommandPayload=]
zh:controller:greenpower: Received unhandled command '0x11' from '4650238'
zh:controller: Received payload: clusterID=33, address=4650238, groupID=2948, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=207, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":166,"commandIdentifier":0},"payload":{"options":0,"srcID":4650238,"frameCounter":2726,"commandID":17,"payloadSize":0,"commandFrame":{}},"command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}}                                                              z2m: Received Zigbee message from 'Philips Hue Tap', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":17,"frameCounter":2726,"options":0,"payloadSize":0,"srcID":4650238}' from endpoint 242 with groupID 2948                                                                                                                                    
```
But frames close to the bulbs mentioned above are picked up as Multicast traffic:
```
zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=184 Len=42]
zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=MULTICAST], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":0,"groupId":2948,"sequence":156}], [packetInfo:{"senderShortId":62718,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":100,"lastHopRssi":-75,"lastHopTimestamp":0}], [messageContents=117500a054fef44600a50a00001100271c8d]
```
You can even see the sourceEndpoint being 242, the green power endpoint. So it seems that the ezsp firmware does not correctly flag these packages as green power packages and ember therefore does not call the green power callback, missing my click.

After ordering a second dongle and sniffing the traffic on my Zigbee network (with the great ember-zli tooling!) and it shows that Zigbee green power frames are being submitted in both cases, meaning that Wireshark identifies in both cases a broadcasted "ZCL Green Power: GP Notification" frame, with minimal expected differences (sequence number, for example). I'll spare you the binary frames, but let me know in case you are interested.

Here I got stuck. I'd love to dive deeper to see where this difference is coming from, so any hints to pinpoint who is messing with my Green Power switch (either the Philips bulb or the dongle) are appreciated!

### What did you expect to happen?

My Green Power device to work everywhere in my house.

### How to reproduce it (minimal and precise)

I honestly have to clue, up to the point "come close to my Philips bulb and try a green power device", and I am more looking for pointers how I can further isolate this issue.

### Zigbee2MQTT version

1.39.1-dev commit: fe6cacd

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff zbdongle-e (with ember)

### Setup

Containerized

Koenkk, 2024-08-09:
> But frames close to the bulbs mentioned above are picked up as Multicast traffic:

Could you provide a bit more debug logging to see what happens after this?

LaurentvdBos, 2024-08-10:
Sure! Here it is with timings included. First I tried close to the faulty bulb:
```
[2024-08-09 15:15:02] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-09 15:15:02] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-08-09 15:15:02] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=2](frmRx=2) Added to rxQueue
[2024-08-09 15:15:02] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=1)
[2024-08-09 15:15:02] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=184 Len=42]
[2024-08-09 15:15:02] debug:         zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=MULTICAST], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":0,"groupId":2948,"sequence":156}], [packetInfo:{"senderShortId":62718,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":100,"lastHopRssi":-75,"lastHopTimestamp":0}], [messageContents=117500a054fef44600a50a00001100271c8d]
[2024-08-09 15:15:03] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-09 15:15:03] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-08-09 15:15:03] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3](frmRx=3) Added to rxQueue
[2024-08-09 15:15:03] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=1)
[2024-08-09 15:15:03] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=184 Len=22]
[2024-08-09 15:15:03] debug:         zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":0,"groupId":2948,"sequence":156}], [messageTag=255]
```
And then I walked to the other side of the room where it all works:
```
[2024-08-09 15:15:10] debug:         z2m: Saving state to file /app/data/state.json
[2024-08-09 15:15:11] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-09 15:15:11] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-08-09 15:15:11] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=4](frmRx=4) Added to rxQueue
[2024-08-09 15:15:11] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=1)
[2024-08-09 15:15:11] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=184 Len=33]
[2024-08-09 15:15:11] debug:         zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=undefined], [gpdLink=207], [sequenceNumber=166], [addr={"applicationId":0,"sourceId":4650238,"endpoint":0}], [gpdfSecurityLevel=FC_MIC], [gpdfSecurityKeyType=NWK], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=2726], [gpdCommandId=17], [mic=479921080], [proxyTableIndex=255], [gpdCommandPayload=]
[2024-08-09 15:15:11] debug:         zh:controller:greenpower: Received unhandled command '0x11' from '4650238'
[2024-08-09 15:15:11] debug:         zh:controller: Received payload: clusterID=33, address=4650238, groupID=2948, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=207, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":166,"commandIdentifier":0},"payload":{"options":0,"srcID":4650238,"frameCounter":2726,"commandID":17,"payloadSize":0,"commandFrame":{}},"command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}}
[2024-08-09 15:15:11] debug:         z2m: Received Zigbee message from 'Philips Hue Tap', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":17,"frameCounter":2726,"options":0,"payloadSize":0,"srcID":4650238}' from endpoint 242 with groupID 2948
[2024-08-09 15:15:11] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap', payload '{"action":"press_3","last_seen":"2024-08-09T15:15:11+02:00","linkquality":207}'
[2024-08-09 15:15:11] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap', payload '{"action":"","last_seen":"2024-08-09T15:15:11+02:00","linkquality":207}'
[2024-08-09 15:15:11] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap/action', payload 'press_3'
[2024-08-09 15:15:11] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-09 15:15:11] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[... cut ...]
[2024-08-09 15:15:12] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=1)
[2024-08-09 15:15:12] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=184 Len=22]
[2024-08-09 15:15:12] debug:         zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":0,"groupId":2948,"sequence":157}], [messageTag=255]
```

Koenkk, 2024-08-12:
This message seems to be ignored because for some of the empty `else if` [here](https://github.com/Koenkk/zigbee-herdsman/blob/16d5cfe058ccb20af0e0d48356b577a0d31ce64b/src/adapter/ember/ezsp/ezsp.ts#L5287). 

I suspect that `ezspGpepIncomingMessageHandler` is only called when the green power frame is directly received by the adapter, in case it's goes via a green power proxy, `ezspIncomingMessageHandler` is used (since the GP frame is than encapsulated in a normal message). 

@LaurentvdBos one thing you can try is to modify the code like this

```js
else if (apsFrame.profileId === ZSpec.GP_PROFILE_ID) {
            this.emit(EzspEvents.INCOMING_MESSAGE, type, apsFrame, packetInfo.lastHopLqi, packetInfo.senderShortId, messageContents);
            // only broadcast loopback in here
        }
```

Then override `zigbee-herdsman` in the Docker container by mounting the following volume `./custom-zigbee-herdsman:/app/node_modules/zigbee-herdsman` (make sure to run a `npm run build` in `zigbee-herdsman` after making the change)

CC: @Nerivec

Nerivec, 2024-08-12:
Interesting!
I parsed the payload manually to check, and it seems it would be fine "as-is".

`ezspGpepIncomingMessageHandler`:
```json
    "payload": {
        "options": 0,
        "srcID": 4650238,
        "frameCounter": 2726,
        "commandID": 17,
        "payloadSize": 0,
        "commandFrame": {}
    },
```
`ezspIncomingMessageHandler `:
```json

    "payload": {
        "options": 21664,
        "srcID": 4650238,
        "frameCounter": 2726,
        "commandID": 17,
        "payloadSize": 0,
        "commandFrame": {
            "raw": {
                "type": "Buffer",
                "data": [
                    175,
                    247,
                    210
                ]
            }
        },
        "gppNwkAddr": 63407,
        "gppGddLink": 210
    },
```

I'll push a fix after current PR is merged, can't hurt to have it, I'll just exclude broadcast loopback to avoid unnecessary emitting.

LaurentvdBos, 2024-08-12:
> one thing you can try is to modify the code like this
> 
> ```js
> } else if (apsFrame.profileId === ZSpec.GP_PROFILE_ID) {
>     this.emit(EzspEvents.INCOMING_MESSAGE, type, apsFrame, packetInfo.lastHopLqi, packetInfo.senderShortId, messageContents);
>     // only broadcast loopback in here
> }
> ```

This fixes the issue, without any doubt: after the change, the green power switch works wherever I am, and after reverting it stops working again in the kitchen (and it works again after applying the change, etc.)

Interestingly enough, the spot in my kitchen happens to be farthest away from the dongle. I am impressed by the range though, since the dongle is on the third floor (with concrete floors)!

Anyways, I have put your change that I used to verify the fix in a PR:

https://github.com/Koenkk/zigbee-herdsman/pull/1148

Feel free to reject / decline, since @Nerivec mentioned to pull it in elsewhere already.

Koenkk, 2024-08-12:
@LaurentvdBos to double check, you do not get duplicate messages (actions) when the frames are directly received by the coordinator?

LaurentvdBos, 2024-08-12:
Nope, all buttons trigger a single event. Here is a snippet of what I see in the log when pressing a button:

```
[2024-08-12 21:22:31] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-12 21:22:31] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-08-12 21:22:31] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=2](frmRx=2) Added to rxQueue
[2024-08-12 21:22:31] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=0)
[2024-08-12 21:22:31] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=47 Len=42]
[2024-08-12 21:22:31] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=MULTICAST], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":256,"groupId":2948,"sequence":240}], [packetInfo:{"senderShortId":62718,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":48,"lastHopRssi":-88,"lastHopTimestamp":0}], [messageContents=117200a054fef44600f90a00001200d90f09]
[2024-08-12 21:22:31] debug:    zh:controller:greenpower: Received unhandled command '0x12' from '62718'
[2024-08-12 21:22:31] debug:    zh:controller: Received payload: clusterID=33, address=62718, groupID=2948, endpoint=242, destinationEndpoint=255, wasBroadcast=false, linkQuality=48, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":114,"commandIdentifier":0},"payload":{"options":21664,"srcID":4650238,"frameCounter":2809,"commandID":18,"payloadSize":0,"commandFrame":{"raw":{"type":"Buffer","data":[217,15,9]}},"gppNwkAddr":4057,"gppGddLink":9},"command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}}
[2024-08-12 21:22:31] debug:    z2m: Received Zigbee message from 'Philips Hue Tap', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[217,15,9],"type":"Buffer"}},"commandID":18,"frameCounter":2809,"gppGddLink":9,"gppNwkAddr":4057,"options":21664,"payloadSize":0,"srcID":4650238}' from endpoint 242 with groupID 2948
[2024-08-12 21:22:31] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap', payload '{"action":"press_4","last_seen":"2024-08-12T21:22:31+02:00","linkquality":48}'
[2024-08-12 21:22:31] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap', payload '{"action":"","last_seen":"2024-08-12T21:22:31+02:00","linkquality":48}'
[2024-08-12 21:22:31] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap/action', payload 'press_4'
[2024-08-12 21:22:32] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-12 21:22:32] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-08-12 21:22:32] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=3](frmRx=3) Added to rxQueue
[2024-08-12 21:22:32] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=0)
[2024-08-12 21:22:32] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=47 Len=22]
[2024-08-12 21:22:32] debug:    zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":256,"groupId":2948,"sequence":240}], [messageTag=255]
```

Looks good to me (but I am admittedly not an expert).

Nerivec, 2024-08-12:
_Actually no the noise I was referring to is to avoid bubbling GP loopback messages up the Z2M chain (since it does nothing but fire events that will get cancelled out at the top)._

What were the circumstances when it triggered twice, versus your previous test?

LaurentvdBos, 2024-08-13:
> What were the circumstances when it triggered twice, versus your previous test?

I used the remote at the spot where it also works originally. My guess would be that the adapter sees the frame now twice: once as "pure" GP frame and once as broadcasted proxy frame from the bulbs nearby. Since these frames are different from each other, two events are triggered. However, this is for some reason not perceived as a `action_3`, something @Koenkk mentioned to check.

LaurentvdBos, 2024-08-16:
@Koenkk Ah yes, if I disable last seen, the empty message disappears from the log. Here it is, in case you are interested:

```
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=4](frmRx=4) Added to rxQueue                                                                  [2024-08-16 11:05:35] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=0)                                                                                              [2024-08-16 11:05:35] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=47 Len=33]
[2024-08-16 11:05:35] debug:         zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=undefined], [gpdLink=10], [sequenceNumber=67], [addr={"applicationId":0,"sourceId":4650238,"endpoint":0}], [gpdfSecurityLevel=FC_MIC], [gpdfSecurityKeyType=NWK], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=2883], [gpdCommandId=34], [mic=4155230387], [proxyTableIndex=255], [gpdCommandPayload=]                                                                                                                            [2024-08-16 11:05:35] debug:         zh:controller:greenpower: Received unhandled command '0x22' from '4650238'                                                                             [2024-08-16 11:05:35] debug:         zh:controller: Received payload: clusterID=33, address=4650238, groupID=2948, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=10, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":67,"commandIdentifier":0},"payload":{"options":0,"srcID":4650238,"frameCounter":2883,"commandID":34,"payloadSize":0,"commandFrame":{}},"command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}}
[2024-08-16 11:05:35] debug:         z2m: Received Zigbee message from 'Philips Hue Tap', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":34,"frameCounter":2883,"options":0,"payloadSize":0,"srcID":4650238}' from endpoint 242 with groupID 2948
[2024-08-16 11:05:35] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap', payload '{"action":"press_1","last_seen":"2024-08-16T11:02:24+02:00","linkquality":10}'    [2024-08-16 11:05:35] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap', payload '{"action":"","last_seen":"2024-08-16T11:02:24+02:00","linkquality":10}'
[2024-08-16 11:05:35] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue Tap/action', payload 'press_1'
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=5](frmRx=5) Added to rxQueue
[2024-08-16 11:05:35] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=0)
[2024-08-16 11:05:35] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=47 Len=42]
[2024-08-16 11:05:35] debug:         zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=MULTICAST], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":0,"groupId":2948,"sequence":58}], [packetInfo:{"senderShortId":62718,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":96,"lastHopRssi":-76,"lastHopTimestamp":0}], [messageContents=110b00a054fef44600430b00002200271cdf]
[2024-08-16 11:05:35] debug:         zh:controller:greenpower: Received unhandled command '0x22' from '62718'
[2024-08-16 11:05:35] debug:         zh:controller: Received payload: clusterID=33, address=62718, groupID=2948, endpoint=242, destinationEndpoint=255, wasBroadcast=false, linkQuality=96, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":11,"commandIdentifier":0},"payload":{"options":21664,"srcID":4650238,"frameCounter":2883,"commandID":34,"payloadSize":0,"commandFrame":{"raw":{"type":"Buffer","data":[39,28,223]}},"gppNwkAddr":7207,"gppGddLink":223},"command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"na [... cut ...]
[2024-08-16 11:05:36] debug:         zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=0)
[2024-08-16 11:05:36] debug:         zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=47 Len=22]
[2024-08-16 11:05:36] debug:         zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"options":0,"groupId":2948,"sequence":58}], [messageTag=255]
```

Seems like the original patch indeed does the trick.

Nerivec, 2024-08-16:
@LaurentvdBos Can you confirm the PR (which turned into a bit more 😉) works as expected?

closed 2024-08-17 by Koenkk

LaurentvdBos, 2024-08-18:
@Nerivec I'll wait for the `latest-dev` tag to pick up the updated `zigbee-herdsman` and then I'll verify it. It still sits on version 0.56.0 (your PR landed in 0.56.1 if I am not mistaken).

LaurentvdBos, 2024-08-18:
Thanks! The new version (z2m claims to be `1.39.1-dev commit: [e95116c](https://github.com/Koenkk/zigbee2mqtt/commit/e95116c)`) solves this issue. So all resolved now.