Koenkk/zigbee2mqtt #23576: [New device support]: TS0601 _TZE284_fzo2pocs

StevenBoelhouwer, 2024-08-09:
### Link

https://www.zemismart.com/products/wifi-motorized-window-opener-electric-chain-tuya-smart-life-app-alexa-google-home-control-110v-240v

### Database entry

Dunno how to look for that

### Comments

Window opener based on Tuya TS0601, external definition is working for me :)

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_fzo2pocs',
        },
    ],
    model: 'ZM90E-DT250N/A400',
    vendor: 'Zemismart',
    description: 'Window opener',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
	options: [exposes.options.invert_cover()],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
		e.enum('reverse_direction', ea.STATE_SET, ['true', 'false']).withDescription('Reverse the motor direction'),
		e.binary('motor_fault', ea.STATE, true, false).withDescription('Motor Fault'),
    ],
    meta: {
        tuyaDatapoints: [
			[1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
			[2, 'position', tuya.valueConverter.coverPositionInverted],
			[3, 'position', tuya.valueConverter.coverPositionInverted],
			[5, 'reverse_direction', tuya.valueConverterBasic.lookup({'true': tuya.enum(0), 'false': tuya.enum(1)})],
			[12, 'motor_fault', tuya.valueConverter.trueFalse1],
		],
    },
};

module.exports = definition;
```

Koenkk, 2024-08-12:
Added, thanks!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `ZM90E-DT250N-A400.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-08-12 by Koenkk

klepper19, 2025-01-08:
If I understand well, this is the quirk for Z2M. Can we add this device to ZHA as well by re-using this code? Sorry, new in this topic. I can do a test though.