Koenkk/zigbee2mqtt #23577: [New device support]: Aqara fp1e - Presence Sensor

balu-, 2024-08-09:
### Link

https://www.aqara.com/en/product/presence-sensor-fp1e/

### Database entry

{"id":66,"type":"EndDevice","ieeeAddr":"0x54ef441000db1e51","nwkAddr":57999,"manufId":4660,"manufName":"aqara","powerSource":"DC Source","modelId":"lumi.sensor_occupy.agl1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":65520,"inClusterList":[0,3,64704],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sensor_occupy.agl1","appVersion":26,"manufacturerName":"aqara","powerSource":4,"zclVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20240520","swBuildId":""}},"manuSpecificLumi":{"attributes":{"247":{"type":"Buffer","data":[3,40,25,5,33,1,0,8,33,26,1,10,33,0,0,12,32,20,16,32,1,18,32,0,101,32,0,102,32,1]}}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":26,"stackVersion":2,"hwVersion":1,"dateCode":"20240520","swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1723220271269}

### Comments

Unfortunately I'm a lost and dont have capabilities to add the functionality.
I'd love if the device gets support, I hope it works similar to https://www.zigbee2mqtt.io/devices/RTCZCGQ11LM.html
so that its not mutch work to get going.

### External definition

```shell
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'aqara',
    description: 'Automatically generated definition',
    extend: [identify()],
    meta: {},
};

module.exports = definition;
```

balu-, 2024-08-11:
I fiddeled around with external definitions, and it seems i got temperature & power loss cound working.
Unfortunately not Motion, which produces Logs like this:

```
z2m: Received Zigbee message from 'FP1E', type 'attributeReport', cluster 'manuSpecificLumi', data '{"351":360}' from endpoint 1 with groupID 0
zhc:lumi: lumi.sensor_occupy.agl1: unknown key 351 with value 360
```

Custom config looks like this:
```
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const lumi = require('zigbee-herdsman-converters/lib/lumi');
const {manufacturerCode} = lumi;

const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'aqara',
    description: 'Presence sensor FP1E',
    extend: [identify()],
    fromZigbee: [lumi.fromZigbee.lumi_specific, lumi.fromZigbee.lumi_presence_region_events], //We will add this later
    toZigbee: [ lumi.toZigbee.lumi_presence,
            lumi.toZigbee.lumi_monitoring_mode,
            lumi.toZigbee.lumi_approach_distance,
            lumi.toZigbee.lumi_motion_sensitivity,
            lumi.toZigbee.lumi_reset_nopresence_status,
            lumi.toZigbee.lumi_presence_region_upsert,
            lumi.toZigbee.lumi_presence_region_delete ], // Should be empty, unless device can be controlled (e.g. lights, switches).*/
    exposes: [ e.presence().withAccess(ea.STATE_GET),
            e.device_temperature(),
            e.power_outage_count() ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
     meta: {},
};

module.exports = definition;
```
(is oriented towards https://github.com/Koenkk/zigbee-herdsman-converters/blob/d28d1994b84aad59c044e3445215fd2d2c77b680/src/devices/lumi.ts#L1783 )

morgendagen, 2024-08-11:
I believe this has to do with https://github.com/Koenkk/zigbee-herdsman-converters/blob/d28d1994b84aad59c044e3445215fd2d2c77b680/src/lib/lumi.ts#L702 only recognising the FP1 as a presence sensor. Adding `lumi.sensor_occupy.agl1` to this list will hopefully bring basic functionality.

morgendagen, 2024-08-11:
Hack to get presence detection in HA:

```
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const lumi = require('zigbee-herdsman-converters/lib/lumi');
const {manufacturerCode} = lumi;

const fzLocal = {
    fp11_presence: {
        cluster: 'manuSpecificLumi',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            payload = {};
            for (const key in msg.data) {
                value = msg.data[key];
                if (key == "322") {
                    payload["presence"] = value == 1;
                }
            }
            return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'aqara',
    description: 'Presence sensor FP1E',
    extend: [identify()],
    fromZigbee: [
        fzLocal.fp11_presence,
        lumi.fromZigbee.lumi_specific,
    ], //We will add this later
    toZigbee: [
    ], // Should be empty, unless device can be controlled (e.g. lights, switches).*/
    exposes: [
        e.presence().withAccess(ea.STATE_GET),
        e.device_temperature(),
        e.power_outage_count(),
    ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint) => {
        device.softwareBuildID = `${device.applicationVersion}`;
        device.save();

        const endpoint = device.getEndpoint(1);
        await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
    },
    meta: {},
};

module.exports = definition;
```

danielgoepp, 2024-08-13:
Nice, thanks @morgendagen! That does at least get basic functionality up. Hopefully full support can come soon. I have been looking for an opportunity to learn more about creating device configs in here, perhaps this is my chance? ;) I just bought two of these to play with.

kkossev, 2024-08-14:
Aqara cluster 0xFCC0, attribute 0x015B (R/W), DataType UINT32 (0x23) is the configurable maximum detection distance in millimeters (default 600 = 6 meters).

Aqara cluster 0xFCC0, attribute 0x0160 (read-only)  : 0x02 = idle (no movement), 0x03 = large movement, 0x04 = small movement

ryanrozich, 2024-08-15:
would love to see this, was just shopping for the FP1E and this was the first hit in my google search for 'fp1e and z2m' - is is fp2 going to be supported?

dinhchinh82, 2024-08-16:
> would love to see this, was just shopping for the FP1E and this was the first hit in my google search for 'fp1e and z2m' - is is fp2 going to be supported?

FP2 is wifi version and unable to work with z2m. You can add FP2 to HA via homekit controller.

elmocito, 2024-08-16:
oh boy. can't wait to have the fp1e running on my z2m instance. If you guys need a tester, shout out! 😎

krabbs, 2024-08-17:
> Hack to get presence detection in HA:
> 
> ```
> const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const lumi = require('zigbee-herdsman-converters/lib/lumi');
> const {manufacturerCode} = lumi;
> 
> 
> const fzLocal = {
>     fp11_presence: {
>         cluster: 'manuSpecificLumi',
>         type: ['readResponse', 'attributeReport'],
>         convert: (model, msg, publish, options, meta) => {
>             payload = {};
>             for (const key in msg.data) {
>                 value = msg.data[key];
>                 if (key == "322") {
>                     payload["presence"] = value == 1;
>                 }
>             }
>             return payload;
>         },
>     },
> };
> 
> const definition = {
>     zigbeeModel: ['lumi.sensor_occupy.agl1'],
>     model: 'lumi.sensor_occupy.agl1',
>     vendor: 'aqara',
>     description: 'Presence sensor FP1E',
>     extend: [identify()],
>     fromZigbee: [
>         fzLocal.fp11_presence,
>         lumi.fromZigbee.lumi_specific,
>     ], //We will add this later
>     toZigbee: [
>         lumi.toZigbee.lumi_presence,
>     ], // Should be empty, unless device can be controlled (e.g. lights, switches).*/
>     exposes: [
>         e.presence().withAccess(ea.STATE_GET),
>         e.device_temperature(),
>         e.power_outage_count(),
>     ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
>     configure: async (device, coordinatorEndpoint) => {
>         device.softwareBuildID = `${device.applicationVersion}`;
>         device.save();
> 
>         const endpoint = device.getEndpoint(1);
>         await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
>     },
>     meta: {},
> };
> 
> module.exports = definition;
> ```

Hi, with this configuration the module is recognized with basic feature and it's correct included. But the sensor is not sending information. Did i miss any point? Some additional bindings?
Thanks.
I would love to see this device

krabbs, 2024-08-27:
My is problem is solved. After including keep pressing the button for a minute. This seems to be a problem of the deconz controller. (Have the same unsolved for aquara thermostat)

ferbulous, 2024-09-03:
Hi, just wondering if you could initiate the ai spatial learning from z2m?

> Hack to get presence detection in HA:
>
> ```
> const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = [...]

Hi, just wondering if you could initiate the ai spatial learning from z2m?

morgendagen, 2024-09-03:
> Hi, just wondering if you could initiate the ai spatial learning from z2m?

I do not have access to an Aqara hub, and am therefore unable to sniff the zigbee trafic. Hopefully someone else can extend the support before long. I do have a small update, since I figured out how to get the distance to the target:

```const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const lumi = require('zigbee-herdsman-converters/lib/lumi');
const {manufacturerCode} = lumi;

const fzLocal = {
    fp11_presence: {
        cluster: 'manuSpecificLumi',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            console.info("msg.data=", msg.data);
            payload = {};
            for (const key in msg.data) {
                value = msg.data[key];
                if (key == "322") {
                    payload["presence"] = value == 1;
                }
                if (key == "351") {
                    // distance in cm
                    payload["target_distance"] = 0.01 * value
                }
            }
            return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'Aqara',
    description: 'Presence sensor FP1E',
    extend: [lumi.modernExtend.lumiZigbeeOTA()],
    fromZigbee: [
        fzLocal.fp11_presence,
        lumi.fromZigbee.lumi_specific,
    ],
    toZigbee: [
        lumi.toZigbee.lumi_presence,
    ],
    exposes: [
        e.presence().withAccess(ea.STATE_GET),
        e.device_temperature(),
        e.power_outage_count(),
        e.numeric('target_distance', ea.STATE_GET).withUnit('m').withDescription('Target distance'),
    ],
    configure: async (device, coordinatorEndpoint) => {
        device.softwareBuildID = `${device.applicationVersion}`;
        device.save();

        const endpoint = device.getEndpoint(1);
        await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
    },
    meta: {},
};

module.exports = definition;
```

krabbs, 2024-09-03:
Nice work! Distance measure is working

Boerny41, 2024-09-15:
I added the external definition to the device, but how exactly do I save it now?

21-Jailbraik-21, 2024-09-15:
Hello, I can't integrate it either.
Is there a better guide than Z2M?
Thanks

Mig-Chip, 2024-09-16:
Hi, how exactly did you implement this? Could you please explain where I need to insert the code so that it's recognized correctly?

Muehli-dev, 2024-09-16:
Just create a new file with the external definition via the File Editor Addon at following location:

/homeassistant/zigbee2mqtt/lumi.sensor_occupy.agl1.js

[image]

You also need to define the external converter at Zigbee2MQTTs configuration.yaml
[image]

wagaman, 2024-09-16:
> Just create a new file with the external definition via the File Editor Addon at following location:
>
> /homeassistant/zigbee2mqtt/lumi.sensor_occupy.agl1.js [...]

something wired happened on my end.

2024-09-16 17:56:18.231 WARNING (MainThread) [homeassistant.helpers.translation] Failed to load integration for translation: Integration 'external_converters' not found.
2024-09-16 17:56:20.423 ERROR (MainThread) [homeassistant.setup] Setup failed for 'external_converters': Integration not found.

Muehli-dev, 2024-09-16:
> something wired happened on my end.
>
> 2024-09-16 17:56:18.231 WARNING (MainThread) [homeassistant.helpers.translation] Failed to load integration for translation: Integration 'external_converters' [...]

Make sure you are in the mqtt configuration.yaml, not the home assistant one

[image]

wagaman, 2024-09-16:
> Make sure you are in the mqtt configuration.yaml, not the home assistant one [...]

Thank you very much :):)

Tingors, 2024-09-17:
Nice work!

jvdmeulen, 2024-09-21:
I've added key 352, that looks to be some sort of movement.
And restricted distance to two decimals

```js
exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const lumi = require('zigbee-herdsman-converters/lib/lumi');
const {manufacturerCode} = lumi;

const fzLocal = {
    fp11_presence: {
        cluster: 'manuSpecificLumi',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            console.info("msg.data=", msg.data);
            let payload = {};
            for (const key in msg.data) {
                const value = msg.data[key];
                if (key === "322") {
                    payload["presence"] = value == 1;
                }
                if (key === "351") {
                    // distance in cm, formatted to 2 decimal places
                    payload["target_distance"] = parseFloat((0.01 * value).toFixed(2));
                }
                if (key === "352") {
                    // Map the value of 352 to the corresponding string
                    const bewegingValues = ['unknown 0', 'unknown 1', 'no presence', 'movement', 'no movement'];
                    payload["movement"] = bewegingValues[value] || `unknown ${value}`;
                }
            }
            return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'Aqara',
    description: 'Presence sensor FP1E',
    extend: [lumi.modernExtend.lumiZigbeeOTA()],
    fromZigbee: [
        fzLocal.fp11_presence,
        lumi.fromZigbee.lumi_specific,
    ],
    toZigbee: [
        lumi.toZigbee.lumi_presence,
    ],
    exposes: [
        e.presence().withAccess(ea.STATE_GET),
        e.device_temperature(),
        e.power_outage_count(),
        e.numeric('target_distance', ea.STATE_GET).withUnit('m').withDescription('Target distance'),
        e.enum('movement', ea.STATE_GET, ['unknown 0', 'unknown 1', 'no presence', 'movement', 'no movement'])
          .withDescription('Beweging status van het apparaat'),
    ],
    configure: async (device, coordinatorEndpoint) => {
        device.softwareBuildID = `${device.applicationVersion}`;
        device.save();

        const endpoint = device.getEndpoint(1);
        await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
    },
    meta: {},
};

module.exports = definition;
```

jvdmeulen, 2024-09-22:
I've added max-range for detection

```js
exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const lumi = require('zigbee-herdsman-converters/lib/lumi');
const {manufacturerCode} = lumi;

const tzLocal = {
    set_max_range: {
        key: ['max_range'],
        convertSet: async (entity, key, value, meta) => {
            const intValue = Math.round(value * 100);
            await entity.write('manuSpecificLumi', {0x015B: {value: intValue, type: 0x23}}, {manufacturerCode: manufacturerCode});
            return {state: {max_range: value}};
        },
    },
};

const fzLocal = {
    fp11_presence: {
        cluster: 'manuSpecificLumi',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            console.info("msg.data=", msg.data);
            let payload = {};
            for (const key in msg.data) {
                const value = msg.data[key];
                // Logging voor specifieke keys
                const hexKeysToLog = [0x0160, 0x015A, 0x015B, 0x015C, 0x015D, 0x015E];
                if (hexKeysToLog.includes(key)) {
                    console.log(`Hex key detected: 0x${key} with value: ${value}`);
                }
                if (key == 322) { //322 / 0x0142
                    payload["presence"] = value == 1;
                }
                if (key == 351) { //351 / 0x015F
                    // distance in cm, formatted to 2 decimal places
                    payload["target_distance"] = parseFloat((0.01 * value).toFixed(2));
                }
                if (key == 352) { //352 / 0x0160
                    // Map the value of 352 to the corresponding string
                    const bewegingValues = ['unknown 0', 'unknown 1', 'no presence', 'movement', 'no movement'];
                    payload["movement"] = bewegingValues[value] || `unknown ${value}`;
                }
                if (key === 232) { // 232 / 0x00E8
                    payload["restart_device"] = Boolean(value);
                    console.log(`Restart device detected: 0x00E8 with value: ${value}`);
                }
            }
            return payload;
        },
    },
};
const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'Aqara',
    description: 'Presence sensor FP1E',
    extend: [lumi.modernExtend.lumiZigbeeOTA()],
    fromZigbee: [
        fzLocal.fp11_presence,
        lumi.fromZigbee.lumi_specific,
        lumi.fromZigbee.lumi_presence_region_events,
    ],
    toZigbee: [
        tzLocal.set_max_range,
        lumi.toZigbee.lumi_presence,
        lumi.toZigbee.lumi_monitoring_mode,
    ],
    exposes: [
        e.presence().withAccess(ea.STATE_GET),
        e.device_temperature(),
        e.power_outage_count(),
        e.numeric('target_distance', ea.STATE_GET).withUnit('m').withDescription('Target distance'),
        e.enum('movement', ea.STATE_GET, ['Unknown 0', 'Unknown 1', 'No presence', 'Movement', 'No movement'])
            .withDescription('Is someone on the move'),
        e.numeric('max_range', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(6.00)
            .withValueStep(.30)
            .withUnit('m')
            .withDescription('Adjust the slider between 0m and 6m with steps of 30cm'),
    ],
    configure: async (device, coordinatorEndpoint) => {
        device.softwareBuildID = `${device.applicationVersion}`;
        device.save();

        const endpoint = device.getEndpoint(1);
        await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
    },
    meta: {},
};

module.exports = definition;
```

hh-pirat, 2024-09-23:
> I've added max-range for detection
> 
> ```js
> exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const lumi = require('zigbee-herdsman-converters/lib/lumi');
> const {manufacturerCode} = lumi;
> 
> const tzLocal = {
>     set_max_range: {
>         key: ['max_range'],
>         convertSet: async (entity, key, value, meta) => {
>             const intValue = Math.round(value * 100);
>             await entity.write('manuSpecificLumi', {0x015B: {value: intValue, type: 0x23}}, {manufacturerCode: manufacturerCode});
>             return {state: {max_range: value}};
>         },
>     },
> };
> 
> const fzLocal = {
>     fp11_presence: {
>         cluster: 'manuSpecificLumi',
>         type: ['readResponse', 'attributeReport'],
>         convert: (model, msg, publish, options, meta) => {
>             console.info("msg.data=", msg.data);
>             let payload = {};
>             for (const key in msg.data) {
>                 const value = msg.data[key];
>                 // Logging voor specifieke keys
>                 const hexKeysToLog = [0x0160, 0x015A, 0x015B, 0x015C, 0x015D, 0x015E];
>                 if (hexKeysToLog.includes(key)) {
>                     console.log(`Hex key detected: 0x${key} with value: ${value}`);
>                 }
>                 if (key == 322) { //322 / 0x0142
>                     payload["presence"] = value == 1;
>                 }
>                 if (key == 351) { //351 / 0x015F
>                     // distance in cm, formatted to 2 decimal places
>                     payload["target_distance"] = parseFloat((0.01 * value).toFixed(2));
>                 }
>                 if (key == 352) { //352 / 0x0160
>                     // Map the value of 352 to the corresponding string
>                     const bewegingValues = ['unknown 0', 'unknown 1', 'no presence', 'movement', 'no movement'];
>                     payload["movement"] = bewegingValues[value] || `unknown ${value}`;
>                 }
>                 if (key === 232) { // 232 / 0x00E8
>                     payload["restart_device"] = Boolean(value);
>                     console.log(`Restart device detected: 0x00E8 with value: ${value}`);
>                 }
>             }
>             return payload;
>         },
>     },
> };
> const definition = {
>     zigbeeModel: ['lumi.sensor_occupy.agl1'],
>     model: 'lumi.sensor_occupy.agl1',
>     vendor: 'Aqara',
>     description: 'Presence sensor FP1E',
>     extend: [lumi.modernExtend.lumiZigbeeOTA()],
>     fromZigbee: [
>         fzLocal.fp11_presence,
>         lumi.fromZigbee.lumi_specific,
>         lumi.fromZigbee.lumi_presence_region_events,
>     ],
>     toZigbee: [
>         tzLocal.set_max_range,
>         lumi.toZigbee.lumi_presence,
>         lumi.toZigbee.lumi_monitoring_mode,
>     ],
>     exposes: [
>         e.presence().withAccess(ea.STATE_GET),
>         e.device_temperature(),
>         e.power_outage_count(),
>         e.numeric('target_distance', ea.STATE_GET).withUnit('m').withDescription('Target distance'),
>         e.enum('movement', ea.STATE_GET, ['Unknown 0', 'Unknown 1', 'No presence', 'Movement', 'No movement'])
>             .withDescription('Is someone on the move'),
>         e.numeric('max_range', ea.STATE_SET)
>             .withValueMin(0)
>             .withValueMax(6.00)
>             .withValueStep(0.30)
>             .withUnit('m')
>             .withDescription('Adjust the slider between 0m and 6m with steps of 30cm'),
>     ],
>     configure: async (device, coordinatorEndpoint) => {
>         device.softwareBuildID = `${device.applicationVersion}`;
>         device.save();
> 
>         const endpoint = device.getEndpoint(1);
>         await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
>     },
>     meta: {},
> };
> 
> module.exports = definition;
> ```

Your code with the max range also works perfectly, and I am very grateful to you. I successfully put the device into operation.

ppanagiotis, 2024-09-25:
```
exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const lumi = require('zigbee-herdsman-converters/lib/lumi');
const {manufacturerCode} = lumi;

const tzLocal = {
    set_max_range: {
        key: ['max_range'],
        convertSet: async (entity, key, value, meta) => {
            const intValue = Math.round(value * 100);
            await entity.write('manuSpecificLumi', {0x015B: {value: intValue, type: 0x23}}, {manufacturerCode: manufacturerCode});
            return {state: {max_range: value}};
        },
    },
};

const fzLocal = {
    fp11_presence: {
        cluster: 'manuSpecificLumi',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            console.info("msg.data=", msg.data);
            let payload = {};
            for (const key in msg.data) {
                const value = msg.data[key];
                // Logging voor specifieke keys
                const hexKeysToLog = [0x0160, 0x015A, 0x015B, 0x015C, 0x015D, 0x015E];
                if (hexKeysToLog.includes(key)) {
                    console.log(`Hex key detected: 0x${key} with value: ${value}`);
                }
                if (key == 322) { //322 / 0x0142
                    payload["presence"] = value == 1;
                }
                if (key == 351) { //351 / 0x015F
                    // distance in cm, formatted to 2 decimal places
                    payload["target_distance"] = parseFloat((0.01 * value).toFixed(2));
                }
                if (key == 352) { //352 / 0x0160
                    // Map the value of 352 to the corresponding string
                    const bewegingValues = ['unknown 0', 'unknown 1', 'no presence', 'movement', 'no movement'];
                    payload["movement"] = bewegingValues[value] || `unknown ${value}`;
                }
                if (key === 232) { // 232 / 0x00E8
                    payload["restart_device"] = Boolean(value);
                    console.log(`Restart device detected: 0x00E8 with value: ${value}`);
                }
            }
            return payload;
        },
    },
};
const definition = {
    zigbeeModel: ['lumi.sensor_occupy.agl1'],
    model: 'lumi.sensor_occupy.agl1',
    vendor: 'Aqara',
    description: 'Presence sensor FP1E',
    extend: [lumi.modernExtend.lumiZigbeeOTA()],
    fromZigbee: [
        fzLocal.fp11_presence,
        lumi.fromZigbee.lumi_specific,
        lumi.fromZigbee.lumi_presence_region_events,
    ],
    toZigbee: [
        tzLocal.set_max_range,
        lumi.toZigbee.lumi_presence,
        lumi.toZigbee.lumi_monitoring_mode,
    ],
    exposes: [
        e.presence().withAccess(ea.STATE_GET),
        e.device_temperature(),
        e.power_outage_count(),
        e.numeric('target_distance', ea.STATE_GET).withUnit('m').withDescription('Target distance'),
        e.enum('movement', ea.STATE_GET, ['Unknown 0', 'Unknown 1', 'No presence', 'Movement', 'No movement'])
            .withDescription('Is someone on the move'),
        e.numeric('max_range', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(6.00)
            .withValueStep(0.30)
            .withUnit('m')
            .withDescription('Adjust the slider between 0m and 6m with steps of 30cm'),
        e
            .enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high'])
            .withDescription('Different sensitivities means different static human body recognition rate and response speed of occupied'),

    ],
    configure: async (device, coordinatorEndpoint) => {
        device.softwareBuildID = `${device.applicationVersion}`;
        device.save();

        const endpoint = device.getEndpoint(1);
        await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
    },
    meta: {},
};

module.exports = definition;
```

added motion_sensitivity setting. I think it works :/

melhaipau, 2024-09-25:
where exactly do i define the external converter?  What line, as every time
 i try add it in the z2mqtt config file get bad indentation errors 
Thanks

JustNZ, 2024-09-25:
> Just create a new file with the external definition via the File Editor Addon at following location:
>
> /homeassistant/zigbee2mqtt/lumi.sensor_occupy.agl1.js [...]

@melhaipau checkout this comment here and the one below on that

melhaipau, 2024-09-25:
> [> Just create a new file with the external definition via the File Editor Addon at following location:
> 
> > /homeassistant/zigbee2mqtt/lumi.sensor_occupy.agl1.js
> > ![image](https://private-user-images.githubusercontent.com/10485531/367804338-24a4b09b-a5f9-4f6e-a2cf-4916b620260e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyOTQ4OTAsIm5iZiI6MTcyNzI5NDU5MCwicGF0aCI6Ii8xMDQ4NTUzMS8zNjc4MDQzMzgtMjRhNGIwOWItYTVmOS00ZjZlLWEyY2YtNDkxNmI2MjAyNjBlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI1VDIwMDMxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAxYTBlZjdjZGRhNGZmYzFhNzMyZjdhOGZlNzM1YmExMGI5MGRlMThjMDc3YmMxYTA2YzM5YWQwZWFiNjQzMGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bPvp4G7g4fdLkEAvSVsxO4hZiVde9JuL9-Ww1yXFLqM)
> > You also need to define the external converter at Zigbee2MQTTs configuration.yaml ![image](https://private-user-images.githubusercontent.com/10485531/367805384-ab980972-6db1-4c18-afdf-0b7770d16957.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyOTQ4OTAsIm5iZiI6MTcyNzI5NDU5MCwicGF0aCI6Ii8xMDQ4NTUzMS8zNjc4MDUzODQtYWI5ODA5NzItNmRiMS00YzE4LWFmZGYtMGI3NzcwZDE2OTU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI1VDIwMDMxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmOTQ5ZTJmMjg5ZTk4ZWE5MTY1ZjA1YWFmYmViNmZlOGIxNGJiNjFkMDI4N2JmOTQyODQyZjcyMzE2MDkwMmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SWJu3QnxI1qkazNgQ-cVFhP_76Cw5_VOhQCvHeYYtWM)]([#23577 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/23577#issuecomment-2353137234))
> 
> @melhaipau that comment here should describe it very well

Hi @JustNZ  thanks for the reply i have already done that 
but when defining the external converter in the config yaml i dont know where to write it

JustNZ, 2024-09-25:
Hey @melhaipau,
so I just added the following lines to my configuration.yaml in the zigbee2mqtt folder (be sure that the .js is in the same folder like the configuration.yaml):
```
external_converters:
  - lumi.sensor_occupy.agl1.js
```

you have there another `devices` block defined within the `external_converters`. Not sure about that one but mine above those work like that.

melhaipau, 2024-09-25:
Thank you i will try that

ianpogi5, 2024-09-25:
> ```
> exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
>
> const lumi = require('zigbee-herdsman-converters/lib/lumi');
> const [...]

I get this error when changing sensitivity:

[image]

ppanagiotis, 2024-09-26:
> > ```
> > exposes = require('zigbee-herdsman-converters/lib/exposes');
> > const e = exposes.presets;
> > const ea = exposes.access;
> > 
> > const lumi = require('zigbee-herdsman-converters/lib/lumi');
> > const {manufacturerCode} = lumi;
> > 
> > const tzLocal = {
> >     set_max_range: {
> >         key: ['max_range'],
> >         convertSet: async (entity, key, value, meta) => {
> >             const intValue = Math.round(value * 100);
> >             await entity.write('manuSpecificLumi', {0x015B: {value: intValue, type: 0x23}}, {manufacturerCode: manufacturerCode});
> >             return {state: {max_range: value}};
> >         },
> >     },
> > };
> > 
> > const fzLocal = {
> >     fp11_presence: {
> >         cluster: 'manuSpecificLumi',
> >         type: ['readResponse', 'attributeReport'],
> >         convert: (model, msg, publish, options, meta) => {
> >             console.info("msg.data=", msg.data);
> >             let payload = {};
> >             for (const key in msg.data) {
> >                 const value = msg.data[key];
> >                 // Logging voor specifieke keys
> >                 const hexKeysToLog = [0x0160, 0x015A, 0x015B, 0x015C, 0x015D, 0x015E];
> >                 if (hexKeysToLog.includes(key)) {
> >                     console.log(`Hex key detected: 0x${key} with value: ${value}`);
> >                 }
> >                 if (key == 322) { //322 / 0x0142
> >                     payload["presence"] = value == 1;
> >                 }
> >                 if (key == 351) { //351 / 0x015F
> >                     // distance in cm, formatted to 2 decimal places
> >                     payload["target_distance"] = parseFloat((0.01 * value).toFixed(2));
> >                 }
> >                 if (key == 352) { //352 / 0x0160
> >                     // Map the value of 352 to the corresponding string
> >                     const bewegingValues = ['unknown 0', 'unknown 1', 'no presence', 'movement', 'no movement'];
> >                     payload["movement"] = bewegingValues[value] || `unknown ${value}`;
> >                 }
> >                 if (key === 232) { // 232 / 0x00E8
> >                     payload["restart_device"] = Boolean(value);
> >                     console.log(`Restart device detected: 0x00E8 with value: ${value}`);
> >                 }
> >             }
> >             return payload;
> >         },
> >     },
> > };
> > const definition = {
> >     zigbeeModel: ['lumi.sensor_occupy.agl1'],
> >     model: 'lumi.sensor_occupy.agl1',
> >     vendor: 'Aqara',
> >     description: 'Presence sensor FP1E',
> >     extend: [lumi.modernExtend.lumiZigbeeOTA()],
> >     fromZigbee: [
> >         fzLocal.fp11_presence,
> >         lumi.fromZigbee.lumi_specific,
> >         lumi.fromZigbee.lumi_presence_region_events,
> >     ],
> >     toZigbee: [
> >         tzLocal.set_max_range,
> >         lumi.toZigbee.lumi_presence,
> >         lumi.toZigbee.lumi_monitoring_mode,
> >     ],
> >     exposes: [
> >         e.presence().withAccess(ea.STATE_GET),
> >         e.device_temperature(),
> >         e.power_outage_count(),
> >         e.numeric('target_distance', ea.STATE_GET).withUnit('m').withDescription('Target distance'),
> >         e.enum('movement', ea.STATE_GET, ['Unknown 0', 'Unknown 1', 'No presence', 'Movement', 'No movement'])
> >             .withDescription('Is someone on the move'),
> >         e.numeric('max_range', ea.STATE_SET)
> >             .withValueMin(0)
> >             .withValueMax(6.00)
> >             .withValueStep(0.30)
> >             .withUnit('m')
> >             .withDescription('Adjust the slider between 0m and 6m with steps of 30cm'),
> >         e
> >             .enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high'])
> >             .withDescription('Different sensitivities means different static human body recognition rate and response speed of occupied'),
> > 
> >     ],
> >     configure: async (device, coordinatorEndpoint) => {
> >         device.softwareBuildID = `${device.applicationVersion}`;
> >         device.save();
> > 
> >         const endpoint = device.getEndpoint(1);
> >         await endpoint.read('manuSpecificLumi', [0x0142], {manufacturerCode: manufacturerCode});
> >     },
> >     meta: {},
> > };
> > 
> > module.exports = definition;
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > added motion_sensitivity setting. I think it works :/
> 
> I get this error when changing sensitivity:
> 
> ![Screenshot from 2024-09-26 05-37-58](https://private-user-images.githubusercontent.com/1063875/370904018-05d27da4-b47a-4821-bc53-970cce3f3a78.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczMzY3MzAsIm5iZiI6MTcyNzMzNjQzMCwicGF0aCI6Ii8xMDYzODc1LzM3MDkwNDAxOC0wNWQyN2RhNC1iNDdhLTQ4MjEtYmM1My05NzBjY2UzZjNhNzgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjZUMDc0MDMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWFlYWNkMmJmMzk5NDk1ZTY1YjJiYjE3OTMwOWI2NTlkMzNlYTkzMDNjYmMwM2E5ZDRiMTMwODQzNGIwZTMxNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Ic_UVx7z2RF8bhdT-5Y961z8WgIaAah7GIRF6-pF9H8)

Please add also:
```
    toZigbee: [
        tzLocal.set_max_range,
        lumi.toZigbee.lumi_presence,
        lumi.toZigbee.lumi_monitoring_mode,
	lumi.toZigbee.lumi_motion_sensitivity,
    ],
```

ianpogi5, 2024-09-26:
> 
> Please add also:
> 
> ```
>     toZigbee: [
>         tzLocal.set_max_range,
>         lumi.toZigbee.lumi_presence,
>         lumi.toZigbee.lumi_monitoring_mode,
> 	lumi.toZigbee.lumi_motion_sensitivity,
>     ],
> ```

Now it works! Great job @ppanagiotis !

morgendagen, 2024-09-27:
I finally got hold of an Aqara hub and will make an effort to create a proper integration during the weekend. Already sniffed the command to initiate AI spatial learning :-)

craigwitter, 2024-09-27:
I realize this comment adds nothing to the thread, but I wanted you to know
that you are officially my hero.

Craig

jvdmeulen, 2024-09-27:
I'm not 100% certain this work, since i'm very new to this, but i think the following should trigger a ai-learning
I don't know how to verify that it's actually working 

replace 'toZigbee', i've add the object with "key"
```js
toZigbee: [
            {
                key: ['reset_presence'],
                convertSet: async (entity, key, value, meta) => {
                    await entity.write('manuSpecificLumi', {0x0157: {value: 1, type: 0x20}});
                    return {state: {reset_presence: true}};
                },
        },
        tzLocal.set_max_range,
        lumi.toZigbee.lumi_presence,
        lumi.toZigbee.lumi_monitoring_mode,
	lumi.toZigbee.lumi_motion_sensitivity,
     ]
```

and in 'exposes' add
```js
e.binary('reset_presence', ea.STATE_SET, 'PRESS', 'RELEASE').withDescription('Reset Presence (press to reset)'),
```

ferbulous, 2024-09-28:
Thanks @jvdmeulen 
I do have to ask what does ‘release’ do?
‘Press’ does seem to start the ai spatial learning and no longer getting false presence from my ceiling fans and aircon

[image]

morgendagen, 2024-09-28:
> Thanks @jvdmeulen I do have to ask what does ‘release’ do? ‘Press’ does seem to start the ai spatial learning and no longer getting false presence from my ceiling fans and aircon

Z2M does not seem to have any way to show a momentary button for invoking an action (e.g. writing a value to the device). One work-around is to use a binary toggle switch. In this instance the device will start spatial learning when flipping the switch either way.

Good to hear that spatial learning is indeed working - haven't had a chance to test it out yet.

marazmarci, 2024-09-28:
> Z2M does not seem to have any way to show a momentary button for invoking an action (e.g. writing a value to the device). One work-around is to use a binary toggle switch. [...]

Z2M can expose a device action as a button, check out the `Identify` button for devices supporting it (for example basically all IKEA devices)!

And also, there's a (weirdly displayed, but working) button action present for FP1(not E) in Z2M called `Reset nopresence status`:
[image]

morgendagen, 2024-09-29:
> Z2M can expose a device action as a button, check out the `Identify` button for devices supporting it (for example basically all IKEA devices)!

That acts more like a latching button, i.e. it doesn't change back to its original state when depressed. Identify and the spatial learning on the FP1E are fire-and-forget.

> And also, there's a (weirdly displayed, but working) button action present for FP1(not E) in Z2M called `Reset nopresence status`

I have settled on using an action button like the one used for identification for the "Reset device" and "Spatial learning" actions in the PR I have been working on.

[image]

jvdmeulen, 2024-09-29:
> I have settled on using an action button like the one used for identification for the "Reset device" and "Spatial learning" actions in the PR I have been working on.

I see you have the valueStep at 0.3, but i think it also works without that. (at least here it looks like it works on other values), if you can verify you might/can leave the stepping out.

closed 2024-09-30 by Koenkk

ferbulous, 2024-10-01:
Hi, now that it's merged how do I get the device to work?
I excluded the external converter file and used z2m edge to add the fp1e but it shows up as unsupported

melhaipau, 2024-10-02:
> Hi, now that it's merged how do I get the device to work? I excluded the external converter file and used z2m edge to add the fp1e but it shows up as unsupported

I just updated my HA and updated Z2MQTT and it worked once it was merged. I couldn’t get it to work the manual way

noharminthat, 2024-10-30:
I’m having an issue with presence where it’s not updating unless I hit the refresh button next to the entity. Movement and target distance work fine though. It does seem to correct itself eventually, but I haven’t monitored it to see exactly how long it takes to self correct. Any ideas? 

[image]

gForce1989, 2024-12-03:
Same problem here.
Often it works great. But sometimes it doesn't work. Distance and movement is recognized but presence stuck on false.

chpio, 2024-12-21:
Might just be the generally very poor zigbee implementation of the aqara lineup. Im also having issues with their buttons/switches, sometimes needing multiple presses to register. Thought they might have fixed that for the FP1E, being an ac powered device, but it seems i've just wasted a lot of money for a broken device. The sensor is really good, but the connectivity issues...

nexus-labs-admin, 2025-04-01:
Same stuff as described @noharminthat is happening to me all the time. I use it as a sensor in my workroom so I notice every day. I did not found any pattern for it, tho. 

Is it something that can be fixed on zigbee2mqtt side or is it an vendor (Aquara) issue?

witooski, 2025-05-09:
Same issue... Restarting the device helps, but its very annoying. I used to have FP1 and never experienced such issue.