Koenkk/zigbee2mqtt #23581: Zemismart ZN2S-RS3E-DH [_TZE284_znvwzxkq]

tuongbui15, 2024-08-10:
### What happened?

Zemismart ZN2S-RS3E-DH has two version: [_TZE284_znvwzxkq] and [_TZE204_znvwzxkq].
Version [_TZE204_znvwzxkq] has been supported #21940

Can you please try include [_TZE204_znvwzxkq] device as a supported device in the Z2M?
Please let me know if you need any further information
[image]

### What did you expect to happen?

Add new supported device

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

Pi4 docker

Koenkk, 2024-08-12:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-12 by Koenkk

tuongbui15, 2024-08-12:
> Added!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Thank you so much.!
Have a nice day.

tuongbui15, 2024-08-14:
> Added!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I want to add one more switch for backlight mode (turn on when sunrise or turn off when sunset).
Thankyou

columnmn, 2025-03-24:
Do you guys happen to have a copy of the converter for this one? I have a similar one from zemismart I'm trying to get working, but suck at the coding part.