Koenkk/zigbee2mqtt #23584: IKEA E2001/E2002 FW2.4.16 - Controlling a group doesn't work any more since upgrade FW

Utopia69, 2024-08-10:
### What happened?

I updated the firmware of a IKEA E2001/E2002 from 2.4.11 to 2.4.16
Since this upgrade I can't control a group binded to the IKEA E2001/E2002 
Unbinding/binding does work, just the buttons don't seems to work
When the switch is binded to 1 or more devices there is no problem

### What did you expect to happen?

Controlling group of lights as it was working with FW2.4.16

### How to reproduce it (minimal and precise)

Update firmware IKEA E2001/E2002 from 2.4.11 to 2.4.16
Bind group

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet Zigbee dongle Plus

### Setup

Add-on in HA

### Debug log

Logs look normal, no errors

`info 2024-08-10 14:38:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sturing Keuken Spots', payload '{"action":"on","battery":100,"device":{"applicationVersion":1,"dateCode":"20240708","friendlyName":"Sturing Keuken Spots","hardwareVersion":1,"ieeeAddr":"0x385cfbfffecf07a2","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":63430,"powerSource":"Battery","softwareBuildID":"2.4.16","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-08-10T14:38:41+02:00","linkquality":176,"update":{"installed_version":33816598,"latest_version":33816598,"state":"idle"}}'
info 2024-08-10 14:38:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sturing Keuken Spots', payload '{"action":"","battery":100,"device":{"applicationVersion":1,"dateCode":"20240708","friendlyName":"Sturing Keuken Spots","hardwareVersion":1,"ieeeAddr":"0x385cfbfffecf07a2","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":63430,"powerSource":"Battery","softwareBuildID":"2.4.16","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-08-10T14:38:41+02:00","linkquality":176,"update":{"installed_version":33816598,"latest_version":33816598,"state":"idle"}}'
info 2024-08-10 14:38:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sturing Keuken Spots/action', payload 'on'
info 2024-08-10 14:38:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sturing Keuken Spots', payload '{"action":"off","battery":100,"device":{"applicationVersion":1,"dateCode":"20240708","friendlyName":"Sturing Keuken Spots","hardwareVersion":1,"ieeeAddr":"0x385cfbfffecf07a2","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":63430,"powerSource":"Battery","softwareBuildID":"2.4.16","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-08-10T14:38:42+02:00","linkquality":180,"update":{"installed_version":33816598,"latest_version":33816598,"state":"idle"}}'
info 2024-08-10 14:38:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sturing Keuken Spots', payload '{"action":"","battery":100,"device":{"applicationVersion":1,"dateCode":"20240708","friendlyName":"Sturing Keuken Spots","hardwareVersion":1,"ieeeAddr":"0x385cfbfffecf07a2","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":63430,"powerSource":"Battery","softwareBuildID":"2.4.16","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-08-10T14:38:42+02:00","linkquality":180,"update":{"installed_version":33816598,"latest_version":33816598,"state":"idle"}}'
info 2024-08-10 14:38:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Sturing Keuken Spots/action', payload 'off'`

newlund, 2024-08-11:
I have the same problem since upgrading to 2.4.16

Colbyjdx, 2024-08-14:
Exact same issue here. Binding to lights individually still works, but not groups.

Koenkk, 2024-08-14:
Probably this is an issue of the firmware itself (maybe they removed group binding functionality). Very likely not something that can be solved from z2m

mmaraa, 2024-08-18:
Exactly same issue here. 😒 Have to make some work around or change switches or wait new fw.

b2un0, 2024-08-23:
read the docs at "Latest Firmware"
https://www.zigbee2mqtt.io/devices/E2001_E2002.html

> Note that the remote loses the capability to bind to a group in the latest firmware.

jamesremuscat, 2024-08-23:
> read the docs at "Latest Firmware" https://www.zigbee2mqtt.io/devices/E2001_E2002.html
> 
> > Note that the remote loses the capability to bind to a group in the latest firmware.

That note has been in there for several firmware versions. Looks like the latest release from Ikea has made it accurate again ;)

ireun, 2024-09-28:
Can you somehow downgrade the FW? This remote is unusable at seems that no FW fix came for two months already. If not - I'm planning to return it to IKEA and buy another one, hopefully with older FW.

JohnnyWestlake, 2024-11-17:
> Probably this is an issue of the firmware itself (maybe they removed group binding functionality). Very likely not something that can be solved from z2m

I find this doubtful given it still works when setting up the groups using a Dirigera hub; if you do so and then turn off the hub, all the devices are still properly bound and controllable directly from the remote.

newlund, 2025-02-02:
> > Probably this is an issue of the firmware itself (maybe they removed group binding functionality). Very likely not something that can be solved from z2m
> 
> I find this unlikely given it still works when setting up the groups using a Dirigera hub; if you do so and then turn off the hub, all the devices (a random assortment of different non-ikea zigbee bulbs in my test case) are still properly bound and controllable directly from the remote, so it definitely still works with ordinary ZigBee groups somehow.

If it still works using the Dirigera hub I guess it is something that could be fixed in Z2M?
Posting this to remove the stale label :)

Utopia69, 2025-04-04:
Should not stale, problem still exits

autourdupc, 2025-05-11:
Ikea E2001/E2002 firmware 2.4.17 can't be set in a group... Need it !

nielsek, 2025-08-30:
not stale

ambidexi, 2025-09-30:
> Can someone write a guide on **How to downgrade from 2.4.16 to 2.4.11**? That seems to be the solution.

I am exactly here. Another issue has a guide on this:
https://github.com/Koenkk/zigbee2mqtt/discussions/23243#discussioncomment-10038494

Downgrading to 2.4.11 first, I think that was the factory firmware. According to the linked thread, it has other issues... We'll see.

Here is the link for the image:
http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/styrbar_dimmer_zingo-0x11cb-2.4.11-prod.ota.ota.signed

autourdupc, 2025-11-30:
Don't stale this issue !
We need a solution for this issue !