Koenkk/zigbee2mqtt #23586: [New device support]: TS0222_TZ3000_8uxxzz4b - Moes Tuya Light intensity sensor

kikejherrera, 2024-08-10:
### Link

https://es.aliexpress.com/item/1005007359584118.html

### Database entry

{"id":126,"type":"EndDevice","ieeeAddr":"0xa4c13880fec5e655","nwkAddr":17330,"manufId":4417,"manufName":"_TZ3000_8uxxzz4b","powerSource":"Battery","modelId":"TS0222","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[1,3,1026,1029,1024,1280,0],"outClusterList":[3,25,10],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":24610}},"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_8uxxzz4b","zclVersion":3,"appVersion":67,"modelId":"TS0222","powerSource":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x804b50fffea7ecd3","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1723313361630}

### Comments

Hello
I've been trying to add this device all day, I haven't seen it in the developer version changes. It is a Moes / Tuya brightness sensor.
I followed the steps in the guide to add devices, got the datapoints and generated the external definition. The device generated fine and was recognised but the values did not update.
I checked that there were already existing converters for other models of the same device (TS0222 Light sensor) and I modified the database.db changing _TZ3000_8uxxzz4b for the two existing models (_TYZB01_4mdqxxnn and _TYZB01_m6ec2pgj), I don't know the reason but this didn't work and it appeared "device not supported".
Finally I have generated a new external definition which is a mix between the existing tuya.ts and the first converter with the datapoints, and it works!!, it recognizes the values and even the battery value, which in the previous attempts did not appear.
I don't know which parts are left over, I've been trying to do something apparently simple and without having enough knowledge.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const legacy = require('zigbee-herdsman-converters/lib/legacy')
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {modelID: 'TS0222', manufacturerName: '_TYZB01_4mdqxxnn'},
        {modelID: 'TS0222', manufacturerName: '_TYZB01_m6ec2pgj'},
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0222',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_8uxxzz4b',
        },
    ],
    model: 'TS0222',
    vendor: 'Tuya',
    description: 'Light intensity sensor',
    fromZigbee: [fz.battery, fz.illuminance, legacy.fromZigbee.TS0222],
    toZigbee: [],
    exposes: [e.battery(), e.illuminance(), e.illuminance_lux()],
    configure: tuya.configureMagicPacket,
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [2, 'illuminance_lux', tuya.valueConverter.raw], // <- mapped the datapoint
            [4, 'battery', tuya.valueConverter.raw],             
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

budcalloway, 2024-08-11:
I exactly face the same issue at the same time :)

gdave321, 2024-08-12:
I happen to have received this exact sensor 2 days ago.
This patch made it work for me, I simply looked for an existing tuya light sensor and added a fingerprint. With this it immediately starting working and battery (which took few hours to show up) and illuminance / lux values match just fine with a hue motion sensor in the same room, i.e. values reported seem to be correct.

[patch4_TZ3000_8uxxzz4b.gz](https://github.com/user-attachments/files/16579306/patch4_TZ3000_8uxxzz4b.gz)

budcalloway, 2024-08-12:
> I happen to have received this exact sensor 2 days ago. This patch made it work for me, I simply looked for an existing tuya light sensor and added a fingerprint. With this it immediately starting [...]

Thanks gdave. Its seems to be the best solution, but I don't know how to use it. Could you guide me/us ? Thanks !

gdave321, 2024-08-13:
(G)unzip it and apply using patch command. 
Or simply add the line:
  { modelID: 'TS0222', manufacturerName: '_TZ3000_8uxxzz4b' },
just below the the existing fingerprint:
  { modelID: 'TS0222', manufacturerName: '_TYZB01_m6ec2pgj' },
in the file tuya.js  ( node_modules/zigbee-herdsman-converters/devices/tuya.js )
and restart.
If that works, somebody has to submit an official change to get in the next z2m release.

kikejherrera, 2024-08-13:
I think it is easier, before touching system files and while @Koenkk , if he considers it appropriate, adds the support in next versions, to put the file with the external converter in the same folder as the configuration.yalm and add it in the list of converters in z2m/settings/external converters.

 link here the file that I mentioned in the first post. You have to change the extension from .txt to .js

[TS0222.txt](https://github.com/user-attachments/files/16596737/TS0222.txt)

budcalloway, 2024-08-13:
Thank you both.
I am using HassOS and could not modify system files.
I used the txt file for external converter and worked well. I don't know why the first one did not worked.
I see now the measures but some features are missing such as temperature and humidity (not of us for me).
Will take a deeper look to understand how to build such file.

kikejherrera, 2024-08-13:
> Thank you both. I am using HassOS and could not modify system files. I used the txt file for external converter and worked well. I don't know why the first one did not worked. I see now the measures [...]

I think this device does not have temperature and humidity, at least in the features of the device in the Tuya cloud they do not appear, although looking at the design I have my doubts if internally there is a sensor but it is not enabled.

budcalloway, 2024-08-13:
> I think this device does not have temperature and humidity, at least in the features of the device in the Tuya cloud they do not appear, although looking at the design I have my doubts if internally [...]

OK. I was missinterpreting the generated_external_definition which stated humidity and temperature.
Well, I thank you again. This is perfectly working ! I still don't understand why it was not OK at first.

nmaster2042, 2024-08-13:
I ordered one to test.

tribakzero, 2024-08-14:
> I think it is easier, before touching system files and while @Koenkk , if he considers it appropriate, adds the support in next versions, to put the file with the external converter in the same folder as the configuration.yalm and add it in the list of converters in z2m/settings/external converters.
> 
> I link here the file that I mentioned in the first post. You have to change the extension from .txt to .js
> 
> [TS0222.txt](https://github.com/user-attachments/files/16596737/TS0222.txt)

Definitely the best approach, thanks!

bastgau, 2024-09-04:
Hello.

It seems that LQI value is not reported :(

[image]

**Zigbee2MQTT version : ** [1.40.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.0) commit: [6fa1dab](https://github.com/Koenkk/zigbee2mqtt/commit/6fa1dab)
**Coordinator type : ** zStack3x0
**Coordinator revision :** 20230507
**Frontend version** : 0.7.4
**zigbee-herdsman-converters version** : 20.8.4
**zigbee-herdsman version** : 0.57.1

nmaster2042, 2024-09-04:
@bastgau : what converter did you used ?
I just received mine.

bastgau, 2024-09-04:
> @bastgau : what converter did you used ?
> I just received mine.

This one : https://github.com/user-attachments/files/16596737/TS0222.txt

nmaster2042, 2024-09-05:
@bastgau : thank you. 
I just tried, no LQI here neither.
Stable zigbee2mqtt stable 1.4.0

bastgau, 2024-09-05:
Today, the LQI appears. Maybe it was a temporary issue just after pairing the device.

nmaster2042, 2024-09-05:
OK will try waiting.

fabianosan, 2024-09-06:
Works fine here, thank you!

[image]

[image]

nmaster2042, 2024-09-08:
> Works fine here, thank you!
>
> image.png (view on web)
>
> image.png (view on web)

Lqi is now avaliable here too.
It needed somme time to update.

nmaster2042, 2024-09-12:
I don't now for you but here it's fine since last message.

I thinky it coule be added in z2m.

closed 2024-12-23 by github-actions[bot]