Koenkk/zigbee2mqtt #23588: Namron 4512721 on/off l3 and on/off l4 do not work anymore

larronni, 2024-08-11:
### What happened?

After upgrading to 1.39.1 and changing to EmberZNet coordinator (from a ZZH), it stopped working

### What did you expect to happen?

Work as before

### How to reproduce it (minimal and precise)

Include the Namron K8
Press on/off on L3 and L4

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

EFR32MG21

### Setup

Add-on on HomeAssistant on a NUC

### Debug log

2024-08-11 11:21:50zh:ember: Delivery of MULTICAST failed for '65533' [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":4,"destinationEndpoint":255,"options":256,"groupId":9732,"sequence":123} messageTag=255]
2024-08-11 11:21:52zh:ember: Error: Failed to register group '9731' in multicast table with status=INVALID_STATE.
2024-08-11 11:21:57zh:ember: Error: Failed to register group '9731' in multicast table with status=INVALID_STATE.
2024-08-11 11:21:58zh:ember: Error: Failed to register group '9732' in multicast table with status=INVALID_STATE.

larronni, 2024-08-16:
Tested with ZZH adapter - works as it should. So it looks like this is something to do with EFR32MG21 Silicon adapter..

Nerivec, 2024-08-16:
What's the adapter specifically?
`INVALID_STATE` is unfortunately a very generic error. Could be the table of the adapter is full. I'd need the full `debug` logs from Z2M start to when that happens, to confirm (to see how many of your devices force to register groups).

The delivery failure could be interference-related though. I'll be pushing a fix in dev in the next few days to improve interference handling with `ember`.

larronni, 2024-08-18:
Think you are on to something. When I restarted Z2M, it started to work again... But now I get an error when adding some other device...  (I will create a new issue for this device/error, not sure if it is related or not...)
Attached is the debug log from startup

[new-log.log](https://github.com/user-attachments/files/16647777/new-log.log)

Nerivec, 2024-08-18:
From what I saw in your other issue, you do seem to be having some troubles with interferences. You may want to try again with latest dev once the improvements I mentioned before are merged (should be within the next couple of days).

As for the table issue, I only see 7 registrations in the logs your provided. It should be capable of more than that with most firmware (16).
Can you confirm which firmware you are using? Try this one if it's different from yours: https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

_Something a bit strange, the ID (in serial path) of your Dongle-E is using the format usually reserved for the Dongle-P (zstack). If you can, try removing the couple of screws on the side of the adapter casing, and sliding it off, until you see the version and date printed on the circuit board._

closed 2024-12-23 by github-actions[bot]