Koenkk/zigbee2mqtt #23589: EROFS: read-only file system, open '/app/data/configuration.yaml' even though no write is necessary

Barsonax, 2024-08-11:
### What happened?

Whenever I try to rename or change a setting of a device I get: `EROFS: read-only file system, open '/app/data/configuration.yaml'` even though I have these in the config:
```
devices: devices.yml
groups: groups.yml 
```
Even though I get the error I do see devices.yml get updated when I for instance rename a device. 

### What did you expect to happen?

I would not expect any writes to configuration.yaml

### How to reproduce it (minimal and precise)

- Make sure `devices: devices.yml` is in your config
- Mount configuration.yaml as a kubernetes configmap
- Rename a device
- An error will popup

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

 20210708

### Adapter

zStack3x0

### Setup

kubernetes, x86

ruifung, 2024-08-18:
In the meantime, my workaround is to specify all manual configuration via environment variables, and let it write to configuration.yaml

Barsonax, 2024-08-18:
Another workaround would be using an init container to copy the config file to its destination so it's not readonly anymore.

ruifung, 2024-08-19:
Well, that would be suboptimal since if z2m writes anything to the file it's lost on pod restart.

lkleen, 2024-10-30:
I'm using the configs section in docker compose to populate container configurations, which is a clean way to do so. As specified by docker compose it is not possible to set a configuration writable: https://docs.docker.com/reference/compose-file/services/#configs

You have no chance to do a proper configuration here and I've never seen another container image which is expecting write access to config files.

djurny, 2024-10-30:
Hi there,
z2m writes to `configuration.yaml` when you change options in the frontend for any device, or if you pair/join new devices. So even if you in your usecase do not expect to pair any devices or use the frontend to change any setting of any device, most likely any other usecase does change settings or pair devices using either MQTT or the z2m frontend.
Is this a semantic issue perhaps? 
Groetjes,

Barsonax, 2024-10-31:
> Hi there,
> z2m writes to `configuration.yaml` when you change options in the frontend for any device, or if you pair/join new devices. So even if you in your usecase do not expect to pair any devices [...]

This isn't the case with the config which I supplied where the writes will be done to separate files.

rlabrecque, 2024-11-17:
> I think it is a semantic issue as config and app data are not clearly separated.

Agreed, I found this thread because I was trying to seperate these in my own setup.

What I roughly ended up doing is:

1. Mount a config directory containing my base configuration.yaml and an entrypoint.sh to /app/config; and mount zigbee2mqtt's /app/data, out to a different directory on the host
2. Override the docker entrypoint to run my entrypoint.sh
3. Have that entrypoint.sh copy the configuration.yaml from /app/config over to /app/data IF it doesn't already exist; and then run the regular entrypoint.

This seems like it would be a fairly easy change to make touching just these two files:

https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/docker-compose.yml

```
volumes:
            - ./data:/app/data
            - ./config:/app/config:ro
```

https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/docker-entrypoint.sh

```
if [ ! -f "$DATA/configuration.yaml" ]; then
    if [ -f "/app/config/configuration.yaml" ]; then
        echo "Copyingconfiguration file..."
        cp /app/config/configuration.yaml "$DATA/configuration.yaml"
    else
        echo "Creating configuration file..."
        cp /app/configuration.example.yaml "$DATA/configuration.yaml"
    fi
fi
```

I don't think this satisfies this ticket completely; but feels like an improvement.

After writing all this I might actually just try mounting `- ./config/configuration.yaml:/app/configuration.example.yaml"` to override the default example.

spacekitteh, 2024-12-11:
I get this just for a `devices.yaml` file. Since I'm not using docker, I can't use this fix :(

Is there any reason why it would need write access to `devices.yaml`?

Barsonax, 2024-12-11:
I tested some more and what zigbee2mqtt is doing is reformating configuration.yaml and replacing any values with their actual values from the env variables. This happens when you make any kind of change from the UI.

It doesn't matter if you put devices and groups in separate files.

This seems unnecessary and breaks any scenario where this file is read-only such as read-only filesystems or config maps.

tardich, 2025-01-15:
I have this error as well, but using zigbee2mqtt as a kubernetes pod from the helm chart.  The configuration.yaml is supposed created, in this case, from the configmap created from the different options, then I see, in the storage, that the resulting configuration.yaml is empty (size=0), but there's an existing configuration_backup_v1.yaml that contains the content of the configuration.yaml I'm expecting. At this point, I have no clue on what to do....  :-/

iSenne, 2025-01-17:
I ran into this issue when upgrading to **Zigbee2MQTT 2.0.0** in Kubernetes, where my configuration.yaml was mounted as a ConfigMap. The error occurred because z2m attempted to update the configuration file, but since it was read-only, the migration failed.

I initially tried the init container workaround, and during that process, I noticed that z2m was attempting to migrate configuration.yaml to version 4, which involved removing legacy settings and restructuring some objects. However, it couldn't perform the migration because my config was read-only.

The solution for me was:

- Let z2m perform the migration once (using a writable config).
- Extract the updated configuration.yaml after migration.
- Ensure that "version: 4" is present in the new config.
- Update my ConfigMap with the migrated config.
- Restart z2m with the updated ConfigMap.

After doing this, I was able to keep my config as a read-only ConfigMap without any further issues.

Hope this helps anyone facing the same problem!

tardich, 2025-01-20:
@iSenne Are you using Koennkk helm chart for z2m deployment?  If so, how do you avoid the CM configuration.yaml in first place to perform the upgrade with a writable config?

iSenne, 2025-01-20:
@tardich I’m not using the Koenkk Helm chart; I’ve created my own. For the upgrade, I temporarily used a writable config, allowed z2m to perform the migration, and then extracted the updated configuration into my ConfigMap. It’s a manual step, but I’ve only had to do this once in the past three years, so it hasn’t been a recurring issue.

tardich, 2025-01-21:
@iSenne OK, did the same to be able to perform my upgrade. Always try not to build my own chart when one's available but the only ones I found interesting for 2.0.0 (one from Koenkk and the one at TrueCharts) do not work as expected. 

Thanks for the tip!

ruifung, 2025-01-21:
I don't use any helm charts and just deployed mine with kustomize.

For me, I specify configuration via environment variables and treat the configuration.yaml file as application state, therefore sidestepping the issue entirely.

So I literally have a configmap that gets used in envFrom. Same goes with a Secret with mqtt credentials, network keys and whatnot.

frank-f, 2025-01-22:
I am using the Helm chart to deploy z2m via ArgoCD.

First issue is, that I need to manually specify the image.tag to actually deploy 2.0.0 (otherwise it always uses 1.37.1)...? That is maybe why not more people have discovered the next issue yet.

Second is, as already mentioned here, that z2m always tries to modify configuration.yaml which will never work. I have no clue what it's trying to do, but the Helm Chart should use a template that already includes these changes.
Also I'm wondering why it's even a fatal error when the configuration file is not writable. I'd be perfectly fine with z2m having to do these changes on every launch. From a container perspective certainly much better than crashing.

@Koenkk I feel this needs more attention. At least updating the configuration.yaml template seems like a fairly small change to me and it would allow 2.x containers to start at all.

Koenkk, 2025-01-22:
It's already being discussed here https://github.com/Koenkk/zigbee2mqtt-chart/issues/27 / https://github.com/Koenkk/zigbee2mqtt-chart/pull/29

closed 2025-03-31 by github-actions[bot]