Koenkk/zigbee2mqtt #23591: [New device support]: MOES Tuya ZigBee Smart Sliding Window Pusher TS0601

effoc, 2024-08-11:
### Link

https://vi.aliexpress.com/item/1005006923987304.html?gatewayAdapt=glo2vnm

### Database entry

Device type     EndDevice Zigbee Model     TS0601 Zigbee Manufacturer     _TZ3210_5rta89nj Support status      Null (how_to_add_support) IEEE Address     0x94b216fffee5f184 Network address     0x60CE / 24782 Power     Battery  Interview completed     True

### Comments

Im way to stupid for this tried the guide to logg the device bur no succes.
Dont know if this is to any help https://community.hubitat.com/t/moes-tuya-zigbee-smart-sliding-window-pusher/139006

### External definition

```shell
const {identify, battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZ3210_5rta89nj',
    description: 'Automatically generated definition',
    extend: [identify(), battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

effoc, 2024-08-19:
https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2270195989

Copy this in to you're /config/zigbee2mqtt/configuration.yaml

homeassistant: true
external_converters:
    - ZC-LP01.js
    
    Make i file under it and name it ZC-LP01.js
    /config/zigbee2mqtt/ZC-LP01.js
    
    Coppy this in to you're file
    
    const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZ3210_5rta89nj']),
    model: 'ZC-LP01',
    vendor: 'Moes',
    description: 'Sliding Window Pusher',
    options: [exposes.options.invert_cover()],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
    ],
    meta: {
        tuyaSendCommand: 0x04,
        tuyaDatapoints: [
            [102, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(0), OPEN: tuya.enum(1), STOP: tuya.enum(2)})],
            [104, 'position', tuya.valueConverter.coverPosition],
            [114, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(0), OPEN: tuya.enum(1), STOP: tuya.enum(2)})],
        ],
    },
    extend: [
        tuya.modernExtend.dpBattery({dp: 4}),
    ],
};

module.exports = definition;

closed 2024-12-23 by github-actions[bot]

CocoLeo3680, 2025-02-16:
Thanks
I have followed these steps but zigbee2mqtt does not recognize the device

Could you help me?

lattice0, 2025-02-17:
Why is this closed as not planned?

alex-ong, 2025-04-12:
Join the existing thread please!

Note that zigbee2mqtt had a configuration version change, you have to put them in a magic folder (`/homeassistant/zigbee2mqtt/external_converters`, or something else depending on how you installed zigbee2mqtt).

effoc, 2025-04-12:
> Thanks I have followed these steps but zigbee2mqtt does not recognize the device
> 
> Could you help me?

Try to restart the computer not only homeassistant