Koenkk/zigbee2mqtt #23597: [New device support]: Hafele lock with TY0A01 module inside

idmilao, 2024-08-12:
### Link

https://hafelehome.com.ph/pages/el7800-plus-digital-door-lock

### Database entry

Last seen N/A Availability Disabled Device type EndDevice Zigbee Model TY0A01 Zigbee Manufacturer _TYST12_ie5b1vhe Support status Not supported (how_to_add_support)  IEEE Address 0x94deb8fffe33205c Network address 0x8561 / 34145 Power Battery  Interview completed True

### Comments

I am a newbie, i tried and also read https://github.com/Koenkk/zigbee2mqtt/issues/4478 but i can't just make it work

### External definition

```shell
const definition = {
    zigbeeModel: ['TY0A01'],
    model: 'TY0A01',
    vendor: '_TYST12_ie5b1vhe',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

kingp0dd, 2024-10-04:
I've been wanting this too

joederpoliveira, 2024-10-23:
> I've been wanting this too

Me too.

closed 2025-01-01 by github-actions[bot]