Koenkk/zigbee2mqtt #23598: ZG-101ZL operation mode error

Sygnus83, 2024-08-12:
### What happened?

z2m: Publish 'set' 'operation_mode' to 'nyomógomb' failed: 'Error: ZCL command 0xa4c13862ca506b12/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49905 - 1 - 190 - 6 - 4 after 10000ms)'

### How to reproduce it (minimal and precise)

Push the opration mode command/event button .

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

 20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Setup

Add-on on HAOS on Intel NUC

ksya, 2024-09-12:
Same here on 1.40 with same coordinator firmware.

oriaj3, 2024-10-15:
Same here: 
Version Zigbee2MQTT
 = [1.40.2]
Coordinator type = EmberZNet
Coordinator firmware = 7.4.3 [GA]
Adapter = Sonoff ZBDongle-e 

Error: 
z2m: Publish 'set' 'operation_mode' to 'Boton_dormitorio_jairo' failed: 'Error: ZCL command 0xa4c1386a0b24206f/1 genOnOff.write({"tuyaOperationMode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '32166'.)'

salcabo, 2024-10-28:
Same here with coordinator: Conbee II 
Version 1.40.2

xb4rt, 2024-11-21:
Same problem with:
[1.41.0]
coordynator
zStack12 20211115 (CC2531)

r0bb10, 2024-12-05:
also here

```
z2m: Publish 'set' 'operation_mode' to 'Loginovo' failed: 'Error: ZCL command 0xa4c138c85a2b3f5a/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 28815 - 1 - 31 - 6 - 4 after 10000ms)'
```

Zigbee2MQTT version 1.42.0-2
Coordinator type zStack3x0 (Sonoff-P)
Coordinator revision 20240710

ChadMatsalla, 2025-01-04:
Same here. I'm using a button that has this on the box:
```
Smart Scene Button
Model: C3006
Tuya Zigbee small circle scene switch TY-CJ-Z-K1
```
z2m: Publish 'set' 'operation_mode' to 'Circle Button 1' failed: 'Error: ZCL command 0xa4c138a2e5fe64e5/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 1552 - 1 - 105 - 6 - 4 after 10000ms)'
```

cedigang, 2025-01-09:
same here
`z2m: Publish 'set' 'operation_mode' to 'Button Bedroom 1' failed: 'Error: ZCL command 0xa4c1380777cdf444/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (30) Error: waiting for response TIMEOUT)'`

wrewolf, 2025-01-21:
Has same 
```log
2025-01-21 17:21:25z2m: Publish 'get' 'operation_mode' to '0xa4c1387c4f6983c9' failed: 'Error: ZCL command 0xa4c1387c4f6983c9/1 genOnOff.read(["tuyaOperationMode"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
```

XAV-01, 2025-03-06:
see https://github.com/Koenkk/zigbee2mqtt/issues/20407

closed 2025-05-13 by github-actions[bot]