Koenkk/zigbee2mqtt #23599: [New device support]: TuYa USB-C repeater

krystof-k, 2024-08-12:
### Link

https://www.aliexpress.com/item/1005007075890698.html

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0x70b3d52b600f2017","nwkAddr":38917,"manufId":4660,"manufName":"_TZ3000_n0lphcok","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,57345],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_n0lphcok","powerSource":1,"zclVersion":3,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"800+TZZSW31ZY101"}},"genOnOff":{"attributes":{"onOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea89768","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"800+TZZSW31ZY101","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1723484409037}

### Comments

I bought a new USB-C TuYa repeater, it is recognized as [TS0001](https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001).

### External definition

```shell
const {onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_n0lphcok',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), identify()],
    meta: {},
};

module.exports = definition;
```

ryanrozich, 2024-08-29:
+1 I also received one of these from AliExpress

[image]
[image]

External Definition

```
const {onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_n0lphcok',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), identify()],
    meta: {},
};

module.exports = definition;
```

guguxfr, 2024-09-20:
I bought one of these too, forgetting to check if it was supported by Z2M, stupid me.

I'm able to have it recognized by Z2M as TS0001 too, and can configure it in OpenHAB, but every time I switch it on by OpenHAB or the physical button, it switches back off after a few seconds.

Notice says one needs to install an app to configure the repeater, so I tried to install it but it does not detect the repeater.

SO, so far no luck.

guguxfr, 2024-10-12:
> > I bought one of these too, forgetting to check if it was supported by Z2M, stupid me.
> > I'm able to have it recognized by Z2M as TS0001 too, and can configure it in OpenHAB, but every time I switch it on by OpenHAB or the physical button, it switches back off after a few seconds.
> > Notice says one needs to install an app to configure the repeater, so I tried to install it but it does not detect the repeater.
> > SO, so far no luck.
> 
> Its just a signal repeater. When you plug it in it is already on and working. the physical button is just for pairing the device, it doesn't do anything else, no off or on. so just add it to your zigbee network and let it do its thing. its not configurable or anything.

Well, that seems strange to me, if the repeater is already on without pushing the button, why is the device sending a status saying it is off ? Also, if you push the button shortly, the device switches to on for a few seconds, then switches back to off. All that does not seem coherent with the fact that ti would be already on all the time.

Also, I did not feel like it enhanced my signal, but honestly I did not make lots of tests, it's just a feeling.

krash22reg, 2024-10-23:
Тоже купил этот продукт, отображается как выключатель. Усиливает ли сигналне понятно.
Было бы прекрасно, если бы кто то исправил и прописал все верно в базе.

[image]

qfoxb, 2025-01-09:
+1 seeing this as well

qfoxb, 2025-01-20:
Not sure if it is due to the incorrect detection, but this device is causing huge issues in my zigbee network, i've heard that some Tuya devices seem to drop zigbee messages from non-tuya devices and it appears to be happening with these, setting up a Universal Electronics XHS2-UE door sensor through the router on this device fails to configure and interview, and even crashes z2m

RHCPNG, 2025-03-22:
Still interested! Thnx

hirvesh, 2025-03-26:
Still interested in having this supported...

pflugj, 2025-06-06:
Hi.
The device is now correctly detected as router and the device picture is also correct.

[image]

Thanks

RHCPNG, 2025-06-06:
I can confirm. Thnx!