Koenkk/zigbee2mqtt #23603: Unparsed frame 0xc4 Skipped and other errors affecting the usability of Zigbee2MQTT

kitaniman, 2024-08-13:
### What happened?

I am running Zigbee2MQTT and added all the devices, but I am getting many errors as shown in the debug log below, and some features such as adding endpoints to groups  do not work or work for an endpoint and not for other endpoints in the same device.

### What did you expect to happen?

Zigbee2MQTT running with no errors or issues.

### How to reproduce it (minimal and precise)

Install ZIgbee2MQTT 1.39.0 on a device with 6.7.10.0 build 423 firmware and EFR32MG1B232F256GM32 adapter.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EFR32MG1B232F256GM32

### Setup

A Linux IoT Gateway with Debian 11 as a systemd file 

### Debug log

error 2024-08-13 15:05:15zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-13 15:17:13zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35) at SerialDriver.emit (node:events:519:28)
error 2024-08-13 15:17:13zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-13 15:19:45zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[141,0,1,5,0]} at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Ezsp.watchdogHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
error 2024-08-13 15:19:55zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[142,0,1,5,0]} at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Ezsp.watchdogHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
error 2024-08-13 15:20:05zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[143,0,1,5,0]} at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Ezsp.watchdogHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
error 2024-08-13 15:20:15zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[144,0,1,5,0]} at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Ezsp.watchdogHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
error 2024-08-13 15:20:25zh:ezsp:ezsp: Watchdog heartbeat timeout Error: Failure send nop:{"type":"Buffer","data":[145,0,1,5,0]} at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:686:23 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Ezsp.watchdogHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:795:13)
error 2024-08-13 15:20:25z2m: Adapter disconnected, stopping
error 2024-08-13 15:21:11zh:ezsp:uart: --> Error: Error: {"sequence":7} after 4000ms
error 2024-08-13 15:21:11zh:ezsp:uart: -!- break rewaiting (7)
error 2024-08-13 15:21:11zh:ezsp:uart: Can't resend DATA frame (6,6,1): 360001640006fdff0100
error 2024-08-13 15:21:11zh:ezsp:cast: No more available slots MulticastId subscription
error 2024-08-13 15:21:15zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35) at SerialDriver.emit (node:events:519:28)
error 2024-08-13 15:21:15zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-13 15:21:15zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35) at SerialDriver.emit (node:events:519:28)
error 2024-08-13 15:21:15zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-13 15:22:06zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35) at SerialDriver.emit (node:events:519:28)
error 2024-08-13 15:22:06zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped

Koenkk, 2024-08-13:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

kitaniman, 2024-08-21:
After downgrading to version 1.35.3, the errors have been resolved, resulting in a more stable and reliable experience overall.