Koenkk/zigbee2mqtt #23605: Curtain switch positioning doesn't work (TS130F)

Coollision, 2024-08-13:
### What happened?

I bought these devices: https://nl.aliexpress.com/item/1005002510471714.html?spm=a2g0o.order_list.order_list_main.5.1dc279d2cdHTjO&gatewayAdapt=glo2nld

They successfully bind as:
Zigbee Model: TS130F
Zigbee Manufacturer: _TZ3000_qa8s8vca

Functionality is mostly there, except for setting specific position, and backlight mode

### What did you expect to happen?

cover to move to the expected position and the backlight to be controlled correctly 

### How to reproduce it (minimal and precise)

buy device, install, pair, calibrate as in documentation, set position or backlight, see nothing happening

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20230507

### Adapter

Sonoff Zb Dongle-P

### Setup

Plain docker on RPI4

Coollision, 2024-08-18:
Any help on how I could solve this or debug it to see what could be the cause of this, would be very much appreciated

Koenkk, 2024-08-18:
Currently we use the standard Zigbee commands to set the position, but maybe it requires something different. The only way to find out is by connecting it to the original hub and see what commands are used there: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

aldifesu, 2024-09-28:
Hello, I have the LoraTap 2nd generation. In their configuration manual, when calibrating, they calibrate both the opening and the descent (with the Tuya coordinator). From my ignorance, I understand that a rise/opening time and another fall/closing time are established.

In fact, in this other Zemismart ZM-CSW032-D model in the specific Z2M settings there is a Time close and Time open to set. This one works perfect.

In Z2M the TS130F model, only a single calibration time is established. I think there would be 2 strokes like in the Zemismart ZM-CSW032-D.

Most blinds take longer to raise than to lower, approximately 2-3 seconds.

oyooyo, 2024-10-14:
I just bought two of these as well and I'm having the same problem. My devices are branded "LoraTap" but apart from that they look exactly the same.
Apparently these are "1st generation" models and positioning simply does not work on those:

[image]

Coollision, 2024-10-15:
@oyooyo thanks for the clarification, it looks like i have the first gen... 😢 (based on the arrows), It's too bad we can't add it through ota or something.
I did manage a workaround. Since they do report their current percentage, I wrote a small app that adds another shutter entity to HA and, based on the percentage set, stops after the right amount of time.

inspiriviri, 2024-11-17:
@Coollision what workaround did you do? I am in the same boat as all others with Loratap. Would you mind sharing the steps of your workaround? Thanks!

Coollision, 2024-11-17:
hey @inspiriviri not sure if my workaround will be of much use to you, but here is what I'm doing:

I'm running an additional container in my home setup where I replace the cover entity with one that is controlled by that container

then based on time (full open / 100 * percentage set), I send a stop command, so that way it passes the WAF

I need to make some time to clean it up and make it more robust but you can see the code [here](https://github.com/Coollision/mqtt-go-playground), think you can also manage something like that in node-red, or maybe through templates, ... for me, it was a fun exercise and works as a temp solution, I'll see what I do in the future

inspiriviri, 2024-11-17:
Thanks @Coollision ! That code is a nice hint and maybe I can do something similar in HA templates. 
BTW, are you aware of any zigbee curtain switch (similar to this Loratap one) which has complete working functionality (time calibrate, position % setting)?

Coollision, 2024-11-17:
@inspiriviri np, not yet, currently only have 2 electrical shutters, and no plans to buy a new controller atm

closed 2025-01-25 by github-actions[bot]