Koenkk/zigbee2mqtt #23607: [New device support]: Nedis SmartLife Rookmelder - Zigbee 3.0

Beer17HWAM, 2024-08-13:
### Link

https://nedis.com/en-us/product/safety-and-security/detectors/smoke-alarm/550782095/smartlife-smoke-detector-zigbee-30-battery-powered-sensor-life-cycle-10-year-en-14604-max-battery-life-60-months-85-db-white-1-pcs

### Database entry

{"id":21,"type":"EndDevice","ieeeAddr":"0x847127fffea481a3","nwkAddr":37278,"manufId":4098,"manufName":"_TYZB01_wqcac7lo","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,1282],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB01_wqcac7lo","appVersion":68,"powerSource":3,"zclVersion":2,"stackVersion":0,"hwVersion":1,"dateCode":"20180531"}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffebb52b8","zoneState":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffebb52b8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"20180531","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1723557274303}

### Comments

Please add support for this device:

https://www.bol.com/nl/nl/p/nedis-smartlife-rookmelder/9300000130473366/?Referrer=ADVNLGOO002063-S--9300000130473366&gad_source=1&gclid=CjwKCAjw1920BhA3EiwAJT3lSYMabZujlQsOGDhxm_XBWgRMHBs17ZFAVDzWERLChHRj8m38y4Wk_RoCTxAQAvD_BwE

It's now detected as a TS0205 but that's not good. It should be the same as a:

KlikAanKlikUit Wireless Smoke Detector | ZSDR-850 or a Heiman Smoke Sensor | HS1SA

State:
{
    "battery": null,
    "battery_low": false,
    "last_seen": "2024-08-13T15:54:34+02:00",
    "linkquality": 148,
    "smoke": false,
    "tamper": false,
    "ac_status": false,
    "battery_defect": false,
    "restore_reports": false,
    "supervision_reports": false,
    "test": false,
    "trouble": false
}

Friendly name
RookmelderWoonkamer
Description
Last seen
15 minutes ago
Availability
Online
Device type
EndDevice
Zigbee Model
TS0205
Zigbee Manufacturer
_TYZB01_wqcac7lo
Description
Smoke sensor
Support status
Supported

IEEE Address
0x847127fffea481a3
Network address
0x919E
Firmware build date
20180531
Manufacturer
[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
[TS0205](https://www.zigbee2mqtt.io/devices/TS0205.html#tuya-ts0205)

It's working but I get an error in de log:

2024-08-13 15:35:52Failed to configure '0x847127fffea481a3', attempt 1 (Error: ZCL command 0x847127fffea481a3/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37278 - 1 - 175 - 1 - 7 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:561:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:69:13) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:287:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/app/lib/extension/configure.ts:122:13))

[Heiman Smoke Sensor](https://zigbee.blakadder.com/Heiman_HS1SA.html)	HS1SA

### External definition

```shell
const {battery, identify, iasZoneAlarm, iasWarning} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0205'],
    model: 'TS0205',
    vendor: '_TYZB01_wqcac7lo',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), iasWarning()],
    meta: {},
};

module.exports = definition;
```

Beer17HWAM, 2024-08-13:
Log error is gone with zigbee2mqtt version 1.39.1 but he still thinks it's a TS0205. I don't know if it needs to unjoin and join again to change this. I don't know if you reconize this device as a Nedis device like the KlikAanKlikUit Wireless Smoke Detector | ZSDR-850 or a Heiman Smoke Sensor | HS1SA.

Beer17HWAM, 2024-12-14:
Please add this item.

Koenkk, 2024-12-15:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `ZBDS10WT.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-12-15 by Koenkk