Koenkk/zigbee2mqtt #23610: Third Reality (3RSB015BZ) Power Source Incorrect

milliman05, 2024-08-13:
### What happened?

The Third Reality Shades (3RSB015BZ) are incorrectly identified as Main (single phase) powered when they are actually battery powered.

Not sure it's related but battery drain is pretty significant on some shades (I own 19 connected through Z2MQTT) - maybe related to how often the devices are pinged for Availability.

### What did you expect to happen?

The power source should be changed to Battery.

### How to reproduce it (minimal and precise)

Device is incorrectly configured on front-end.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.2 [GA]

### Adapter

EmberZNet

### Setup

Add-on in Home Assistant through Proxmox VE

mrmaximas, 2024-08-13:
please post database.db entry for this device

milliman05, 2024-08-13:
Is this what you need:

`{"id":176,"type":"EndDevice","ieeeAddr":"0x282c02bfffea050d","nwkAddr":21001,"manufId":4659,"manufName":"Third Reality, Inc","powerSource":"Mains (single phase)","modelId":"3RSB015BZ","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,6,258],"outClusterList":[6,258,25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":114,"batteryVoltage":53}},"genBasic":{"attributes":{"modelId":"3RSB015BZ","manufacturerName":"Third Reality, Inc","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":2,"hwVersion":2,"swBuildId":"v1.00.72"}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xdc8e95fffe08aa90","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0xdc8e95fffe08aa90","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}}},"appVersion":72,"stackVersion":2,"hwVersion":2,"swBuildId":"v1.00.72","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1517847420},"lastSeen":1723584567509}`

mrmaximas, 2024-08-13:
this device reported > "powerSource":1

milliman05, 2024-08-13:
They show as mains powered in this screenshot:
[image]

mrmaximas, 2024-08-14:
> They show as mains powered in this screenshot: [...]

"powerSource":1  = mains powered
"powerSource":3 = battery powered

milliman05, 2024-08-15:
I suppose I'm still confused.  Does that mean this is something fixable, because they are very much battery powered?

closed 2024-12-22 by github-actions[bot]