Koenkk/zigbee2mqtt #23613: [New device support]:

slayer, 2024-08-13:
### Link

https://www.aliexpress.com/item/1005006883575693.html

### Database entry

none

### Comments

Hey

I've tried to connect  "[Tuya Zigbee Smart Circuit Breaker Photovoltaic Bidirectional Metering Energy Meter](https://www.aliexpress.com/item/1005006883575693.html)"
[image]

[image]
[image]
[image]

Is there external converter for this device?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_lb0fsvba',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

mrmaximas, 2024-08-13:
@slayer please find database.db file in the z2m dir, for example /zigbee2mqtt/data/database.db

slayer, 2024-08-14:
thought it is sqlite db, but it's plain file, thanks

```json
{"id":37,"type":"Router","ieeeAddr":"0xa4c1385dc56909fd","nwkAddr":4681,"manufId":4417,"manufName":"_TZE204_lb0fsvba","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\rSO.i\u000eSO.i\u000eSO.i\u000fSO.i\u000fSO.i\u000fSO.i\u0010SO.i�RO.i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_lb0fsvba","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1723628114269}
```

mrmaximas, 2024-08-14:
https://github.com/Koenkk/zigbee2mqtt/issues/23502

slayer, 2024-08-14:
Thanks, I've tried converter from that thread but it is not usable, since Circuit Breaker fails to `wrong_frequency_threshold`  error, and I don't know how to change this threshold (or even is it real `wrong_frequency_threshold` or something else)

[image]

mimich38, 2024-11-18:
Hi
I am not a geek , this product is recognize under Tuya ... but noot under MQTT
Regards

closed 2025-01-25 by github-actions[bot]